raspiblitz/raspiblitz #526: Waiting for final blockchain sync going backwards

DCress12, 2019-04-16:
Hello lovely people.

My waiting for final blockchain sync going backwards? I have been running my Blitzs for about 1 month with no problems but now this? I upgraded to 1.1 which helped for about a day and now back again to sync that does not end and is going backwards.

Can any one help.

Thanks

hupajupe, 2019-04-16:
Hello,
What are you using for your storage? I had the same problem as i was using an old 2.5" laptop hard drive connected to my PI with a USB adapter. The problem was resolved when i switched over to using an SSD instead. 

Also if your internet connection is too slow you may never sync the chain.

DCress12, 2019-04-17:
I am using a new Toshiba Canvio basic from the Amazon shopping lists.

My internet is 100Mbps down and 6Mbps up so not bad.

One other factor is that I recently set up a Pfsense router that my Blitzs is now on. It did work for a day or so before my problem started so I do not think that is the problem?

fluidvoice, 2019-04-17:
Try installing a [Bitcoin node](https://bitcoincore.org/en/download/) on your PC and see if it seems to be syncing at a good pace... catching up to the current block.  If not, then it could be your ISP slowing down the bitcoin traffic? or perhaps your router is prioritizing other internet traffic? dunno.

nicewedda, 2019-05-02:
Hi there, it seems that I have a similar problem with blockchain sync. I haven't been able to finish the synchronization for over a month because it keeps interrupting or the computer reboots. i don't know what it is, but it keeps telling me that i have to log in again and unlock the wallet or re-sync cause the blockchain is corrupted.

My internet connection is very fast (1000Mbps up and down) and if I installing bitcoin node on my PC I get the whole blockchain in 7-9 hours.

I'm on raspiblitz v 1.0 and try now v 1.1, maybe it works better...

fluidvoice, 2019-05-02:
> Hi there, it seems that I have a similar problem with blockchain sync. I haven't been able to finish the synchronization for over a month because it keeps interrupting or the pie reboots. i don't know what it is, but it keeps telling me that i have to log in again and unlock the wallet or re-sync cause the blockchain is corrupted.
> 
> My internet connection is very fast (1000Mbps up and down) and if I installing bitcoin node on my PC I get the whole blockchain in 7-9 hours.
> 
> I'm on raspiblitz v 1.0 and try now v 1.1, maybe it works better...

1. Open a new issue. 
2. supply your debug log output: 
`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.2/home.admin/XXdebugLogs.sh && sudo chmod +x /home/admin/XXdebugLogs.sh && /home/admin/XXdebugLogs.sh | nc termbin.com 9999`
3. test your power supply:
`wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh `

DCress12, 2019-05-04:
I ended up starting from a fresh install.

I used the new backup feature.

Log into shh 

Ctrl + C

/home/admin/config.scripts/lnd.rescue.sh backup

After the fresh install

/home/admin/config.scripts/lnd.rescue.sh restore

It work but I lost 3/4 of my founds :(

 Still early days for lightning but it is getting there.

fluidvoice, 2019-05-04:
Glad it's working for you now. please close this issue.

closed 2019-05-04 by DCress12