raspiblitz/raspiblitz #527: Can't connect via ssh: "load pubkey "/Users/username/.ssh/id_rsa": invalid format"

Whatyes, 2019-04-16:
So I started setting up my raspiblitz via ssh and downloaded the BTC blockchain via torrent. A few hours later I wanted to check the current status but I can't connect via ssh. It gives me the error message
"load pubkey "/Users/username/.ssh/id_rsa": invalid format"

thelwyn, 2019-04-17:
you can try to ssh again with the -v flag to get a debug

thelwyn, 2019-04-17:
the default way to connect to the raspiblitz by ssh is password authentication, did you try to change it to connect with keys instead?

Whatyes, 2019-04-17:
I didn't change anything so I was surprised it tried to connect with keys. 
How can I change it back to password authenticaton? 
Will provide a debug in a few hours.

Whatyes, 2019-04-17:
> user-MBP-5:~ user$ ssh admin@192.168.178.108 -v
> OpenSSH_7.9p1, LibreSSL 2.7.3
> debug1: Reading configuration data /etc/ssh/ssh_config
> debug1: /etc/ssh/ssh_config line 48: Applying options for *
> debug1: Connecting to 192.168.178.108 [192.168.178.108] port 22.
> debug1: Connection established.
> load pubkey "/Users/user/.ssh/id_rsa": invalid format
> debug1: identity file /Users/user/.ssh/id_rsa type -1
> debug1: identity file /Users/user/.ssh/id_rsa-cert type -1
> debug1: identity file /Users/user/.ssh/id_dsa type -1
> debug1: identity file /Users/user/.ssh/id_dsa-cert type -1
> debug1: identity file /Users/user/.ssh/id_ecdsa type -1
> debug1: identity file /Users/user/.ssh/id_ecdsa-cert type -1
> debug1: identity file /Users/user/.ssh/id_ed25519 type -1
> debug1: identity file /Users/user/.ssh/id_ed25519-cert type -1
> debug1: identity file /Users/user/.ssh/id_xmss type -1
> debug1: identity file /Users/user/.ssh/id_xmss-cert type -1
> debug1: Local version string SSH-2.0-OpenSSH_7.9

This is the debuged ssh command
How can I change back to password authentication?

thelwyn, 2019-04-18:
Deleting the /Users/username/.ssh/id_rsa file should probably make it (if you don't use ssh key connection with other machines of course, make a backup somewhere before deleting).

And I would have a look at the ssh config files:

$HOME/.ssh/config
/etc/ssh/ssh_config

and see if some options are wrongly defined (see [http://www.delafond.org/traducmanfr/man/man5/ssh_config.5.html](url)

Especially PasswordAuthentication, PubkeyAuthentication, PreferredAuthentications

What is strange is that usually by default ssh try to use keys, if no key is found, it defaults to password authentication. I would expect the same if the key was invalid.

Have you generated SSH keys before?

thelwyn, 2019-04-18:
Oh. Are you on a Windows machine and trying to connect with Putty?

nyxnor, 2021-05-17:
Please close this.

closed 2021-05-18 by rootzoll