raspiblitz/raspiblitz #531: Error Raspiblitz: Structure needs cleaning

Jeans5, 2019-04-17:
Hi, after copying the blocks and chainstate to the Raspiblitz and it was syncing the blockchain. I think it reached block 450.000 or so and suddenly it crashed. This is my error:

Not sure what happened.

Thanks for the help!

> 2019-04-17T12:47:41Z [0%]...[16%]...[33%]...[50%]...[66%]...[83%]...[99%]...[DONE].
> 2019-04-17T12:48:03Z No coin database inconsistencies in last 6 blocks (18658 transactions)
> 2019-04-17T12:48:03Z  block index           53298ms
> 2019-04-17T12:48:03Z Opening LevelDB in /home/bitcoin/.bitcoin/indexes/txindex
> 2019-04-17T12:48:05Z Fatal LevelDB error: IO error: /home/bitcoin/.bitcoin/indexes/txindex/086625.ldb: Structure needs cleaning
> 2019-04-17T12:48:05Z You can use -debug=leveldb to get more complete diagnostic messages
> 2019-04-17T12:48:05Z 
> 
> ************************
> EXCEPTION: 15dbwrapper_error       
> Fatal LevelDB error: IO error: /home/bitcoin/.bitcoin/indexes/txindex/086625.ldb: Structure needs cleaning       
> bitcoin in AppInit()       
> 
> 2019-04-17T12:48:05Z Shutdown: In progress...
> 2019-04-17T12:48:05Z scheduler thread interrupt
> 2019-04-17T12:48:05Z Shutdown: done
>

fluidvoice, 2019-04-17:
You copied from where and using what method?  SCP, rsync, etc.?  The exact commands you used.

Jeans5, 2019-04-17:
I copied it from another HDD through SSH (with SCP) to the raspberry:

> scp -r    blocks/    admin@192.168.1.118:/mnt/hdd/bitcoin/

fluidvoice, 2019-04-17:
1. You need to verify the original copy of the blockchain DB on the source HDD works properly (is not corrupted).  You can do this by starting up the bitcoin daemon (**bitcoind**) on a Windows/Linux machine using that blockchain DB.  This is how I do this on Linux. Change -datadir to the directory containing your "blocks" and other folders.  On Windows omit "sudo":
`sudo bitcoind -printtoconsole -datadir=/run/media/brad/BLOCKCHAIN/bitcoin`
2. SCP while easy to use does not guarantee file integrity since it does not use file checksums.  You could try to copy the blocks again using **rsync** as follows (from the source machine):
`rsync -cave ssh blocks/ admin@192.168.1.118:/mnt/hdd/bitcoin/`
Note: it might appear this is locked up or doing nothing for a long time (like 20 minutes). It will take this time to calculate all the file checksums **before** it begins to copy the files - but it will only copy the files that are different (checksum) or not in the destination dir.

Jeans5, 2019-04-19:
Many thanks for the comprehensive explanation.
I already started with following the troubleshoot guide from Stadicus before I saw your comment. ==> (https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_70_troubleshooting.md).
It got resolved by following the steps on "Is the disk mounted" .. strange but it worked.

closed 2019-04-19 by Jeans5

fluidvoice, 2019-04-20:
@rootzoll if Raspiblitz readme doesn't point people to this debug guide, maybe it should.  Or perhaps better fork this info and add to it what other things we've learned. https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_70_troubleshooting.md#troubleshooting

fluidvoice, 2019-04-20:
> Many thanks for the comprehensive explanation.
> I already started with following the troubleshoot guide from Stadicus before I saw your comment. ==> [...]

Do you mean that to fix your problem you did ALL the steps between these two lines?
**Is the hard disk mounted?**
and
**Are ip ports accessible through the firewall?**
This section basically describes how to properly check the hard disk for errors using `fsck`

Jeans5, 2019-04-30:
Correct. (the URL doesn't work anymore)

rootzoll, 2019-04-30:
RaspiBlot Trouble shooting is now here:
https://stadicus.github.io/RaspiBolt/raspibolt_70_troubleshooting.html

Added to the README under Support