raspiblitz/raspiblitz #532: LND not ready yet ... waiting another 60 seconds.: report Problem 1555643184

Jeans5, 2019-04-19:
Hi, I'm launching ./70initLND.sh and keeps getting several errors:

Thanks for the help.

```
*** Checking bitcoin ***
OK - bitcoind is running

*** LND Config ***
OK - exists

*** Starting LND ***
Starting LND ... give 120 seconds to init.
1555643184
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
1555643244
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
1555643304
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
1555643364
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
1555643424
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
^C

```

fluidvoice, 2019-04-20:
Sounds like a network problem.

rootzoll, 2019-04-28:
Improving in this situation (when LND is not getting ready during setup) is something I will add for future realease to consider.

Jeans5, 2019-04-30:
Any suggestions to check or resolve this? On the screen I got this: 

`Public 201.214.50.85:18333 Not reachable`

But I did port forwarding and my node is reachable when checking on bitnodes: 
https://bitnodes.earn.com/#join-the-network

rootzoll, 2019-04-30:
@Jeans5 can you open a new issue on this?

Jeans5, 2019-05-04:
@rootzoll , it resolved by just rebooting my node.

ctbon, 2020-12-12:
I'm having the same issue. It worked in my initial setup but I was unable to SSH on that setup after completing the entire setup except syncing the blockchain so I flashed the SD card again and was able to SSH and begin the process. However, LND will not sync.
`OK - chainnetwork is working

*** Check LND Config ***
OK - exists

*** Init LND Service & Start ***
stopping lnd ..
TOR was not selected
Starting LND Service ...
Created symlink /etc/systemd/system/multi-user.target.wants/lnd.service → /etc/systemd/system/lnd.service.

waiting .
waiting ..
waiting ...

          ┌────────────────────────────────────────────────────────┐
          │  Starting LND - please wait ..                         │  
          │  ┌──────────────────────────────────────────────────┐  │  
          │  │                         0                        │  │  
          │  └──────────────────────────────────────────────────┘  │  
          ├────────────────────────────────────────────────────────┤  
          │               <  OK  >        <Cancel>                 │  
          └────────────────────────────────────────────────────────┘  
                                                                      

1607804538
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
1607804598
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
1607804658
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
1607804718
`
Not expecting tech support but you might find this useful and if anyone has any tips, I would love to hear them. :)

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll