raspiblitz/raspiblitz #534: Still unable to upgrade to 1.1

lucbran, 2019-04-21:
After one week attempts (including a complete blockchain reinstall)  I am still having probs with upgrading to 1.1.
I wanted to reinstall 1.0 version but unable to locate the dowanload link (I found  https://github.com/rootzoll/raspiblitz/releases but this is not the image file)...
where can I find this? Thanks

fluidvoice, 2019-04-21:
> After one week attempts (including a complete blockchain reinstall) I am still having probs with upgrading to 1.1.
> I wanted to reinstall 1.0 version but unable to locate the dowanload link (I found [...]

Still having what problems? In your previous issue you said the upgrade completed, so what problem did you see?...

> "So I tried after reinstate original name BRNode from sudo nano /mnt/hdd/raspiblitz.conf and reburnt the SD card.  After that sys recognised the original Alias name and **the whole process of updating to 1.1 version was finalized**"

lucbran, 2019-04-21:
Actuallu it worked for a while ... then looks like blockchain went out of sync and had to sync all over again...
Now after both torrents have been downloaded, going ahead I get following on screen:
 │ Waiting for bitcoind to get ready.                                           │  
 │ error: Could not connect to the server 127.0.0.1:8332  Make sure the         │  
 │ bitcoind server is running and that you are connecting to the correct RPC 

any idea?

fluidvoice, 2019-04-21:
> Actuallu it worked for a while ... then looks like blockchain went out of sync and had to sync all over again...
> Now after both torrents have been downloaded, going ahead I get following on screen: [...]

run these commands and post the output:
`sudo cat /mnt/hdd/bitcoin/bitcoin.conf`
`sudo cat /mnt/hdd/lnd/lnd.conf`
`sudo cat /etc/systemd/system/bitcoind.service`
`sudo systemctl status bitcoind -n2 --no-pager`
`sudo bitcoin-cli getblockchaininfo`
`/home/admin/XXdebugLogs.sh`

lucbran, 2019-04-22:
Hi, thanks for your assistance:
sudo cat /mnt/hdd/bitcoin/bitcoin.conf
# mainnet/testnet
testnet=0

# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=1

# Connection settings
rpcuser=raspibolt
rpcpassword=Patr151620
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

datadir=/mnt/hdd/bitcoin
rpcport=8332
rpcallowip=127.0.0.1
rpcbind=127.0.0.1:8332

# lnd configuration

[Application Options]
debuglevel=debug
restlisten=0.0.0.0:8080
tlsextraip=0.0.0.0
maxpendingchannels=5
alias=**S.P.R.**
color=#68F442
nat=false

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# Domain, could use https://freedns.afraid.org
#tlsextradomain=lightning.yourhost.com

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

[Bitcoind]
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=Patr151620
bitcoind.rpchost=127.0.0.1
bitcoind.zmqpubrawblock=tcp://*:28332
bitcoind.zmqpubrawtx=tcp://*:28333

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6

# lnd configuration

[Application Options]
debuglevel=debug
restlisten=0.0.0.0:8080
tlsextraip=0.0.0.0
[... 89 lines cut ...]
*** LAST 20 LND INFO LOGS ***
2019-04-21 18:53:39.351 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-04-21 18:53:39.352 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-04-21 18:53:39.353 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-21 18:53:39.360 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-21 18:53:39.360 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-21 18:53:39.362 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-04-21 18:40:42.805 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-04-21 18:40:42.806 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-04-21 18:40:42.994 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-21 18:40:43.037 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-21 18:40:43.038 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-21 18:40:43.039 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-04-21 19:00:34.691 [INF] LNWL: Opened wallet
2019-04-21 19:00:35.158 [INF] LTND: Primary chain is set to: bitcoin
2019-04-21 19:00:35.196 [INF] LTND: Shutdown complete2019-04-21 19:01:35.858 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-04-21 19:01:35.858 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-04-21 19:01:35.861 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-21 19:01:35.872 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-21 19:01:35.872 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-21 19:01:35.873 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

*** LAST 20 RTL LOGS ***
-- Logs begin at Sun 2019-04-21 18:17:01 BST, end at Mon 2019-04-22 06:55:20 BST. --
Apr 21 18:40:41 S.P.R.-Nexus systemd[1]: Started RTL daemon.

fluidvoice, 2019-04-22:
I see two obvious problems in the logs:

> error: Could not locate RPC credentials. No authentication cookie could be found, and **RPC password is not set.** See -rpcpassword and -stdinrpcpass. Configuration file: [...]

So it looks like 1) your blockchain DB is possibly corrupted. and 2) RPC username + password is not set correctly via these config file entries:

`bitcoin.conf` file
**rpcuser=raspibolt
rpcpassword=**

`lnd.conf` file
**bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=**

lucbran, 2019-04-22:
agree but i guess raspibolt is default user as i dont remember having ever setup userid... how to change this?

fluidvoice, 2019-04-22:
> agree but i guess raspibolt is default user as i dont remember having ever setup userid... how to change this?

You don't need to change this.  Just make sure your password is correct - what you entered during the install/setup for "passwordB"

lucbran, 2019-04-22:
Looks like pwd in bitcoin.conf and lnd.conf files match so I guess this is probably due to blockchain corrupted... will have to reinstall it all over...

fluidvoice, 2019-04-22:
> Looks like pwd in bitcoin.conf and lnd.conf files match so I guess this is probably due to blockchain corrupted... will have to reinstall it all over...

Why were the pwd's empty above then, you removed just to not show them in this issue posting?
It also does not explain the error in the log...

> error: **Could not locate RPC credentials**. No authentication cookie could be found, and **RPC password is not set.**

lucbran, 2019-04-22:
yeah they were removed when posting...anyway starting a fresh reinstallation even if i already know funds will not be recovered with seed... thanks for your help

lucbran, 2019-04-23:
ok this is becoming quite embarassing... I formatted HD (not canceled files but just formatted) and went trhough the whole installation process..Downloaded again whole blockchain.
Created wallet and when syncronizing lnd I get Unable to resolve Host S.P.Q.R.
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

what is going on? Should I have formatted HD with full cancellation of data or what el;se?

fluidvoice, 2019-04-23:
> ok this is becoming quite embarassing... I formatted HD (not canceled files but just formatted) and went trhough the whole installation process..Downloaded again whole blockchain.
> Created wallet and [...]

Verify your node's public IP address with:
`curl -s http://v4.ipv6-test.com/api/myip.php` 
and check that it's set correctly; see this: https://github.com/rootzoll/raspiblitz/issues/50#issuecomment-453739075

fluidvoice, 2019-04-23:
Btw, you can search the past Raspiblitz issues for the errors you are seeing by clicking on the "issues" tab above then remove the "is:open" filter (so you search even closed issues), and then past in the text you want to search for.

lucbran, 2019-04-24:
Hi searched everywhere but even after new fresh installation and blockchain download, I get message : unable to resoolve host and follwoing screen appears asking to download the whole blockchain again

/home/rufus/Desktop/Screenshot from 2019-04-24 19-47-23.png

I guess i will wait new updated version and try to reinstall rt all over..

fluidvoice, 2019-04-24:
> Hi searched everywhere but even after new fresh installation and blockchain download, I get message : unable to resoolve host and follwoing screen appears asking to download the whole blockchain again
> 
> I guess i will wait new updated version and try to reinstall rt all over..
>

I doubt waiting will change anything.  You most likely need to replace some poor quality hardware... power supply, HD, HD case, cables etc.   Or you're doing something wrong with downloading and copying the blockchain without corrupting it.

lucbran, 2019-04-25:
You are probably right.. it all was previously working when I got a problem in the blockchain.. HDD may be the problem... will replace and restart it over... thanks

fluidvoice, 2019-04-25:
Run the power-supply test script.  The majority of peoples' problems are from a weak PS:

PowerTest script (all in one line):
`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/4abbdd18aa0ec44553d1fd875f2c5a6b98d9721c/home.admin/config.scripts/blitz.powertest.sh && sudo chmod +x blitz.powertest.sh && ./blitz.powertest.sh`

if that one fails to run try this one:

`wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh
`

lucbran, 2019-04-29:
Ok first one failed... second one returns: 59.1'C  600/1400 MHz 1.2V.

what doe s it mean?

fluidvoice, 2019-04-29:
> Ok first one failed... second one returns: 59.1'C 600/1400 MHz 1.2V.
> 
> what doe s it mean?

It means your power supply is poor quality.  Voltage of 1.2V is not sufficient and is causing a lot of CPU throttling (slowed down) from 1400MHz to only 600MHz.

lucbran, 2019-05-01:
I purchased starter kit which include power supply as instructed
https://www.amazon.co.uk/ABOX-Raspberry-Desktop-Starter-Upgraded/dp/B07G786RJ2/ref=sr_1_fkmrnull_1?crid=1KWQD0YVNQ6HW&keywords=abox+raspberry+pi+3+b%2B+model+b+plus+desktop+starter+kit&qid=1556743049&s=computers&sprefix=Abox+raspberry%2Ccomputers%2C186&sr=1-1-fkmrnull

Isn't enough?

fluidvoice, 2019-05-01:
> I purchased starter kit which include power supply as instructed [...]

I don't think so. This one looks good:
[image]

lucbran, 2019-05-01:
yo u mean shopping list requirement Powersupply >=3A (choose big & stable output) not sufficient? Any alternative on amazon?

fluidvoice, 2019-05-01:
> yo u mean shopping list requirement Powersupply >=3A (choose big & stable output) not sufficient? Any alternative on amazon?

The specification "3A" is irrelevant because many of them do not actually put out 5V at that current load.
yes: https://www.amazon.com/dp/B07MC7B9X3

lucbran, 2019-05-02:
ok will check if amazon in Italy has this kind of adapters..
What about USB cable to HDD? Do you think problem could be also caused by the usb cable quality?

lucbran, 2019-05-05:
So, thanks to your precious advice I solved my main 2 problems:
undervoltage problem : solved with new power ssupply 5.25V
blockchain config problem: pwas caused by hostname set to word containing dots. Changed raspi-config hostname and everything went smooth.
Now raspiblitz screen working but when trying menu #detailed wallet balance# I get: ./00mainMenu.sh: line 429: lnchannels: command not found
Press ENTER to return to main menu.

I am thinking to try 1.2 version..what you think?

fluidvoice, 2019-05-05:
> So, thanks to your precious advice I solved my main 2 problems:
> undervoltage problem : solved with new power ssupply 5.25V
> blockchain config problem: pwas caused by hostname set to word containing [...]

try running `/home/admin/91addBonus.sh` then 00mainMenu.sh

lucbran, 2019-05-05:
done but stll returns ./00mainMenu.sh: line 429: lnchannels: command not found

fluidvoice, 2019-05-05:
> done but stll returns ./00mainMenu.sh: line 429: lnchannels: command not found

try this:
`wget https://stadicus.github.io/RaspiBolt/resources/lnbalance`
`wget https://stadicus.github.io/RaspiBolt/resources/lnchannels`
`chmod +x lnbalance && chmod +x lnchannels`
`sudo cp lnchannels /usr/local/bin && sudo cp lnbalance /usr/local/bin`
`/home/admin/00mainMenu.sh`

lucbran, 2019-05-05:
No words...That's the solution!!! now everything working..Thanks very much..
will now wait few days and try the 1.2 version..
thanks again my friend!

fluidvoice, 2019-05-05:
glad its working. Please close this issue.

closed 2019-05-06 by lucbran