raspiblitz/raspiblitz #535: Is the lack of Free Mem somehow related to the Power Cable?

CommanderPoe, 2019-04-21:
I have noticed that everytime i run some sort of "heavy" process on the pi like: - Copy the whole LND folder to make a backup or - Install Zap Connet. My Free Memory goes to values below 80 which i think is kindda critical. I have experienced several times this issue and a quick restart of the pi usually fix it, but I would like to understand why this is happening? Is that related to the Power Cable? 

This is usually the mem issues it goes as low as 60.
[image]

the Cable i bought is the one that comes with the ABOX, 5V 3A

fluidvoice, 2019-04-21:
Cable and power supply ratings mean nothing as they often don't reflect reality (lies).
I can imagine power supply, or any other, problems that impact network data could possibly cause LND to use more RAM as it dynamically handles channel connectivity. I'm sorta guessing here but I've seen LND at times using excessive RAM and I'm assuming this is from some "abnormal" condition.

CommanderPoe, 2019-04-21:
> Cable and power supply ratings mean nothing as they often don't reflect reality (lies).
> I can imagine power supply, or any other, problems that impact network data could possibly cause LND to use more RAM as it dynamically handles channel connectivity. I'm sorta guessing here but I've seen LND at times using excessive RAM and I'm assuming this is from some "abnormal" network condition.

I thought it could be some sort of punctual issue with my system since i never see anyone in here complaining about the RAM. But as far as i understood from your answer just make the quick restart and move along...

fluidvoice, 2019-04-21:
> I thought it could be some sort of punctual issue with my system since i never see anyone in here complaining about the RAM. But as far as i understood from your answer just make the quick restart [...]

see my edited msg above.  You should run the power-test script to test your power supply.

fluidvoice, 2019-04-21:
PowerTest script (all in one line):
`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/4abbdd18aa0ec44553d1fd875f2c5a6b98d9721c/home.admin/config.scripts/blitz.powertest.sh && sudo chmod +x blitz.powertest.sh && ./blitz.powertest.sh`

if that one fails to run try this one:

`wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh`

fluidvoice, 2019-04-21:
If I remember correctly a result of voltage drops below 1.25V signal an inferior power supply and probably should be replaced.

CommanderPoe, 2019-04-22:
> wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh

thanks for this... seems like i need a new power supply according to the test.

[image]
 
I think u guys advised one right? would you share it with me? Im in Europe btw and thanks again

CommanderPoe, 2019-04-22:
Well it seems like im in need of a new power supply according to the test 

[image]

Guess u guys recommended one in the past right? Would u share it with me? Im in Europe btw

fluidvoice, 2019-04-22:
See the shopping lists on the main RaspiBlitz page.  @rootzoll keeps them pretty up to date with what we know at the time to be the best choices.

fluidvoice, 2019-04-22:
Even if your memory problem persists after new PS, at least you have removed one possible (and common!) source of problems.

closed 2019-04-22 by CommanderPoe

amcitaly, 2019-04-26:
> 
> 
> Well it seems like im in need of a new power supply according to the test
> 
> [image]
> 
> I think u guys recommended one in another thread right? Would u share it with me? Im in Europe btw

This is mine, but i never received messages of undervoltage problems...
[image]

fluidvoice, 2019-04-26:
> This is mine, but i never received messages of undervoltage problems...

try running this and see if anything shows up..
`sudo cat /var/log/syslog | grep -i undervolt`

amcitaly, 2019-04-26:
> 
> 
> > This is mine, but i never received messages of undervoltage problems...
> 
> try running this and see if anything shows up..
> `sudo grep "Under-voltage detected!" /var/log/syslog`

Ooops it seems I have a problem 
[image]

fluidvoice, 2019-04-27:
> Ooops it seems I have a problem

Actually it looks fine.  It says zero undervoltage reports.
So "Under-voltage detected!" was not in there?  I think it's OK.

fluidvoice, 2019-04-27:
But 1.23V is pretty bad. I'd replace that PS. Btw, found out my PS is equally as bad.

amcitaly, 2019-04-27:
> 
> 
> > Ooops it seems I have a problem
> 
> Actually it looks fine. It says zero undervoltage reports.
> So "Under-voltage detected!" was not in there? I think it's OK.

Yes probably is fine sorry, when i saw the red I though there was a problem but in fact is "0". I have 2 screen for the raspberry, with one it gives me some undervoltage problem just on starting and I don't use anymore. With the one I use now never have problems. Well I will change the PS on next upgrade to verify, thx

fluidvoice, 2019-04-27:
> I have 2 screen for the raspberry, with one it gives me some undervoltage problem

That's interesting. Which screen is this that caused under-voltage?  I would have thought all screens consume about the same amount of current.

amcitaly, 2019-04-27:
Is the most beautiful LCD the one that cause the problem. Apparently they are the same but the one that make problem has the pin on the opposite side of the one I use now. I look on the back and they are similar, if you think can help i will post a pic on the back to compare them. Is this that i bought case+lcd+pen (the problematic one)
[image]

amcitaly, 2019-04-27:
> > I have 2 screen for the raspberry, with one it gives me some undervoltage problem
> 
> That's interesting. Which screen is this that caused under-voltage? I would have thought all screens consume about the same amount of current.

Is the most beautiful LCD the one that cause the problem. Apparently they are the same but the one that make problem has the pin on the opposite side of the one I use now. I look on the back and they are similar, if you think can help i will post a pic on the back to compare them. Is this that i bought case+lcd+pen (the problematic one)
[image]