raspiblitz/raspiblitz #537: UK shopping list power supply links broken - need update

fluidvoice, 2019-04-22:
See the shopping lists on the main RaspiBlitz page.  @rootzoll keeps them pretty up to date with what we know at the time to be the best choices.
EDIT: I just noticed both UK shopping lists power supply links are no longer valid.

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issues/535#issuecomment-485437642_

CommanderPoe, 2019-04-22:
the spaniard/german list offer 4 different options for the same brand... Do u think it will help if i choose the most powerful (20W 5V 4A)? also the on/off switch its important to me.

[image]

fluidvoice, 2019-04-22:
It's hard to say.  Sometimes the printed spec's are an outright lie.  Sometimes they substitute and ship something else than what was ordered or what reviewers commented on.  I just saw in the picture that this PS showed "20awg" on the wire (20 gauge wire thickness) which is good. The smaller the number the better so 20awg is good or even 18awg but unlikely you find that and prob it's not necessary.  An on/off switch could work if it's good quality but could also drop voltage if it's not.  Best is to use a switch on your AC power outlet or "power bar"

closed 2019-04-23 by rootzoll