raspiblitz/raspiblitz #538: IDEA: Synergy with NodeLauncher

rootzoll, 2019-04-23:
A great start start for your own node is the NodeLauncher:
https://github.com/lightning-power-users/node-launcher

It would be great to make it easy to switch between NodeLauncher & RaspiBlitz. Two user cases:

- You start with NodeLauncher on your laptop - you download and verify the blockchain and maybe even have your LND funded. Now you like to move it over to RaspiBlitz. The NodeLauncher could now be a source for the Blockchain and could copy over the LND-data in a lnd.rescue backup file kind of style.

- Your RaspiBlitz has some problems. You could use the NodeLauncher to rescue your LND data over and cashout/continue there.

fluidvoice, 2019-04-23:
What's the TL;DR of node-launcher? it's quick n easy way to spin up a full node + LN using some magical implementation/packaging code?

CommanderPoe, 2019-04-24:
If we can pull out something like this, that would be awesome, as far as im concerned node launcher is pretty user friendly, for example i always keep a copy of the blockchain on my desktop computer in case something goes wrong with my hdd on the pi so i wouldnt mind to have node launcher installed on my that pc to recover my data or simply to start from scratch and move it all to the pi. This is indeed a great idea @rootzoll

rootzoll, 2019-04-24:
I think the process of installing, syncing and copying over a self-validated blockchain form your laptop to your RaspiBlitz is still a very complicated process for users. So I was thinking maybe of a Desktop-Companion-App for RaspiBlitz. But providing it for Win/OSX/Linux ... sounded like too much work to start with. But then i got reminded that something similar already exists: The NodeLauncher and it could even do more then just providing the blockchain data. Just wanted to note the idea here for now ... but if anybody has some python skills and like to connect NodeLauncher and RaspiBlitz that would be awesome.

fluidvoice, 2019-04-25:
> I think the process of installing, syncing and copying over a self-validated blockchain form your laptop to your RaspiBlitz is still a very complicated process for users.

Why do they have to copy it?  You can run bitcoind pointing to the same external USB HDD.
at most you might have to run `chown` on the files recursively.

jbhoz, 2019-04-27:
I would love to see this happen!

rootzoll, 2020-08-08:
I think v1.6 has now a good tutorial how to interact with a bitcoind and use a blockchain from your laptop. Closing this issue.

closed 2020-08-08 by rootzoll