raspiblitz/raspiblitz #539: Telegram community

cryptomulde, 2019-04-23:
Are you guys interested in creating a telegram community?
I think it would be cool to have some active raspiblitz'ers bundled in a telegram community.
What do you think? Especially what does @rootzoll think? I would volunteer as an admin.

fluidvoice, 2019-04-23:
Telegram is centralized and closed source.  We can't use anything better than that?

cryptomulde, 2019-04-23:
Nearly every crypto project is active on telegram, this is why I proposed it. I also like telegram a lot. It's the best messenger I've used so far. I agree with centralized, but we are currently using a centralized platform to exchange these messages. ;)

gen-i, 2019-04-23:
I would be for discord, telegram is quite nice but you lose the overview just if you want to help someone

openoms, 2019-04-23:
GitHub provides a searchable and linkable database of issues focused on the actual software. A chatgroup comes with a lot of of noise and valuable information gets lost easily.

CommanderPoe, 2019-04-24:
im all in for Telegram too, I volunteer for admin too, no shitcoin talk allowed!

cryptomulde, 2019-04-24:
Telegram is more to build a community and exchange information. Github for software development. I like the style they do it with WasabiWallet if someone is familiar with it.

rootzoll, 2019-04-24:
We had an IRC channel for some time. But I am neither a big IRC chatter and I dont have Telegram installed (personal choice). So these are not natural for me and I would not be around there.

I think for personal RaspiBlitz support - twitter and GitHub are a good mix. Twitter for public small questions and GitHub for searchable public issues. 

But I could see a chat channel like Telegram useful for a community exchange on "Node Management" - a place for people to offer/make channels and exchange on how to run a good and stable node for the lightning network. Does anybody know if there is already a chat/telegram channel for that?

cryptomulde, 2019-04-24:
I know Btcpayserver users, trying to create a "merchant directory", where merchants can exchange information, open channels and help others. But a more general approach like a "LN liquidity" group sounds like a nice idea, I don't know any yet.

CommanderPoe, 2019-04-24:
Not only lightning liquidity, also small questions by newbies regarding small issues that can be solved very quickly, that way it could help to mitigate a bit the charge and traffic in here, there are loads of repeated/unnecessary issues opened and closed. Maybe a telegram channel like that could help and purify a bit those issues and maybe @rootzoll could focus more on the important issues. Even if this is the final result, couple of people managing the telegram channel could come here and report weekly with well elaborated texts.

rootzoll, 2019-04-28:
I think a "RaspiBlitz Nodemanager Group" is a cool idea - it can cover all those little questions during onboarding/setup and connect the community to exchange on how to best manage your RaspiBlitz Node after SetUp.

I personally dont have Telegram - so if there is any other easy to use chatplatform that works mobile/desktop we can use for this that would be great. Maybe Slack/Rocketchat or similar? Should be decided by the person comitting to make the admin-moderation - if Telegram is the choice maybe I will install just on Desktop ;) So let me know if someone wants to commit on this ... then I will be happy to promote.

CommanderPoe, 2019-04-29:
@cryptomulde maybe you and I can moderate such a telegram group ?

cryptomulde, 2019-04-29:
I still vouch for Telegram. 
Yes we can start the group together. Do you create it @CommanderPoe? 
Think I exceeded my telegram group creations limits. 
I will start the coordination with @CommanderPoe if you guys are okay with those steps.

philbertw4, 2019-04-30:
since a lot of the activity is already happening here on github, what about a gitter room? People who don't have github accounts could just sign in with twitter or something else.

rootzoll, 2019-04-30:
@cryptomulde and @CommanderPoe sounds cool - go ahead :) And let me know how to join.

drvnoo, 2019-05-01:
What did you name the chat? I have wanted a chat for a long time, feels like this community really needs it

cryptomulde, 2019-05-01:
Group was created on telegram platform. https://t.me/raspiblitz

cryptomulde, 2019-05-16:
Today I tweeted the first time about the group. We are growing nicely. Currently 33 Blitz'ers. The idea seems to work and people join to ask easy questions, like intended. Will come back here and post once we hit 100 members :P

drvnoo, 2019-06-11:
106 users on telegram now!

rootzoll, 2019-06-11:
Any reports from the group what are the most common problems to fix or to make easier in the upcomming releases?

woeisme, 2019-07-26:
is there any alternative to Telegram for support currently?

openoms, 2019-07-26:
@woeisme
Telegram for chat and github for forum-like interaction works pretty well. Setting up a Mattermost could be a next step if someone would want to host it (see https://chat.btcpayserver.org/ )

woeisme, 2019-07-29:
> 
> 
> @woeisme
> Telegram for chat and github for forum-like interaction works pretty well. Setting up a Mattermost could be a next step if someone would want to host it (see https://chat.btcpayserver.org/ )

It doesn't work pretty well for those of us that won't use Telegram however, so you're only getting feedback from the ones who it works well for, since you only hear from them via chat. It's a self-fulfilling prophecy.

openoms, 2019-07-29:
@woeisme I meant that there is good activity and meaningful interaction on the Telegram channel. Can't deny that the UX is pretty good too. 
Please share which app would you suggest? Signal / Wire / Wickr? 
There could be multiple support channels, the like-minded people (me included) will gravitate towards the one which strikes the best balance between privacy/security and usability.

woeisme, 2019-08-07:
> @woeisme I meant that there is good activity and meaningful interaction on the Telegram channel. Can't deny that the UX is pretty good too.
> Please share which app would you suggest? Signal / Wire / [...]

I'm only really familiar with Signal and would trust that best and of course good old IRC, is there any way to mirror the telegram chat into IRC perhaps?

openoms, 2019-08-15:
@woeisme are you using Keybase?
We could try https://keybase.io/team/raspiblitz

woeisme, 2019-08-17:
Hmm I signed up but never used it, I'd be willing to try it however. 
I think I had privacy concerns because of how they attempt to link various disparate accounts together and I wasn't comfortable with one company having that information.

> 
> 
> @woeisme are you using Keybase?
> We could try https://keybase.io/team/raspiblitz

phenomeus, 2020-06-17:
why not use this? https://gitter.im

DaniW42, 2020-08-13:
+1 for Gitter, why not just give it a try?

phenomeus, 2020-08-13:
And for the development environment its much better, so basically for @rootzoll and @openoms and the whole team ;) go for Gitter :)

rootzoll, 2020-08-13:
From my view its up to the person that commits to do the chat support to decide what platform fits best. @openoms if i should somewhere open an account let me know.

phenomeus, 2020-08-14:
@openoms why not try Gitter? Its perfect in the github world because you can browse history, search, have channels, and also crosslink to PR and other parts in the github repo. 

Its like slack or teams but in the github/gitlab world.

openoms, 2020-08-25:
I've never used gitter and don't feel the need to have another platform. We are active on Github, Twitter, Telegram, mm.fulmo.org and Keybase already. There is even an unused IRC channel on Freenode. The technical discussion is plenty and clear enough for me on Github, but occasionally very happy to have a video (or if very lucky in person) chat. 
Same as @rootzoll let me know if it worth to join somewhere.

phenomeus, 2020-08-27:
isn't telegram the main support channel for real time chat? and thus its a bad experience when you get the people entered that can't really follow a good chat history or have powerful search tools. 

gitter is the same experience as Irc or slack but can implement GitHub the best because its from the same people. and the best part: you can login in with GitHub / gitlab account. for me its a no brainer to try this platform. its working with apps, desktop and even in the browser. :) 

twitter is not for support really, its just good for announcements.

openoms, 2020-10-10:
Got two new language specific groups started on Telegram in Spanish and Russian:
https://t.me/raspiblitz_ES
https://t.me/Raspiblitz_ru_chat

ChuckNorrison, 2020-10-29:
Hey i like telegram a lot and iam heavy using it for a lot of stuff. Telegram is able to get some open source bots developed. It would be a great project to get a raspi telegram bot running with some monitoring/control behaviour. I did something like that in the past as conversation handler with inline keyboards. 

Anyway, i like to chat too :)

Telegram and Slack/Discord are the best choices today to build up specialized communities

nyxnor, 2021-04-18:
This is done.
There is keybase, mattermost (fulmo) and telegram.
The most open source is mattermost, keybase is rarely used by the Blitz community and Telegram is pike, because of the group features. I also would like to move off telegram, but the major bitcoin groups are there.
Pls close this.

ChuckNorrison, 2021-04-19:
A chat function within the blitz with i.e Sphynx Relais chat could be interesting to connect all blitz nodes to a chat on purpose ✌️

nyxnor, 2021-04-19:
@ChuckNorrison Openoms is running a tribe https://github.com/rootzoll/raspiblitz/issues/2089#issuecomment-805789854 :+1:

openoms, 2021-04-19:
Telegram certainly has the most activity now. 
Seems sufficient together with the discussions on github.
Closing.

closed 2021-04-19 by rootzoll