raspiblitz/raspiblitz #540: Is it safe/possible to upgrade to lnd v0.6 while still using v1.1 of Raspiblitz?

Z-and, 2019-04-23:
Subject says it all, just wondering if it's safe to upgrade to lnd 0.6 instead of waiting for v1.2 of Raspiblitz to come out.

walletofsatoshi, 2019-04-24:
This is next critical upgrade for raspiblitz considering static channel backup on new lnd upgrade should be priority

drunkenclam, 2019-04-25:
Yes, you can just stop lnd and then exchange the lnd and lncli binary files, they're in /usr/local/bin. Then restart.

frennkie, 2019-04-27:
@Z-and I can confirm.. completely safe.

Z-and, 2019-04-29:
Great thanks! At this point I might just wait for v1.2 as it seems to be close to completion

bereska, 2019-05-01:
@Z-and, @drunkenclam, how exactly did you guys upgraded your lnd to 0.6? a step-by-step guide would be grade for less tech people like me) thanks

fluidvoice, 2019-05-01:
> @Z-and, @drunkenclam, how exactly did you guys upgraded your lnd to 0.6? a step-by-step guide would be grade for less tech people like me) thanks

On way would be to install/upgrade Raspiblitz to (pre-release) v1.2 which uses LND 0.6-beta (beta meaning pre-release).  Keep in mind v1.2 is also not yet formally released so you might hit some issues, but I installed it fine running on my Pi.  The install/upgrade is the same process as v1.1... burn the SD card, etc...

[image]

bereska, 2019-05-01:
@fluidvoice, thank you. The reason I asked about a manual upgrade of lnd to 0.6.1 is to check the performance improvement lnd devs were talking about. And I have a BTCPay server running on the same pi. Besides I manually added some packages and dependancies. So re-burning SD would mean re-installing and fine-tuning all that again

fluidvoice, 2019-05-01:
> @fluidvoice, thank you. The reason I asked about a manual upgrade of lnd to 0.6.1 is to check the performance improvement lnd devs were talking about. And I have a BTCPay server running on the same [...]

Prob a question better asked to the LND dev's or just find and follow their upgrade doc's.

drunkenclam, 2019-05-01:
@bereska If you installed BTCpay server and other packages you should also be able to download the LND binaries and copy the files to /usr/local/bin folder? If not, take a few minutes and google how to do that, it helps you much when doing basic commandline work...

bereska, 2019-05-01:
@drunkenclam, thank you. I do know how to handle binaries. I just thought there is something raspiblitz specific not to break anything

fluidvoice, 2019-05-01:
Are you guys sure that the new LND release does not contain any config changes or changes other than simply updated binaries?

drunkenclam, 2019-05-01:
That's exactly how you update LND. If you download the current 0.6 beta, all you get are those two binary files...

I'm running on LND master (compiling from source on the raspi) and I update nearly every week, depending on what was merged. You won't break anything this way.

bereska, 2019-05-10:
for those who want to do it the right way https://gist.github.com/daniel-vera-g/92b33ffc6b7d4754056254f92ede4a1b

frennkie, 2019-10-03:
@Z-and I believe this is outdated now -> closing.

Please urgently make sure that you are running at least `LND` version **0.7.1** as a security flaw has been discovered which was fixed in that version.

closed 2019-10-03 by frennkie