raspiblitz/raspiblitz #541: Lcd does not indicate

Lupus-Pluvia, 2019-04-24:
hello have the following problem despite installation, the display does not show me the info display on the resberry flash.  just stay wise only at the start comes a bite text.  Display is Kedai Version 6.1.3 please help

fluidvoice, 2019-04-24:
Post a picture of what you see on the LCD and HDMI screen during bootup, a picture of the back of the LCD and the link to where you bought the LCD.  If you plug a keyboard into the Pi are you able to access the terminal/command-line when pressing Ctrl-Alt-F2 ?

Lupus-Pluvia, 2019-04-25:
[image]
[image]
[image]
[image]
[image]

I bought it here
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.de%2Fulk%2Fitm%2F191921112753
From the principle, the output should be dipped that it is not output via HDMI.

Lupus-Pluvia, 2019-04-25:
[image]
The display looks like this except that it is the version 6.1.3 and from 2018 can only photograph the back tonight

fluidvoice, 2019-04-25:
It it listed in one of the shopping lists?

Lupus-Pluvia, 2019-04-25:
[image]

had forgotten another picture

Lupus-Pluvia, 2019-04-25:
No it is not thought I could take what I had with me

Lupus-Pluvia, 2019-04-25:
but if the display from the list is used to 100% it can be ordered

Lupus-Pluvia, 2019-04-25:
I have now ordered the display from the list, I have to pay attention to during commissioning or is that automatically?

fluidvoice, 2019-04-25:
> had forgotten another picture

that looks good/normal, seems like it's booting up correctly.  This is perhaps a driver issue (if) that LCD is actually supported.

fluidvoice, 2019-04-25:
> I have to pay attention to during commissioning or is that automatically?

I don't understand the question.

fluidvoice, 2019-04-25:
> I have now ordered the display from the list

What is the link to the new LCD you bought?

Lupus-Pluvia, 2019-04-25:
this is the link that was in the list

closed 2019-04-25 by Lupus-Pluvia

fluidvoice, 2019-04-25:
> this is the link that was in the list

There are many lists.  One for each country - USA, UK, Germany, France, Spain, etc.  I'm just curious which one you bought.