raspiblitz/raspiblitz #542: free mem dropping

philbertw4, 2019-04-25:
I wasn't sure whether to re-open #423 or not, or to start a new issue. In that issue it was mentioned that the free memory problem has been addressed for the 1.1 release. However, I have been running 1.1 for the last couple weeks, and, unfortunately, seemed to have had the same issue.

This morning I noticed on the display that the free memory was displayed in red. It was down to 72M / 927M when I rebooted it. Everything seems fine after the reboot so far.

Please let me know if there is a particular command to run that would help in debugging.

fluidvoice, 2019-04-25:
No. Log rotation had to do with disk space not memory usage.
**How many channels do you have open and connected peers?**
A screenshot would help.  I think (but not sure) it's also possible that network connectivity problems could increase LND memory usage.

philbertw4, 2019-04-25:
Thanks for the quick reply. I have 4 channels open (all quite small in capacity) and 5 connected peers. I am running it over tor which might be making it even less well connected.

CommanderPoe, 2019-04-25:
As i said in #535 I noticed the RAM dropping significantly when some "heavy" (if we can call it that way) task is performed, for example when backing up the lnd folder and previously due to the lack of space cuz the var was filling up pretty quickly (solved with V1.1). 
I ordered a new power supply since the tests i ran showed it was underperforming, so hopefully the new cable solves it. Anyways the reboot always does the magic, but its pretty annoying to shutdown lnd and bitcoind to reboot and wait for the channels to synch again.

amcitaly, 2019-04-25:
I have fix the problem after many tries, but the compromize (@fluis to run just 2 channel on the same peer

CommanderPoe, 2019-04-25:
> I have fix this problem after many tries, but the compromize (@fluidvoice thanks) for me is to run just 2 channel on the same peer. If not I let the autopilot on, but I don't add any more channel. Now the memory is always around 350-400 and is all very stable. Maybe try close all channels and off the AutoPilot, then after restart put on the AutoPilot and let him go and don't add manually any channels for some day to see how it react

I would never recommend to anyone to keep the autopilot ON... At least not at this early stage of the network, autopilot will get very efficient in the future, but today is way much better if one manage they channels manually. Apart from that i dont think Autopilot is the main reason for the low levels of RAM.

fluidvoice, 2019-04-25:
> Anyways the reboot always does the magic, but its pretty annoying to shutdown lnd and bitcoind to reboot and wait for the channels to synch again.

I'm wondering if it would be sufficient (It's certainly way faster) to just stop and restart the LND service - I'm assuming that LND is the process gobbling up all the memory.  So try this and let me know.  If it works perhaps we can change the scripts to detect low memory condition and just prompt the user to restart LND... or, just do it automatically (say using some config setting like LowMemoryRestart=lnd.service)

amcitaly, 2019-04-25:
> I would never recommend to anyone to keep the autopilot ON... At least not at this early stage of the network, autopilot will get very efficient in the future, but today is way much better if one [...]

Thank you I will try to put it off on the next reboot and manually open a channel. Anyway what I did is let the AutoPilot detect and open one channel, then put all the amount on this (1M sats) so is done. I think the AutoPilot without any satoshi left in the balance will not look for any other channel to open. For my experience, even with 2 channels open (if on different nodes) the problems starts and the raspiblitz crash and become very unstable.

CommanderPoe, 2019-04-25:
> > Anyways the reboot always does the magic, but its pretty annoying to shutdown lnd and bitcoind to reboot and wait for the channels to synch again.
> 
> I'm wondering if it would be sufficient (It's certainly much faster) to just stop and restart the LND service - I'm assuming that LND is the process using up most of the memory. So **try the command below** and let me know if it frees up sufficient memory. If it works perhaps we can change the scripts to detect low memory condition and just prompt the user to restart LND...

Maybe i poorly expresed myself, when i said reboot i meant it only for LND, a simple LND reset usually works as expected.

>  or, just do it automatically (say using some config setting like LowMemoryRestart=lnd.service)

This is interesting, im gonna try it.

CommanderPoe, 2019-04-25:
> Thank you I will try to put it off on the next reboot and manually open a channel. Anyway what I did is let the AutoPilot detect and open one channel, then put all the amount on this (1M sats) so is [...]

Maybe your issue is more related to power supply, refer to the thread #535 and run the respected powertests and see the results you get, you might be in need of a new cable. I ordered one as well to try it out since mine was underperforming.

fluidvoice, 2019-04-25:
> > or, just do it automatically (say using some config setting like LowMemoryRestart=lnd.service)
> 
> This is interesting, im gonna try it.

I didn't mean that such a setting exists.  It doesn't. What I meant was that we could modify the script to use a config like that.

fluidvoice, 2019-04-25:
> > > or, just do it automatically (say using some config setting like LowMemoryRestart=lnd.service)
> > 
> > 
> > This is interesting, im gonna try it.
> 
> I didn't mean that such a setting exists. It doesn't. What I meant was that we could modify the scripts to create and use a config setting like that.

Perhaps might be a setting for the amount of free memory (low-water mark) such that LND and Bitcoind get restarted when memory goes below the set value.  This allows the number to be variable.  
Eg...  
LowMemResetBitcoinMB=32
LowMemResetLndMB=32

@rootzoll what do you think?

fluidvoice, 2019-04-25:
@CommanderPoe please let us know if **restarting only LND** is sufficient to fix the observed problems due to low memory condition. This will make it easier to automate this in the scripts.

CommanderPoe, 2019-04-26:
> @CommanderPoe please let us know if **restarting only LND** is sufficient to fix the observed problems due to low memory condition. This will make it easier to automate this in the scripts.

At least for me it does the magic. I have had the RAM issue for a variety of issues and always a simple restart of lnd fixes it.

philbertw4, 2019-04-30:
Sorry for the late reply. I've not experienced the issue again on my end after restarting the entire pi. If I do experience it, I'll restart LND only first and see if that fixes it.

nyxnor, 2021-04-18:
pls close this.