raspiblitz/raspiblitz #543: Endless loop lnd wallet and filtering blockchain

fdvanoye, 2019-04-25:
Hi All,

I had my node running for a few weeks and last week it crashed because of the under voltage issue. I fixed the under voltage issue and tried to recover the system. System starts ok, can unlock the wallet and after that it I get the Lightning Filtering Blockchain screen with ?/ 573137. After a while it goes back to the wallet unlock screen and loops endlessly. How can I resolve this?

livestradamus, 2019-04-25:
Mine is looping as well but mine is the initial set up.

fdvanoye, 2019-04-25:
I flashed the card again, ran ./60finishHDD.sh, ran  70initLND.sh but the problem persists. Anybody got other suggestions?

fluidvoice, 2019-04-25:
Guys, nobody can help you if you don't supply detailed information about your problem ... and, **open your own issue**, don't post in someone else's issue:
See these:
https://github.com/rootzoll/raspiblitz#support 
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-do-i-generate-a-debug-report

fdvanoye, 2019-04-25:
my  generate debug report

*** RASPIBLITZ LOGS ***
blitzversion: 1.1
chainnetwork: bitcoin / main
 18:30:46 up  2:01,  2 users,  load average: 0.59, 0.48, 0.41

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-04-25 16:30:22 BST; 2h 0min ago
  Process: 812 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 830 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─830 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bi…id

Apr 25 16:30:22 Txxxxxxx systemd[1]: Starting Bitcoin daemon...
Apr 25 16:30:22 Txxxxxxx systemd[1]: Started Bitcoin daemon.

*** LAST 5 ERROR LOGS ***
-- Logs begin at Thu 2019-04-25 16:29:17 BST, end at Thu 2019-04-25 18:30:46 BST. --
Apr 25 16:30:22 Txxxxxxx systemd[1]: Starting Bitcoin daemon...
Apr 25 16:30:22 Txxxxxxx systemd[1]: Started Bitcoin daemon.

*** LAST 20 INFO LOGS ***
2019-04-25T16:42:48Z ping timeout: 1200.044013s
2019-04-25T16:43:01Z socket recv error Connection reset by peer (104)
2019-04-25T16:43:02Z socket recv error Connection reset by peer (104)
2019-04-25T16:44:30Z UpdateTip: new best=000000000000000000123dbb26a8551c611b7f218522706ce31fe60b9f70bcb3 height=573202 version=0x20400000 log2_work=90.5763 tx=406246404 date='2019-04-25T16:44:21Z' progress=1.000000 cache=16.1MiB(143183txo) warning='38 of last 100 blocks have unexpected version'
2019-04-25T17:02:54Z socket receive timeout: 1201s
2019-04-25T17:03:06Z socket recv error Connection reset by peer (104)
2019-04-25T17:04:08Z socket send error Connection reset by peer (104)
2019-04-25T17:06:36Z ThreadRPCServer incorrect password attempt from 127.0.0.1:45696
2019-04-25T17:10:39Z socket send error Connection timed out (110)
2019-04-25T17:12:27Z socket recv error Connection timed out (110)
2019-04-25T17:12:51Z socket recv error Connection reset by peer (104)
2019-04-25T17:14:44Z ERROR: AcceptBlockHeader: Consensus::ContextualCheckBlockHeader: 000000000000001daed8bac77bdf571441649e51732e91bec984cf97a39005a1, bad-diffbits, incorrect proof of work (code 16)
2019-04-25T17:15:31Z UpdateTip: new best=00000000000000000009a0365997ea38cf7c52088ebfe52788a0bbbbee773530 height=573203 version=0x3fff0000 log2_work=90.576321 tx=406249907 date='2019-04-25T17:15:45Z' progress=1.000000 cache=17.5MiB(155450txo) warning='38 of last 100 blocks have unexpected version'
2019-04-25T17:15:36Z socket send error Connection timed out (110)
2019-04-25T17:18:49Z Pre-allocating up to position 0x600000 in rev01612.dat
2019-04-25T17:18:49Z UpdateTip: new best=000000000000000000059ebd54efdddd65fc449a5d1ac5a267cb50f29d402c57 height=573204 version=0x3fffe000 log2_work=90.576343 tx=406252505 date='2019-04-25T17:18:44Z' progress=1.000000 cache=18.1MiB(161079txo) warning='39 of last 100 blocks have unexpected version'
2019-04-25T17:19:35Z socket recv error Connection reset by peer (104)
2019-04-25T17:26:00Z ping timeout: 1200.005900s
2019-04-25T17:28:16Z socket recv error Connection reset by peer (104)
2019-04-25T17:30:36Z socket send error Broken pipe (32)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Thu 2019-04-25 18:12:29 BST; 18min ago
  Process: 1420 ExecStart=/usr/local/bin/lnd --externalip=${publicIP} (code=killed, signal=TERM)
 Main PID: 1420 (code=killed, signal=TERM)

Apr 25 18:12:29 Txxxxxxx systemd[1]: Stopping LND Lightning Daemon...
Apr 25 18:12:29 Txxxxxxx systemd[1]: Stopped LND Lightning Daemon.

*** LAST 5 LND ERROR LOGS ***
-- Logs begin at Thu 2019-04-25 16:29:17 BST, end at Thu 2019-04-25 18:30:47 BST. --
Apr 25 18:07:36 Txxxxxxx systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Apr 25 18:07:36 Txxxxxxx systemd[1]: Stopped LND Lightning Daemon.
Apr 25 18:07:36 Txxxxxxx systemd[1]: Started LND Lightning Daemon.
Apr 25 18:12:29 Txxxxxxx systemd[1]: Stopping LND Lightning Daemon...
Apr 25 18:12:29 Txxxxxxx systemd[1]: Stopped LND Lightning Daemon.

*** LAST 20 LND INFO LOGS ***
2019-04-25 16:36:46.773 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-25 16:36:46.802 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-25 16:36:46.802 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-25 16:36:46.802 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-04-25 16:51:58.994 [INF] LNWL: Opened wallet
2019-04-25 16:51:59.557 [INF] LTND: Primary chain is set to: bitcoin
2019-04-25 16:53:00.859 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-04-25 16:53:00.859 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-04-25 16:53:00.863 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-25 16:53:00.874 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-25 16:53:00.875 [INF] RPCS: password RPC server listening on [::]:10009
[... cut ...]
2019-04-25 18:07:37.198 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-25 18:07:37.204 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-25 18:07:37.205 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-25 18:07:37.205 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config

fluidvoice, 2019-04-25:
@fdvanoye looks like you're having network connectivity problems.  This may be causing LND to restart itself (I'm not sure what LND does in this situation) but you need to have reliable connectivity.  If you are using WiFi you might try using wired/Ethernet or investigate router problems.

fdvanoye, 2019-04-26:
I wasn't using Wifi and had a straight connection into a hub. Connection is ok because I can log into the system. Just made sure and switched from hub direct to router and restarted network and PI. This is the new output log.  The filtering lightning blockchain count (now 555734) seems to increase VERY slowly and always does an auto exit and asks for the wallet password. Very frustrating. This is the latest ./XXdebugLogs.sh output. 

*** RASPIBLITZ LOGS ***
blitzversion: 1.1
chainnetwork: bitcoin / main
 16:01:37 up  7:30,  3 users,  load average: 0.15, 0.28, 0.33

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-04-26 08:21:49 BST; 7h ago
  Process: 852 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 871 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─871 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bit…id

Apr 26 08:21:49 Tibxfool systemd[1]: Starting Bitcoin daemon...
Apr 26 08:21:49 Tibxfool systemd[1]: Started Bitcoin daemon.

*** LAST 5 ERROR LOGS ***
-- Logs begin at Fri 2019-04-26 08:20:37 BST, end at Fri 2019-04-26 16:01:37 BST. --
Apr 26 08:21:49 Tibxfool systemd[1]: Starting Bitcoin daemon...
Apr 26 08:21:49 Tibxfool systemd[1]: Started Bitcoin daemon.

*** LAST 20 INFO LOGS ***
2019-04-26T12:55:09Z UpdateTip: new best=0000000000000000001a3f39abcc5f365fde982ed45f3b764a5e4bbdf27335ff height=573324 version=0x20000000 log2_work=90.5789 tx=406556849 date='2019-04-26T12:55:06Z' progress=1.000000 cache=31.0MiB(274273txo) warning='42 of last 100 blocks have unexpected version'
2019-04-26T12:55:32Z ThreadRPCServer incorrect password attempt from 127.0.0.1:43378
2019-04-26T13:00:39Z UpdateTip: new best=000000000000000000105ac8800ca5e4c708b3b3ef4cced036f1bc8a40cafadd height=573325 version=0x20000000 log2_work=90.578921 tx=406559947 date='2019-04-26T13:00:10Z' progress=1.000000 cache=31.4MiB(278023txo) warning='42 of last 100 blocks have unexpected version'
2019-04-26T13:15:10Z UpdateTip: new best=00000000000000000012fb15e8cb57640b4429e4861fb390c32dfb9d442f3a8e height=573326 version=0x20000000 log2_work=90.578942 tx=406562109 date='2019-04-26T13:14:54Z' progress=1.000000 cache=32.9MiB(291382txo) warning='42 of last 100 blocks have unexpected version'
2019-04-26T13:24:21Z UpdateTip: new best=00000000000000000019a63ecefe36142b34635b8a33d8825637b9eb45422766 height=573327 version=0x3711c000 log2_work=90.578964 tx=406563881 date='2019-04-26T13:23:58Z' progress=1.000000 cache=33.4MiB(296122txo) warning='43 of last 100 blocks have unexpected version'
2019-04-26T13:27:35Z UpdateTip: new best=0000000000000000000589b297068272b9ab194d9f06f82c3fcd096aedb8bce8 height=573328 version=0x20000000 log2_work=90.578985 tx=406565715 date='2019-04-26T13:27:03Z' progress=1.000000 cache=33.7MiB(299188txo) warning='43 of last 100 blocks have unexpected version'
2019-04-26T13:40:15Z Pre-allocating up to position 0x5000000 in blk01613.dat
2019-04-26T13:40:15Z UpdateTip: new best=0000000000000000001a4a3e2af3cfd9ad2ec7fedfe85b4245ee3c9726f13079 height=573329 version=0x20000000 log2_work=90.579006 tx=406568489 date='2019-04-26T13:40:05Z' progress=1.000000 cache=34.3MiB(305180txo) warning='43 of last 100 blocks have unexpected version'
2019-04-26T13:42:36Z UpdateTip: new best=00000000000000000026eb3bda7421ccd5db801e645435090746f72e4f859c61 height=573330 version=0x3fff0000 log2_work=90.579027 tx=406570132 date='2019-04-26T13:42:26Z' progress=1.000000 cache=34.6MiB(307680txo) warning='44 of last 100 blocks have unexpected version'
2019-04-26T14:19:44Z UpdateTip: new best=0000000000000000001fc98b497e494d755654a37bc75e602e38a039e0f435cf height=573331 version=0x20000000 log2_work=90.579049 tx=406573236 date='2019-04-26T14:19:30Z' progress=1.000000 cache=35.8MiB(319302txo) warning='44 of last 100 blocks have unexpected version'
2019-04-26T14:20:46Z Pre-allocating up to position 0x900000 in rev01613.dat
2019-04-26T14:20:46Z UpdateTip: new best=0000000000000000001bcdc3dad0c0d0e8b5a1abcb83caa75bd17a78e0617db4 height=573332 version=0x20000000 log2_work=90.57907 tx=406576457 date='2019-04-26T14:20:15Z' progress=1.000000 cache=36.4MiB(324250txo) warning='44 of last 100 blocks have unexpected version'
2019-04-26T14:22:08Z UpdateTip: new best=000000000000000000048d938ba3163d6e1d7bf75c4b442f1dde4f3a01959296 height=573333 version=0x20000000 log2_work=90.579091 tx=406578802 date='2019-04-26T14:21:45Z' progress=1.000000 cache=36.7MiB(327177txo) warning='43 of last 100 blocks have unexpected version'
2019-04-26T14:31:23Z ThreadRPCServer incorrect password attempt from 127.0.0.1:47906
2019-04-26T14:42:05Z UpdateTip: new best=0000000000000000002234c11a531172c34b567b06f63828fd343663282ed799 height=573334 version=0x20000000 log2_work=90.579113 tx=406581866 date='2019-04-26T14:41:35Z' progress=1.000000 cache=37.5MiB(334904txo) warning='42 of last 100 blocks have unexpected version'
2019-04-26T14:47:44Z ThreadRPCServer incorrect password attempt from 127.0.0.1:48736
2019-04-26T14:49:15Z ThreadRPCServer incorrect password attempt from 127.0.0.1:48868
2019-04-26T14:57:27Z ThreadRPCServer incorrect password attempt from 127.0.0.1:49314
[... cut ...]
2019-04-26 15:58:28.665 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-26 15:58:28.669 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-26 15:58:28.669 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-26 15:58:28.670 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config

fdvanoye, 2019-04-26:
Maybe this log will help ==>

 sudo journalctl -u lnd -b --no-pager -n20
-- Logs begin at Fri 2019-04-26 08:20:37 BST, end at Fri 2019-04-26 16:14:49 BST. --
Apr 26 15:32:23 Tibxfool systemd[1]: Started LND Lightning Daemon.
Apr 26 15:47:44 Tibxfool systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Apr 26 15:47:44 Tibxfool systemd[1]: lnd.service: Unit entered failed state.
Apr 26 15:47:44 Tibxfool systemd[1]: lnd.service: Failed with result 'exit-code'.
Apr 26 15:48:44 Tibxfool systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Apr 26 15:48:44 Tibxfool systemd[1]: Stopped LND Lightning Daemon.
Apr 26 15:48:44 Tibxfool systemd[1]: Started LND Lightning Daemon.
Apr 26 15:49:15 Tibxfool lnd[2014]: unable to connect to bitcoind: status code: 401, response: ""
Apr 26 15:49:15 Tibxfool systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Apr 26 15:49:15 Tibxfool systemd[1]: lnd.service: Unit entered failed state.
Apr 26 15:49:15 Tibxfool systemd[1]: lnd.service: Failed with result 'exit-code'.
Apr 26 15:50:15 Tibxfool systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Apr 26 15:50:15 Tibxfool systemd[1]: Stopped LND Lightning Daemon.
Apr 26 15:50:15 Tibxfool systemd[1]: Started LND Lightning Daemon.
Apr 26 15:57:28 Tibxfool systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Apr 26 15:57:28 Tibxfool systemd[1]: lnd.service: Unit entered failed state.
Apr 26 15:57:28 Tibxfool systemd[1]: lnd.service: Failed with result 'exit-code'.
Apr 26 15:58:28 Tibxfool systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Apr 26 15:58:28 Tibxfool systemd[1]: Stopped LND Lightning Daemon.
Apr 26 15:58:28 Tibxfool systemd[1]: Started LND Lightning Daemon.

fluidvoice, 2019-04-26:
lets see if your syslog shows anything. Paste the URL link that results from running this:
`sudo cat /var/log/syslog | nc termbin.com 9999`

Also, lets see the results of testing your power supply.
PowerTest script (all in one line):

`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/4abbdd18aa0ec44553d1fd875f2c5a6b98d9721c/home.admin/config.scripts/blitz.powertest.sh && sudo chmod +x blitz.powertest.sh && ./blitz.powertest.sh`

if that one fails to run try this one:

`wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh`

fdvanoye, 2019-04-26:
wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh
--2019-04-26 16:37:16--  https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.36.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.36.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1835 (1.8K) [text/plain]
Saving to: ‘XXpowertest.sh.2’

XXpowertest.sh.2                        100%[============================================================================>]   1.79K  --.-KB/s    in 0s

2019-04-26 16:37:17 (16.2 MB/s) - ‘XXpowertest.sh.2’ saved [1835/1835]

***
RaspiBlitz powertest v0.1
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
***

sysbench 0.4.12:  multi-threaded system evaluation benchmark

Running the test with following options:
Number of threads: 4

Doing CPU performance benchmark

Threads started!
52.6'C 1400 MHz  1.3688V
59.1'C 1400 MHz  1.3688V
60.1'C 1400 MHz  1.2625V
60.1'C 1200/1400 MHz 1.2625V
60.7'C 1200/1400 MHz 1.2625V
61.2'C 1200/1400 MHz 1.2625V
62.3'C 1200/1400 MHz 1.2625V
Done.

Maximum prime number checked in CPU test: 10000

Test execution summary:
    total time:                          35.0226s
    total number of events:              10000
    total time taken by event execution: 140.0414
    per-request statistics:
         min:                                 11.90ms
         avg:                                 14.00ms
         max:                                 68.52ms
         approx.  95 percentile:              16.95ms

Threads fairness:
    events (avg/stddev):           2500.0000/56.94
    execution time (avg/stddev):   35.0103/0.01

58.0'C 1400 MHz  1.3688V
56.9'C 1400 MHz  1.3688V
56.4'C 1400 MHz  1.3688V
55.8'C 1400 MHz  1.3688V
55.8'C 1400 MHz  1.3688V
55.3'C 1400 MHz  1.3688V

fdvanoye, 2019-04-26:
sudo cat /var/log/syslog | nc termbin.com 9999 ===>> done

fluidvoice, 2019-04-27:
> sudo cat /var/log/syslog | nc termbin.com 9999 ===>> done

This should have resulted in printing out a URL link that we can view the output on termbin.com
Where is the link?

fluidvoice, 2019-04-27:
> 52.6'C 1400 MHz 1.3688V
> 59.1'C 1400 MHz 1.3688V
> 60.1'C 1400 MHz 1.2625V
> 60.1'C 1200/1400 MHz 1.2625V
> 60.7'C 1200/1400 MHz 1.2625V
> 61.2'C 1200/1400 MHz 1.2625V
> 62.3'C 1200/1400 MHz 1.2625V

You can see the voltage drops during the test from 1.3688V to 1.2625V and some throttling happens 1400 to 1200Mhz. Now what is the exact limit of voltage drop before problems occur (file corruption, network problems, etc.) we don't know yet.  But you want a power supply that does not drop voltage under load.

fdvanoye, 2019-04-27:
sorry, got it now: https://termbin.com/3xuz7

fluidvoice, 2019-04-27:
> sorry, got it now: https://termbin.com/3xuz7

It looks OK.  I don't see any problems in there.

fluidvoice, 2019-04-27:
> Apr 26 15:49:15 Tibxfool lnd[2014]: unable to connect to bitcoind: status code: 401, response: ""

I didn't notice this error before.  A previous issue had same with resolution: https://github.com/rootzoll/raspiblitz/issues/336#issuecomment-465833372

> Feb 20 19:34:50 raspihelx2 lnd[21174]: unable to create chain control: unable to connect to bitcoind: status code: 401, response: ""
> Feb 20 19:34:50 raspihelx2 lnd[21174]: 2019-02-20 19:34:50.912 [INF] LTND: Shutdown complete
> Feb 20 19:34:50 raspihelx2 systemd[1]: lnd.service: Main process exited, code=exited,
> 
> **LND cannot connect to bitdoind .. please check if both configs contain the same RPC password:**
> `sudo cat /mnt/hdd/bitcoin/bitcoin.conf`
> `sudo cat /mnt/hdd/lnd/lnd.conf`

fdvanoye, 2019-04-27:
Thanks for all your help guys!. Decided to do a clean setup because from the start I had a good internal connection with my Zap IOS wallet and ran into problem trying to get external working. With this second problem it would be best to do a clean install. Made a backup of the wallet and will pull the chain from another node. Good exercise ;-) Thanks again!

closed 2019-04-27 by fdvanoye