raspiblitz/raspiblitz #545: Mount HDD - Wrong fs type

StyriaTrading, 2019-04-25:
At the initialization of the HDD I get this error message:

init HDD ...

*** Checking if HDD is connected ***
OK - HDD found at sda1

*** HDD Size Check ***
OK: HDD seems big enough

*** Formatting the HDD ***
WARNING ALL DATA ON HDD WILL GET DELETED - CAN TAKE SOME TIME
Wait until you get a OK or FAIL
mke2fs 1.43.4 (31-Jan-2017)
/dev/sda1 contains a exfat file system labelled 'Lightning'
Creating filesystem with 489062384 4k blocks and 122265600 inodes
Filesystem UUID: f940d54e-e799-4351-995e-4d37b23f56cc
Superblock backups stored on blocks: 
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
        4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968, 
        102400000, 214990848

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (262144 blocks): done
Writing superblocks and filesystem accounting information: done       

format ext4 done - wait 6 secs
OK - HDD is now formatted in ext4

*** Adding HDD to the System ***
started from state(waitsetup)
OK - HDD found as sda

*** Check HDD ***
OK - HDD is formatted with ext4 and is named BLOCKCHAIN
Adding line to /etc/fstab ...
UUID=f940d54e-e799-4351-995e-4d37b23f56cc /mnt/hdd ext4 noexec,defaults 0 2
OK - HDD is listed in /etc/fstab

*** Mount HDD ***
mount: wrong fs type, bad option, bad superblock on /dev/sda1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
FAIL - was not able to mount

Anyone an idea how to fix or what is wrong?

fluidvoice, 2019-04-25:
post the output of running `sudo parted -l`

StyriaTrading, 2019-04-30:
admin@RaspiBlitz:~ $ lsblk -fa
NAME        FSTYPE LABEL      UUID                                 MOUNTPOINT
loop0                                                              
loop1                                                              
loop2                                                              
loop3                                                              
loop4                                                              
loop5                                                              
loop6                                                              
loop7                                                              
sda                                                                
└─sda1      ext4   BLOCKCHAIN 543e2619-0854-405f-977c-3605e9e6c577 
mmcblk0                                                            
├─mmcblk0p1 vfat   boot       AF14-47C0                            /boot
└─mmcblk0p2 ext4   rootfs     b57a7299-0677-4dc6-b4b0-c10f107f0f6d /

admin@RaspiBlitz:~ $ sudo parted -l
Model: Generic Flash Disk (scsi)
Disk /dev/sda: 2003GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  2003GB  2003GB  primary  ext4

Model: SD SD16G (sd/mmc)
Disk /dev/mmcblk0: 15.7GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  50.2MB  46.0MB  primary  fat32        lba
 2      50.3MB  7851MB  7800MB  primary  ext4

fluidvoice, 2019-04-30:
I don't see anything obviously wrong there.  Let's see if dmesg shows anything. Run this n post the resulting URL link:
`dmesg | nc termbin.com 9999`

fluidvoice, 2019-04-30:
also...
`sudo cat /etc/fstab`

StyriaTrading, 2019-04-30:
admin@RaspiBlitz:~ $ **sudo cat /etc/fstab**
proc            /proc           proc    defaults          0       0
PARTUUID=03d9b017-01  /boot           vfat    defaults          0       2
PARTUUID=03d9b017-02  /               ext4    defaults,noatime  0       1
UUID=543e2619-0854-405f-977c-3605e9e6c577 /mnt/hdd ext4 noexec,defaults 0 2
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

admin@RaspiBlitz:~ $ **/home/admin/40addHDD.sh**

*** Adding HDD to the System ***
started from state(waitsetup)
OK - HDD found as sda

*** Check HDD ***
OK - HDD is formatted with ext4 and is named BLOCKCHAIN
UUID is already in /etc/fstab
OK - HDD is listed in /etc/fstab

*** Mount HDD ***
mount: wrong fs type, bad option, bad superblock on /dev/sda1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
FAIL - was not able to mount

admin@RaspiBlitz:~ $ **dmesg | nc termbin.com 9999**
https://termbin.com/rhka

fluidvoice, 2019-04-30:
dmesg log shows HDD failures - which *may* be caused by your weak power supply. The log also shows under-voltage detected...

> [   20.791561] Voltage normalised (0x00000000)
> [   22.871595] Under-voltage detected! (0x00050005)
> [   44.742755] EXT4-fs error (device sda1): ext4_find_extent:921: inode #8: comm mount: pblk 244350975 bad header/extent: invalid magic - magic 0, entries 0, max 0(0), depth 0(0)
> [   44.773442] jbd2_journal_init_inode: Cannot locate journal superblock
> [   44.773464] EXT4-fs (sda1): Could not load journal inode
> [   56.151535] Voltage normalised (0x00000000)
> [   62.401580] Under-voltage detected! (0x00050005)
> [   66.551550] Voltage normalised (0x00000000)
> [  283.558153] EXT4-fs error (device sda1): ext4_find_extent:921: inode #8: comm mount: pblk 244350975 bad header/extent: invalid magic - magic 0, entries 0, max 0(0), depth 0(0)
> [  283.579490] jbd2_journal_init_inode: Cannot locate journal superblock
> [  283.579500] EXT4-fs (sda1): Could not load journal inode
> [  457.236555] EXT4-fs error (device sda1): ext4_find_extent:921: inode #8: comm mount: pblk 244350975 bad header/extent: invalid magic - magic 0, entries 0, max 0(0), depth 0(0)
> [  457.257937] jbd2_journal_init_inode: Cannot locate journal superblock
> [  457.257949] EXT4-fs (sda1): Could not load journal inode

fluidvoice, 2019-04-30:
Let's see what the PowerTest script tells us. I'm betting it will confirm bad PS...

PowerTest script (run all in one line):

`wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh`

StyriaTrading, 2019-04-30:
I tried it with another PS --> Power Test seems normal for me:

***
RaspiBlitz powertest v0.1
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
***

./XXpowertest.sh: line 22: sysbench: command not found
42.9'C  600/1200 MHz 1.2V
42.4'C 1200 MHz  1.2438V
41.9'C  600/ 600 MHz 1.2438V
42.4'C  600/1200 MHz 1.2438V
42.4'C  600/1200 MHz 1.2438V
41.9'C  600/ 600 MHz 1.2438V
42.4'C  600/ 600 MHz 1.2438V
42.9'C  600/1200 MHz 1.2438V
42.4'C 1200 MHz  1.2438V
42.4'C 1200 MHz  1.2438V
42.4'C 1200 MHz  1.2438V
42.9'C  600/1200 MHz 1.2438V
42.9'C  600/1200 MHz 1.2438V
42.9'C  600/1200 MHz 1.2438V
42.9'C  600/ 600 MHz 1.2438V

So may be an issue with the Harddisk... it's a new one, I tried it several times on Linux and Windows, worked fine... maybe you have a suggestion for me how to format the HDD to get it mounted. Thank you!

fluidvoice, 2019-04-30:
First off the power supply is weak. I recommend you buy a new one.  1.2V is def not enough and the test shows your RPi CPU is throttling (slowed down) in an attempt to compensate for the low voltage. 
Does your HDD have it's own power supply or it's connected only via USB to your RPi?

> So may be an issue with the Harddisk... it's a new one, I tried it several times on Linux and Windows, worked fine... maybe you have a suggestion for me how to format the HDD to get it mounted. Thank you!

I don't think the problem is how it's formatted. The Raspiblitz scripts do the formatting.
I think it's your power supply.

StyriaTrading, 2019-04-30:
Thank you very much for the help! Last question: What minimum specs for the PS would you recommend, so that I am on the safe side?

fluidvoice, 2019-04-30:
> Thank you very much for the help! Last question: What minimum specs for the PS would you recommend, so that I am on the safe side?

Unfortunately the specs on the web sites and printed on them are pretty much meaningless.  The reason is they tell you what the no-load voltage is and the max Amps are but that does not mean they will actually deliver both at the same time.  Meaning, most of the cheap ones lose voltage as the power (Amps) demanded goes up (one could say that spec's are a lie).  The best thing is to get one that is over-volted, say 5.25V  (not sure if 5.5V exists n might be too much) so that under a heavy load they will still stay close to 5V output.  If you read through the Amazon reviews you can often find guys who have gone to the trouble of testing and measuring the PS's output under full load.
Oh... I just dug out an old tweet of mine.  This looks like a good candidate:
https://twitter.com/fluidvoice/status/1113654715349438464

fluidvoice, 2019-04-30:
Sorry, here's the link to it: https://www.adafruit.com/product/1995

fluidvoice, 2019-04-30:
> Thank you very much for the help! Last question: What minimum specs for the PS would you recommend, so that I am on the safe side?

Btw, you didn't answer the question - does your HDD have it's own PS or it's just connected to the Pi's USB port only?

StyriaTrading, 2019-04-30:
Great! The external 2.5" HDD (Seagate) is WITHOUT own PSU - it's directly connected to the Pi's-USB-Port via the USB-Cable which was shipped with the HDD.

fluidvoice, 2019-04-30:
> Great! The external 2.5" HDD (Seagate) is WITHOUT own PSU - it's directly connected to the Pi's-USB-Port via the USB-Cable which was shipped with the HDD.

You can cheaply add a separate power supply to the HDD by using a "usb y-cable" and something like a spare USB phone wall charger - 1Amp or perhaps even less is usually enough. 

[image]

fluidvoice, 2019-04-30:
Which Seagate drive and PS do you currently have? Just curious.

StyriaTrading, 2019-05-07:
Sorry for the late reply - I tested the whole thing with another HDD (WD Elements) and it worked properly. Seems that the Seagate HDD has some damage.

closed 2019-05-07 by StyriaTrading