raspiblitz/raspiblitz #546: ENHANCEMENT: free memory monitor service

fluidvoice, 2019-04-25:
Or more generally a "resource monitor" service which keeps an eye on both SD disk space and free memory?

> > > or, just do it automatically (say using some config setting like LowMemoryRestart=lnd.service)
> > 
> > 
> > This is interesting, im gonna try it.
> 
> I didn't mean that such a setting exists. It doesn't. What I meant was that we could modify the scripts to create and use a config setting like that.

Perhaps might be a setting for the amount of free memory (low-water mark) such that LND and Bitcoind get restarted when memory goes below the set value.  This allows the number to be variable.  
Eg...  
LowMemResetBitcoinMB=32
LowMemResetLndMB=32

@rootzoll what do you think?  Is this an advisable/feasible thing to add?

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issues/542#issuecomment-486687525_

fluidvoice, 2019-04-25:
For example, perhaps **_background.sh** should check the free/available RAM once per hour the same as it is checking for under-voltage here:
https://github.com/rootzoll/raspiblitz/blob/7459786b80975fc8c2a57c8458f5587480d3a52f/home.admin/_background.sh#L65

nyxnor, 2021-04-18:
Memory is in stable consumption now, rarely this case occur anymore. Pls close this.

closed 2021-04-19 by rootzoll