raspiblitz/raspiblitz #548: Consider Bitcoin 0.18 for 1.2 milestone

frennkie, 2019-04-27:
As far as I see https://github.com/bitcoin/bitcoin/milestone/35 Bitcoin 0.18 is expected to be released "soon". So it might make sense to track this for the 1.2 milestone of RaspiBlitz.

rootzoll, 2019-04-27:
v1.2 is already in final testing. I think for the v1.2 I will stick to the old bitcoin version. 

But I mark this ticket for the next v1.3 release .. which should not take to long after v1.2.

fluidvoice, 2019-04-27:
yes. and better to wait to find out if any LN problems w/resolution with the new release first.

openoms, 2019-07-30:
Just clarified with @roasbeef: lnd is supporting the bitcoind 0.18.0 release already! The RPC changes to `sendrawtransaction` is merged in to the master post 0.18.0, so could be a problem only when building bitcoin from source. 
Updating to 0.18 will come with a higher `assumevalid` value so will make the Initial Block Download quicker.

rootzoll, 2019-08-05:
I updated the v1.3 branch now including LND 0.7.1 and BitcoinCore 0.18.0 ... tests look good on my machine. Will be part of the v1.3 final release.

fluidvoice, 2019-08-05:
> I updated the v1.3 branch now including LND 0.7.1 and BitcoinCore 0.18.0 ... tests look good on my machine. Will be part of the v1.3 final release.

You might want to hold off release until we can get Tor completely working. It seems it's failing for some even on the Pi4

rootzoll, 2019-08-05:
Sure .. fixing TOR is still on the TODO list for the v1.3 release ... or are the new versions of LND/bitcoind adding to the Tor issue?

rootzoll, 2019-08-11:
Updated to latest bitcoin core 0.18.1 release.

closed 2019-09-03 by rootzoll