raspiblitz/raspiblitz #550: Include PowerTest script one of the first things in the Install doc's

fluidvoice, 2019-04-27:
90% of issues I've looked at lately show PS's with voltage drop.  The problem source needs to be removed from the get-go.  I recommend bolding this as part of the first steps in the Raspiblitz installation guide.

> At this point I'm recommending that everyone run the PowerTest with all your normal peripherals plugged in (LCD, HDD, etc.)

Thanks for this one. Ran the test but not shure how to read/interpretet the results. Are they good or bad?

```
***
RaspiBlitz powertest v0.1
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
***

sysbench 0.4.12:  multi-threaded system evaluation benchmark

Running the test with following options:
Number of threads: 4

Doing CPU performance benchmark

Threads started!
44.5'C  600/1200 MHz 1.2V
48.3'C  600/1200 MHz 1.2V
49.9'C  600/1200 MHz 1.2V
51.0'C  600/1200 MHz 1.2V
51.5'C  600/1200 MHz 1.2V
52.1'C  600/1200 MHz 1.2V
53.2'C  600/1200 MHz 1.2V
53.2'C  600/1200 MHz 1.2V
53.2'C  600/1200 MHz 1.2V
53.2'C  600/1200 MHz 1.2V
53.7'C  600/1200 MHz 1.2V
Time limit exceeded, exiting...
(last message repeated 3 times)
Done.

Maximum prime number checked in CPU test: 10000

Test execution summary:
    total time:                          60.0314s
    total number of events:              6376
    total time taken by event execution: 240.0375
    per-request statistics:
         min:                                 27.79ms
         avg:                                 37.65ms
         max:                                162.01ms
         approx.  95 percentile:              69.93ms

Threads fairness:
    events (avg/stddev):           1594.0000/45.37
    execution time (avg/stddev):   60.0094/0.01

52.6'C  600/1200 MHz 1.2V
50.5'C  600/1200 MHz 1.2V
49.4'C  600/1200 MHz 1.2V
49.4'C  600/1200 MHz 1.2V
```

_Originally posted by @gate40 in https://github.com/rootzoll/raspiblitz/issues/450#issuecomment-487273137_

rootzoll, 2019-04-28:
In v1.2 the power/stress test is run as the first thing on install process automatically. And the results are explained to the user. If the power outlet is weak/failed or raspi gets too hot the user can continue, but is extremly advised to power down again and improve hardware before starting the further setup.

I think this covers this issue, correct?

fluidvoice, 2019-04-28:
For during the install, yeah, that sounds good.  But it might be helpful to also mention this in the doc's where it's talking about the required hardware (and the shopping lists) so that users will hopefully buy the correct/sufficient PS instead of finding out later and having to buy/order another one.  
  Ideally, maybe with reports from users, we could compile a list of Power Supplies and their performance - so have a list of good and bad ones, and ensure that the ones recommended in the shopping lists will work well.

rootzoll, 2019-04-30:
@fluidvoice On the README.md page of v1.2 branch there is now a link to a v1.2 image to download.
https://github.com/rootzoll/raspiblitz/tree/v1.2#installing-the-software

This image is a candidate to be the final v1.2 release - but there are still some small things I would like to try out until the end of the week, before merging v1.2 branch into master.

closed 2019-05-13 by rootzoll