raspiblitz/raspiblitz #552: Format HDD error

rpjcbs, 2019-04-29:
Getting the below error when doing initial set up:

format ext4 done - wait 6 secs
FAIL - was not able to format the HDD to ext4 with the name 'BLOCKCHAIN'

I'm using a Seagate 2TB HDD which I formatted to ExFAT before connecting to my Pi.

fluidvoice, 2019-04-29:
What is your output of running this from a SSH command line shell?
`lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL`

rpjcbs, 2019-04-29:
So after I got that error, I turned my Pi off and on and it's now downloading the blockchain... weird

rpjcbs, 2019-05-01:
@fluidvoice - Scrap that, the torrent blockchain and download have been stuck on 0% and 3% for over 48 hours.

So I went to try to ultra-slow method and got this error:

*** Activating Blockain Sync ***
mkdir: cannot create directory ‘/mnt/hdd/bitcoin’: Read-only file system
OK - sync is activated

*** Checking HDD ***
FAIL - HDD is empty.
admin@RaspiBlitz:~ $

fluidvoice, 2019-05-01:
Looks most likely a power supply problem.  Insufficient power causing HDD failure and/or corruption.
This "Read-only file system" error is a common symptom of that.  Which power supply and HDD are you using?  That Seagate doesn't have it's own PS, just plugged into the Pi's USB right?  You should prob power the HDD separately with a powered USB hub or a USB drive "Y-cable".

rpjcbs, 2019-05-02:
@fluidvoice thanks for your help, I purchased this - https://www.amazon.co.uk/gp/product/B00P0C4M5M/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 - off of Amazon and it came today. Just plugged it into my Pi, the HDD and power and it's downloaded further than it did yesterday.

Will leave that for a day or two and see how it goes, thanks again

fluidvoice, 2019-05-02:
> @fluidvoice thanks for your help, I purchased this - https://www.amazon.co.uk/gp/product/B00P0C4M5M/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 - off of Amazon and it came today. Just plugged [...]

both of the y-cable USB are plugged into the Pi or one is plugged into a separate power supply?
I recommend using separate PS.

rpjcbs, 2019-05-04:
@fluidvoice separate power supplies.

Blockchain downloaded, just doing all the other stuff now. Letting it run it's course, thanks for the help.

closed 2019-05-13 by rootzoll