raspiblitz/raspiblitz #553: During blockchain download, download stalls and message appears no screen session found

coinicarus, 2019-04-29:
I'm installing a new node and for some reason, my blockchain download gets stuck and this message-no screen session found appears in the cmd window in the download blockchain status area, see screenshot below. i am ready to scrap my torrent download and start a new one. I do have the screen connected and on it i just see the normal light blue screen with the login and IP info, so i dont think that error has anything to do with the actual screen. thank you for your time and assistance. 
[image]

fluidvoice, 2019-04-30:
Could be a network problem. If you are using WiFi try switching to wired/Ethernet.  
The "screen" is not a physical screen. It's a virtual terminal session, "screen" is a Linux command.

coinicarus, 2019-04-30:
> Could be a network problem. If you are using WiFi try switching to wired/Ethernet.
> The "screen" is not a physical screen. It's a virtual terminal session, "screen" is a Linux command.

Thank you for your help. I was using a wired connection, i started over from scratch again and have now gotten past this point  :)

coinicarus, 2019-04-30:
see previous comment for resolution, i didnt figure out the root cause. i restarted my blockchain download from the beginning and it went through perfectly this time.

closed 2019-04-30 by coinicarus