raspiblitz/raspiblitz #554: Undervoltage Reports

2000jago, 2019-04-30:
I have been having undervoltage reports for a while now so I ordered the power supply recommended from the shopping list which just arrived.  I shut down the node hooked up the new powersupply and to my disappointment I STILL see "6 Undervoltage reports found..." message after start up again.

Now it could just be that these are "old" voltage reports, but I can see no way to "clear" them to see if new ones are being generated.  (I would have assumed all existing undervoltage reports would be auto-cleared on a reboot but perhaps not?).

Where are these reports located and how can I delete them in order to know if new ones are still being generated?

fluidvoice, 2019-04-30:
You're better off running the PowerTest.

PowerTest script (all in one line):

`wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh`

2000jago, 2019-04-30:
Thanks!  After running the test I (still) see "6 Undervoltage reports found..."

Test result was as follows:  

***
RaspiBlitz powertest v0.1
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
***

sysbench 0.4.12:  multi-threaded system evaluation benchmark

Running the test with following options:
Number of threads: 4

Doing CPU performance benchmark

Threads started!
68.8'C 1200/1400 MHz 1.2375V
73.1'C 1200/1400 MHz 1.2375V
74.1'C 1200/1400 MHz 1.2375V
75.2'C 1200/1400 MHz 1.2375V
75.8'C 1200/1400 MHz 1.2375V
75.8'C 1200/1400 MHz 1.2375V
77.4'C 1200/1400 MHz 1.2375V
76.8'C 1200/1400 MHz 1.2375V
Done.

Maximum prime number checked in CPU test: 10000

Test execution summary:
    total time:                          45.8014s
    total number of events:              10000
    total time taken by event execution: 183.1451
    per-request statistics:
         min:                                 13.89ms
         avg:                                 18.31ms
         max:                                146.86ms
         approx.  95 percentile:              37.08ms

Threads fairness:
    events (avg/stddev):           2500.0000/32.21
    execution time (avg/stddev):   45.7863/0.01

77.4'C 1200/1400 MHz 1.2375V
73.1'C 1200/1400 MHz 1.2375V
72.0'C 1200/1400 MHz 1.2375V
70.9'C 1200/1400 MHz 1.2375V
70.4'C 1200/1400 MHz 1.2375V
69.3'C 1200/1400 MHz 1.2375V
70.4'C 1200/1400 MHz 1.2V

2000jago, 2019-04-30:
How to reset/clear these undervoltage reports?   Anyone?

bitsam, 2019-04-30:
I was getting under voltage errors and had to rebuild my node twice due to filesystem corruption. I decided to use a hdd with its own power supply. Since then, now under voltage errors.

rootzoll, 2019-04-30:
> How to reset/clear these undervoltage reports? Anyone?
The undervoltage reports are counted from the systemlog file - you can delete it to reset counts:
`sudo rm /var/log/syslog`

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll