raspiblitz/raspiblitz #559: [v1.2 testing] 05hardwareTest.sh temperature format

fluidvoice, 2019-05-01:
**(8060 centiGrad)** shown is improper format.  Should this be "**80.6 degrees Celsius**"?
The normal English words for temper
[image]
ature are Fahrenheit and Celsius.
I ran the test while still syncing just for kicks - my CPU is smoking! ;-)

fluidvoice, 2019-05-01:
Also a small typo here:
[image]

fluidvoice, 2019-05-01:
And "FAIL detected" is confusing here.  Nothing has actually failed, has it?  My system is running fine, if a tad warmly ;-)

> *** Hardware Test Report ***
> 
> RaspiBlitz Hardwaretest v0.1
> Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
> 74.1'C 1200/1400 MHz 1.2563V
> --> Temp FAIL detected
> --> Temp WARN detected
> 76.8'C 1200/1400 MHz 1.2563V
> --> Temp FAIL detected
> --> Temp WARN detected
> 77.9'C 1200/1400 MHz 1.2563V
> --> Temp FAIL detected
> --> Temp WARN detected
>

rootzoll, 2019-05-01:
I chaned temp FAIL to CRITICAL. Typo corrected.
Yeah the Centigrad was a quick way to go forward because I was not finding I working way to convert the number into float again. Its just for us in the support forums, to have a number at all. I leave to make a working conversion to a PR for a future version ;)

closed 2019-05-01 by rootzoll