raspiblitz/raspiblitz #560: [v1.2 testing] upgrades should retain WiFi and keyboard settings

fluidvoice, 2019-05-01:
This is not new to v1.2 but when I installed v1.2 I had to run raspi-config again to turn on WiFi (I don't use ethernet) and set my keyboard to the correct layout.  It's kinda a PITA to have to do this EVERY time a new Raspiblitz version is installed.  Ideally, the **/etc/wpa_supplicant/wpa_supplicant.conf** file is retained in **/mnt/hdd** somewhere that can be restored during the install procedure. Or alternately the information is prompted from the user and appended to **wpa_supplicant.conf** such as below.  The keyboard setting I'm not sure where that is stored.

> network={
> 	ssid="YOUR-SSID-STRING"
> 	psk="yourpsk123"
> }
>

frennkie, 2020-09-19:
@rootzoll Is this something we would want to include in the raspiblitz.conf?

Alternatively people could put this into the new `custom-installs.sh`, right?

These seem to be the relevant `raspi-config` parameters:

```
# System Configuration
sudo raspi-config nonint do_configure_keyboard us
sudo raspi-config nonint do_change_timezone America/New_York
sudo raspi-config nonint do_change_locale LANG=en_US.UTF-8

# set WiFi Country
sudo raspi-config nonint do_wifi_country %s"
``` 

I haven't found a way to setup ssid and wpa2-psk - but @fluidvoice mentioned this can simply be copied into /etc/wpa_supplicant/wpa_supplicant.conf .

rootzoll, 2020-09-23:
I think since v1.6 we already had a backup of wifi stettings that was restored on updates. I took the oppertunity to optmize that with the now merged PR #1584 that even prevents the extra reboot and should make this solution even better working on non-raspberrypi platforms. Putting this on final testing for v1.6.1 release.

Timezone/Keyboard dont have a high prio, because in normal setup you dont have a keyboard connected to the raspberrypi. So I will close this issue with the v1.6.1 release. If someone likes to create an extra PR with for backup/restore of the timezone/keyboard settings I will be happy to merge for future releases.

rootzoll, 2020-10-12:
OK finally tested with setting a RaspiBlitz to WIFI with a `wpa_supplicant.conf` on `boot` sd card partition and then on a recovery/update those WFI settings got preserved. Closing issue for v1.6.1 release.

closed 2020-10-12 by rootzoll