raspiblitz/raspiblitz #561: [v1.2 testing] bitcoind errors in debug.log not cleared upon restarting

fluidvoice, 2019-05-01:
I got in a circular error state where the script hits this error every time it boots up.  The debug.log should be fresh on each boot.  Everything was working so I'm not sure why 'Please restart with -reindex or -reindex-chainstate to recover' showed up in the debug.log, perhaps my date/time not set correctly?  Anyway, I deleted debug.log and rebooted.  Then I got an error about debug.log not found.  But after doing a few things (or waiting?) not sure, the debug.log was created again - probably when the bitcoind.service was auto-restarted.  Then everything progressed as normal.

https://github.com/rootzoll/raspiblitz/blob/aaed704014a077da66bd4c1f8034618af9fa70a2/home.admin/00mainMenu.sh#L170

>         reindex=$(sudo cat /mnt/hdd/${network}/debug.log | grep -c 'Please restart with -reindex or -reindex-chainstate to recover')

fluidvoice, 2019-05-01:
Actually everything did not work right away after that. Then when running mainMenu I go:

> ****************************************
> * The LND service is not running.
> ****************************************
> If you just started some config/setup, this might be OK.
> 
> 2 RESTARTS DETECTED - LND might be in a error loop
> 
> -> Use following command to debug: /home/admin/XXdebugLogs.sh
> -> To force Main Menu run: /home/admin/00mainMenu.sh
> -> To try restart: sudo shutdown -r now
> 
so I did `sudo reboot now`

fluidvoice, 2019-05-01:
After that I ctrl-c out of that status screen and saw this:
[image]

fluidvoice, 2019-05-01:
`/home/admin/XXdebugLogs.sh | nc termbin.com 9999`
https://termbin.com/485n

fluidvoice, 2019-05-01:
OK, so then tried again from scratch:
`sudo reboot now`
waited for the node to come up, then SSH in and let it go to the menu:
[image]

rootzoll, 2019-05-01:
I added the deletion of bitcoin & lnd log files on every boot. I thinkthats OK now, because the log errors are still in the systemd journals and seen in XXdebugLogs.sh in the case we have to debug a error case where the system starts to reboot. The error in journal logs are not used to detect the error loops.

rootzoll, 2019-05-01:
That the "chain not synced" in the status screen is a non critical timing issue - happens sometime.

rootzoll, 2019-05-01:
I incresed the restartCount to 4 for LND before it will detect a loop error - so just in case a error goes away with an additional unlocking

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll