raspiblitz/raspiblitz #563: detailed wallet balance error

fdvanoye, 2019-05-03:
Hi All,

After a fresh new build I am having problems running the balance command from the menu and get the following error. How is it that after a fresh install the error occurs. Used the same image as previous installs, raspiblitz-v1.1-2019-03-18.img.gz. Very strange because I never had the error after a new install. Zap IOS wallet connects ok and shows correct balance.

**./00mainMenu.sh: line 419: lnbalance: command not found**

  RaspiBlitz v1.1  Tibxfool
               bitcoin Fullnode + Lightning Network
        ,/     -------------------------------------------
      ,'/      load average: 1.14, 1.16, 1.24, CPU 59°C
    ,' /       Free Mem 350M / 927M  Free HDD 594G (64%)
  ,'  /_____,  ssh admin@192.168.178.11 ▼3.0GiB ▲1.8GiB
 .'____    ,'
      /  ,'    bitcoin v0.17.0.1 mainnet Sync OK (100.00%)
     / ,'      Public 83.84.179.176:8333 38 connections
    /,'
   /'          LND 0.5.2-beta wallet 96927 sat
               0/1 Channels 353812 sat 7 peers

Thanks for your help

fluidvoice, 2019-05-05:
This is a small error in the v1.2 scripts.  To fix it try this from a SSH command line:

`wget https://stadicus.github.io/RaspiBolt/resources/lnbalance && wget https://stadicus.github.io/RaspiBolt/resources/lnchannels`
`chmod +x lnbalance && chmod +x lnchannels`
`sudo cp lnchannels /usr/local/bin && sudo cp lnbalance /usr/local/bin`

fluidvoice, 2019-05-05:
oh, sorry @fdvanoye I thought you were using v1.2

Stadicus, 2019-05-05:
I was not aware that other scripts depend on my repository, otherwise I might have been more careful with renaming stuff. Could be a good idea to port these two scripts directly into the RaspiBlitz repo, as they are not mine but by @robclark56 anyway.

geco91, 2019-05-08:
Recommend this to be fixed in v1.3 - adding scripts to RaspiBlitz repo makes sense.

rootzoll, 2019-05-13:
Listed to be fixed in v1.3 release.

fdvanoye, 2019-05-24:
Was away for a few weeks abroad, sorry for the late reply. Ran the commands as suggested by fluidvoice. Lightning filtering blockchain after a few weeks rest. Hope to get 1.3 installed soon, thx for your help!

fdvanoye, 2019-05-25:
Morning, the scripts fixed the problem :-). Will go for the 1.2 upgrade now. Thanks!

fdvanoye, 2019-05-27:
FYI, installed 1.2 and had the same wallet balance error again. Ran the suggested fluidvoice commands and got it working again.

closed 2019-05-27 by fdvanoye