raspiblitz/raspiblitz #564: LND service not running (missing rpcuser/password)

HonmaSokyu, 2019-05-03:
I just updated my raspi to 1.2 version and I cant connect anymore?

Debug Log:
https://termbin.com/wuqn

Main screen:
https://imgur.com/0J9VVRo

fluidvoice, 2019-05-03:
might be a networking problem.  Are you using ethernet or wifi? 
post the URL of running this:
`sudo cat /mnt/hdd/bitcoin/bitcoin.conf && sudo cat /mnt/hdd/bitcoin/debug.log | nc termbin.com 9999`
also note that you might have a power supply problem as 1.2 volts is below adequate level...

> *** HARDWARE TEST RESULTS ***
> UndervoltageReports in Logs: 0
> powerFAIL=0
> powerWARN=1
> powerMIN='1200000 microVolt'
> tempFAIL=0
> tempWARN=6
> tempMAX='6770 centiGrad'
> IMPORTANT: There are some hardware issues with your setup.
> 'Run Hardwaretest' in main menu or: **sudo /home/admin/05hardwareTest.sh**
>

HonmaSokyu, 2019-05-03:
Thanks for the fast reply :)

Im using Ethernet, like before with 1.1 version.

URL: https://termbin.com/npyu

I bought the power supply for EU of Amazon from the list on raspiblitz. Which one to buy then?

fluidvoice, 2019-05-03:
> Thanks for the fast reply :)
> 
> Im using Ethernet, like before with 1.1 version.
> 
> URL: https://termbin.com/npyu
> 
> I bought the power supply for EU of Amazon from the list on raspiblitz. Which one to buy then?

Lots of network socket errors:
> 2019-05-03T17:48:40Z socket send error Broken pipe (32) 

and failure to reach TOR hosts...
> 2019-05-03T17:47:44Z Socks5() connect to e6g4kddog7nac23w.onion:8333 failed: host unreachable
> 2019-05-03T17:47:50Z Socks5() connect to ndndword5lpb7eex.onion:8333 failed: host unreachable

fluidvoice, 2019-05-03:
I want to see what's in your config files:
`sudo cat /mnt/hdd/bitcoin/bitcoin.conf`
`sudo cat /mnt/hdd/lnd/lnd.conf`

HonmaSokyu, 2019-05-03:
Bitcoin.conf:
rpcport=8332
rpcallowip=127.0.0.1
rpcbind=127.0.0.1:8332
onlynet=onion
addnode=fno4aakpl6sg6y47.onion
addnode=toguvy5upyuctudx.onion
addnode=ndndword5lpb7eex.onion
addnode=6m2iqgnqjxh7ulyk.onion
addnode=5tuxetn7tar3q5kp.onion

LND.conf:
# lnd configuration

[Application Options]
debuglevel=info
maxpendingchannels=5
alias=UlaLa
color=#68F442
nat=false

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# REST open to all connections on Port 8080
restlisten=0.0.0.0:8080
# Domain, could use https://freedns.afraid.org
tlsextradomain=tty.sdf.org
tlsextraip=0.0.0.0

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

[Bitcoind]
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=
bitcoind.rpchost=127.0.0.1
bitcoind.zmqpubrawblock=tcp://*:28332
bitcoind.zmqpubrawtx=tcp://*:28333

[autopilot]
autopilot.active=0
autopilot.maxchannels=5
autopilot.allocation=0.6

fluidvoice, 2019-05-03:
That's the problem. Your **bitcoin.conf** file is missing a bunch of entries.  Now the question is how/why is this?  I didn't have this problem when I upgraded to v1.2

HonmaSokyu, 2019-05-03:
Dont know. I didnt change anything, dont even know how to change something in those files?

So we just have to change some lines? Please guide me ;) Will donate some sats :D

fluidvoice, 2019-05-03:
I was trying to see if there is a Raspiblitz log that might show this file getting over-written or something. 
When you installed, the menu system prompted you to re-enter some passwords right?

HonmaSokyu, 2019-05-03:
Right...

fluidvoice, 2019-05-03:
looks possible it was overwritten here. Need to verify with @rootzoll 
https://github.com/rootzoll/raspiblitz/blob/480a67729081cdd81357f508c0174b9ce05fa38e/home.admin/config.scripts/internet.tor.sh#L191

fluidvoice, 2019-05-03:
please prepend the title of this issue with "[v1.2]" to bring more attention to it.

HonmaSokyu, 2019-05-03:
WIth 1.1 i tried DNS and it didnt work. So then i tried with TOR and worked just fine from the start. I connected to peers and opened channels and funded them with RTL web and ZAP mobile.

Then today i upgraded to 1.2 and from start it worked but then i tried to "Forwarding Ports by autossh Tunnel" so i could connect outside my network and I failed :)

Allready tried flashing SD card again but no luck :)

Try again from scratch? or? :)

fluidvoice, 2019-05-03:
> i upgraded to 1.2 and from start it worked
 
Interesting. I had similar... v1.2 worked at first (without TOR) then failed after I added TOR and poked around with some other things (that I don't remember), but mine failed in a different way than you with different error.

fluidvoice, 2019-05-03:
try resetting your bitcoin.conf file to default:
`sudo cp /home/admin/assets/bitcoin.conf /mnt/hdd/bitcoin/bitcoin.conf`
then edit **/mnt/hdd/bitcoin/bitcoin.conf** `sudo nano /mnt/hdd/bitcoin/bitcoin.conf`
and change "rpcpassword=" set to your passwordB
then restart:
`sudo reboot now`

fluidvoice, 2019-05-03:
> Then today i upgraded to 1.2 and from start it worked but then i tried to "Forwarding Ports by autossh Tunnel" so i could connect outside my network and I failed :)

so what exactly did you do before it started failing, you ran this?... **`/home/admin/config.scripts/internet.sshtunnel.py`**

HonmaSokyu, 2019-05-03:
Thanks! That helped. Getting back online :)
Send me an invoice for 100k sat ill pay it :)

I ran this " /home/admin/config.scripts/internet.sshtunnel.py on test@raspiblitz.com "10009<20009" " and enabled DNS.

fluidvoice, 2019-05-03:
I have a tip link on my profile :)

fluidvoice, 2019-05-03:
@rootzoll is there a potential problem here?

> I ran this " /home/admin/config.scripts/internet.sshtunnel.py on [test@raspiblitz.com](mailto:test@raspiblitz.com) "10009<20009" " and enabled DNS.

HonmaSokyu, 2019-05-03:
Just for clarification "test@raspiblitz.com" was replaced by "tty.sdf.org".

StyriaTrading, 2019-05-11:
Have the same issue, appears on my Pi with v1.2 when I deactivated TOR in the advanced settings - maybe this helps in finding the bug.

fluidvoice, 2019-05-12:
> Have the same issue, appears on my Pi with v1.2 when I deactivated TOR in the advanced settings - maybe this helps in finding the bug.

When you deactivated TOR, what exactly happened and you saw?

rootzoll, 2019-05-13:
Since v1.2 it should make sure on every restart, that the RPC passwords are in sync between butcoind/lnd config - but not sure whats happening if they are completly missing. I will put this on the test list for 1.3 release.

TODO: Test switching between TOR/non-TOR ... see if bitcoind/lnd conf stays stable

rootzoll, 2019-07-01:
OK with v1.3 the missing of RPC username/password on the switch off of TOR should be fixed.

closed 2019-07-01 by rootzoll