raspiblitz/raspiblitz #565: LND service not running

vamp111, 2019-05-03:
after upgrading to v 1.2 and fully synced LN blckchain I get this error on the screen of the Raspi, as well as when i SSH into it
[image]
[image]

earlier I got two errors one for low voltage and one for high temperature, not sure if this matters.

I managed to get this information on the errors:
19 RESTARTS DETECTED - LND might be in a error loop
May 03 23:22:26 FreeTradeBulgaria lnd[24659]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/channeldb/graph.go:778 +0xe8 fp=0x306fc5c sp=0x306fc08 pc=0x4fe2f8
May 03 23:22:26 FreeTradeBulgaria lnd[24659]: github.com/lightningnetwork/lnd/routing.(*ChannelRouter).syncGraphWithChain(0x34021e0, 0x5f1e000, 0x99c7)
May 03 23:22:26 FreeTradeBulgaria lnd[24659]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/routing/router.go:607 +0x69c fp=0x306fcf0 sp=0x306fc5c pc=0x61f874
May 03 23:22:26 FreeTradeBulgaria lnd[24659]: github.com/lightningnetwork/lnd/routing.(*ChannelRouter).Start(0x34021e0, 0x0, 0x0)
May 03 23:22:26 FreeTradeBulgaria lnd[24659]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/routing/router.go:454 +0x300 fp=0x306fd48 sp=0x306fcf0 pc=0x61ee34
May 03 23:22:26 FreeTradeBulgaria lnd[24659]: main.(*server).Start.func1()
May 03 23:22:26 FreeTradeBulgaria lnd[24659]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/server.go:1141 +0x408 fp=0x306fdc0 sp=0x306fd48 pc=0x904cd4
May 03 23:22:26 FreeTradeBulgaria lnd[24659]: sync.(*Once).Do(0x31f4008, 0x30e5dec)
May 03 23:22:26 FreeTradeBulgaria lnd[24659]:         /usr/local/go/src/sync/once.go:44 +0xb8 fp=0x306fdd8 sp=0x306fdc0 pc=0x74328
May 03 23:22:26 FreeTradeBulgaria lnd[24659]: main.(*server).Start(0x31f4000, 0xb4d9ea, 0x2e)
May 03 23:22:26 FreeTradeBulgaria lnd[24659]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/server.go:1075 +0x64 fp=0x306fdf8 sp=0x306fdd8 pc=0x8e7d44

here is the Log file: /home/admin/XXdebugLogs.sh | nc termbin.com 9999

fluidvoice, 2019-05-03:
post the link from running these:
`sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`
`sudo cat /mnt/hdd/bitcoin/debug.log | nc termbin.com 9999`

vamp111, 2019-05-03:
https://termbin.com/8sqb

https://termbin.com/e0x0

fluidvoice, 2019-05-04:
grep through the LND log for "[ERR]" and you'll see lots of the same error message repeated and I don't know why and what it means:
> [ERR] LTND: Unable to lookup witness: no witnesses

could it just be network connection issue?

vamp111, 2019-05-04:
well i see some connection related errors:

256 RESTARTS DETECTED - LND might be in a error loop
1556943302 ERROR: LND RPC is not responding. LND may have problems starting up. Check logs, config files and systemd service. **Org-Error: [lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: **Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"**
************************************
* The LND service is not running.**
****************************************
If you just started some config/setup, this might be OK.

256 RESTARTS DETECTED - LND might be in a error loop
1556943302 ERROR: LND RPC is not responding. LND may have problems starting up. Check logs, config files and systemd service. Org-Error: [lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
May 04 05:32:42 FreeTradeBulgaria lnd[18597]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/cursor.go:262 +0x10c fp=0x31e76e8 sp=0x31e7698 pc=0x22b87c
May 04 05:32:42 FreeTradeBulgaria lnd[18597]: github.com/coreos/bbolt.(*Cursor).seek(0x31e7768, 0x31e77a4, 0x29, 0x29, 0x4a1a280, 0x1, 0x1, 0xff, 0x32f8300, 0x15, ...)
May 04 05:32:42 FreeTradeBulgaria lnd[18597]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/cursor.go:159 +0x70 fp=0x31e772c sp=0x31e76e8 pc=0x22b0c8
May 04 05:32:42 FreeTradeBulgaria lnd[18597]: github.com/coreos/bbolt.(*Bucket).Get(0x4ee20c0, 0x31e77a4, 0x29, 0x29, 0x0, 0x0, 0x0)
May 04 05:32:42 FreeTradeBulgaria lnd[18597]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/bucket.go:260 +0x98 fp=0x31e7778 sp=0x31e772c pc=0x228674
May 04 05:32:42 FreeTradeBulgaria lnd[18597]: github.com/lightningnetwork/lnd/channeldb.fetchChanEdgePolicy(0x4ee20c0, 0x61c8b8bc, 0x8, 0x8, 0x5f7af46e, 0x21, 0x1d9, 0x4ee2260, 0x325b9d0, 0x0, ...)
May 04 05:32:42 FreeTradeBulgaria lnd[18597]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/channeldb/graph.go:3642 +0xa0 fp=0x31e77dc sp=0x31e7778 pc=0x506874
May 04 05:32:42 FreeTradeBulgaria lnd[18597]: github.com/lightningnetwork/lnd/channeldb.fetchChanEdgePolicies(0x4ee2160, 0x4ee20c0, 0x4ee2260, 0x61c8b8bc, 0x8, 0x8, 0x0, 0x46a2a6c1, 0x4ff763ae, 0x65831e93, ...)
May 04 05:32:42 FreeTradeBulgaria lnd[18597]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/channeldb/graph.go:3696 +0x138 fp=0x31e7818 sp=0x31e77dc pc=0x506bbc
May 04 05:32:42 FreeTradeBulgaria lnd[18597]: github.com/lightningnetwork/lnd/channeldb.delChannelByEdge(0x4ee20c0, 0x4ee2160, 0x4ee2200, 0x4ee2300, 0x4ee2260, 0x3811400, 0xa8ce200, 0x72b3f1b6, 0x46a2a6c1)
May 04 05:32:42 FreeTradeBulgaria lnd[18597]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/channeldb/graph.go:1773 +0x178 fp=0x31e793c sp=0x31e7818 pc=0x50022c

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

vamp111, 2019-05-04:
here the latest debuglog file
https://termbin.com/siy8

fluidvoice, 2019-05-04:
send the full LND logs:
`sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`
`sudo cat /var/log/syslog | nc termbin.com 9999`

vamp111, 2019-05-04:
https://termbin.com/hvsf
https://termbin.com/oy81q

fluidvoice, 2019-05-04:
This seems similar to the problem you reported with v1.1 where LND was crashing.
Was your system running file/stable on v1.1 before you installed v1.2 or was it still having problems?

fluidvoice, 2019-05-04:
It looks like a possible HDD error on **/dev/sda** which I assume is your blockchain HDD

> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.040200] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.040230] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x5 [current] 
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.040238] sd 0:0:0:0: [sda] tag#0 ASC=0x24 ASCQ=0x0 
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.040248] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 0b 90 10 30 00 00 08 00
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.040256] print_req_error: critical target error, dev sda, sector 193990704
> May  3 23:00:45 FreeTradeBulgaria lnd[29769]: unexpected fault address 0x61eab008
> May  3 23:00:45 FreeTradeBulgaria lnd[29769]: fatal error: fault
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.429255] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.429271] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x5 [current] 
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.429282] sd 0:0:0:0: [sda] tag#0 ASC=0x24 ASCQ=0x0 
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.429296] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 0b 90 10 30 00 00 08 00
> May  3 23:00:45 FreeTradeBulgaria kernel: [ 5882.429307] print_req_error: critical target error, dev sda, sector 193990704
> May  3 23:00:45 FreeTradeBulgaria lnd[29769]: [signal SIGBUS: bus error code=0x2 addr=0x61eab008 pc=0x22b7b4]
> May  3 23:00:45 FreeTradeBulgaria lnd[29769]: goroutine 1 [running]:
> May  3 23:00:45 FreeTradeBulgaria lnd[29769]: runtime.throw(0xb19bcd, 0x5)
> May  3 23:00:45 FreeTradeBulgaria lnd[29769]: #011/usr/local/go/src/runtime/panic.go:617 +0x5c fp=0x224d4e4 sp=0x224d4d0 pc=0x3ef6c
> May  3 23:00:45 FreeTradeBulgaria lnd[29769]: runtime.**sigpanic**()

fluidvoice, 2019-05-04:
Are you running an external USB HDD plugged into just the Pi USB port?
You might want use a separate power supply for the HDD, as it's more robust, reliable that way.
You can use a powered USB hub or a USB Y-cable with a around 1Amp USB power supply (wall charger)

fluidvoice, 2019-05-04:
Also run the Harware test to see if it points to a possible source of your disk/data corruption:
`sudo /home/admin/05hardwareTest.sh`

fluidvoice, 2019-05-04:
@vamp111 I'm curious why you opened a LND issue when this is a hardware issue?

vamp111, 2019-05-05:
yes, hdd is powered from the Raspi with USB all from the german amazon shopping list 

i runned the hardware test, got warning on temperature and power
this massage appeares
Your power supply was FAILING the stress test (1200000 microVolt).         │
                     │ Most reports of data loss are caused by weak power supplies.               │
                     │ Also a lot of RaspiBlitz setups fail because of weak power supplies.       │
                     │ To SHUTDOWN and upgrade the Power Supply is HIGHLY RECOMMENDED.

Your RaspiBlitz is getting MUCH TOO HOT (7200 centiGrad).                  │
                     │ The system is getting very slow when hot - thats not a NO GO but bad.      │
                     │ An upgrade of the Heat Management is HIGHLY RECOMMENDED.

vamp111, 2019-05-05:
> This seems similar to the problem you reported with v1.1 where LND was crashing.
> Was your system running file/stable on v1.1 before you installed v1.2 or was it still having problems?

ver. 1.0 was running fine, all problems began after upgrading to 1.1. I reported this to LND https://github.com/lightningnetwork/lnd/issues/2874 which supposed to be solved in ver 0.6

vamp111, 2019-05-05:
ver. 1.2 not even starting the LND service, I dont know if it matters, the Raspi was on during this wait period of 29 days, blockchain was syncing and sometimes LND started by itself and even showed some channels (very rarely)

fluidvoice, 2019-05-05:
> FAILING the stress test (1200000 microVolt).

If you are seeing only 1.2 volts in the test this is too low and can be causing HD I/O errors and/or file corruption.  I recommend changing your power supply, or powering your HDD with separate PS.

fluidvoice, 2019-05-05:
> sometimes LND started by itself and even showed some channels (very rarely)

Working sometimes and not others - could be intermittent power problem or network problem.  Just a guess, but I'm still betting this all is caused by the PS.

nsollazzo, 2019-05-05:
i have the same issue, but my hdd is plugged into a Hub with a 5V/2A adapter.
here's my log: https://termbin.com/4060

fluidvoice, 2019-05-05:
> i have the same issue, but my hdd is plugged into a Hub with a 5V/2A adapter.
> here's my log: https://termbin.com/4060

1. Open your own separate issue here (click "..." and select "reference in new issue")
2. "UndervoltageReports in Logs: **682**" ... regardless of your HDD power, your Pi PS is probably insufficient.  I'd replace it.

HansiRumker, 2019-05-06:
Hello together, same Problem,
2 RESTARTS DETECTED - LND might be in a error loop
May 06 15:01:14 BitcoinServerTHREE lnd[2102]: unable to read channel for chain_h                                                                                                                                                             ash=6fe28c0ab6f1b372c1a6a246ae63f74f931e8365e15a089c68d6190000000000, node_key=0                                                                                                                                                             319fec13dcb6df5b6a0cee49e85b4f6c48e5387d62d67858d6d46615b6e683f81: unexpected EO   
HHD ist new, i put in the Backupfile, but the same on an new Raspi, HHD new, 3A, Harwaretest OK.
Can you help me ?

fluidvoice, 2019-05-06:
> Hello together, same Problem,
> 2 RESTARTS DETECTED - LND might be in a error loop
> May 06 15:01:14 BitcoinServerTHREE lnd[2102]: unable to read channel for chain_h [...]

Open your own issue and give output from running...
-> Use following command to debug: `/home/admin/XXdebugLogs.sh`

HansiRumker, 2019-05-06:
Thanks here ist it.
****************************************
* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

3 RESTARTS DETECTED - LND might be in a error loop

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

admin@BitcoinServerTHREE:~ $ /home/admin/XXdebugLogs.sh

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 15:43:07 up 54 min,  2 users,  load average: 0.67, 0.41, 0.52

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-05-06 14:51:13 BST; 51min ago
  Process: 2038 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 2032 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 2097 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─2097 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoi…id

May 06 14:51:12 BitcoinServerTHREE systemd[1]: Starting Bitcoin daemon...
May 06 14:51:13 BitcoinServerTHREE systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Mon 2019-05-06 15:43:07 BST. --
May 06 14:51:12 BitcoinServerTHREE systemd[1]: Starting Bitcoin daemon...
May 06 14:51:13 BitcoinServerTHREE systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-05-06T14:07:55Z socket recv error Connection reset by peer (104)
2019-05-06T14:11:42Z socket recv error Connection reset by peer (104)
2019-05-06T14:12:55Z socket recv error Connection reset by peer (104)
2019-05-06T14:14:44Z New outbound peer connected: version: 70015, blocks=574855, peer=137
2019-05-06T14:16:29Z socket recv error Connection reset by peer (104)
2019-05-06T14:17:55Z socket send error Connection reset by peer (104)
2019-05-06T14:18:14Z socket recv error Connection reset by peer (104)
2019-05-06T14:21:35Z socket send error Broken pipe (32)
2019-05-06T14:21:35Z socket recv error Connection reset by peer (104)
2019-05-06T14:22:10Z UpdateTip: new best=00000000000000000021037df6661e3f7ee7d104b927d4b0a18b6896c800dbdf height=574856 version=0x20000000 log2_work=90.611491 tx=410373078 date='2019-05-06T14:21:58Z' progress=1.000000 cache=7.3MiB(64647txo) warning='35 of last 100 blocks have unexpected version'
2019-05-06T14:23:05Z socket recv error Connection reset by peer (104)
2019-05-06T14:26:28Z socket recv error Connection reset by peer (104)
2019-05-06T14:26:29Z socket recv error Connection reset by peer (104)
2019-05-06T14:27:55Z socket send error Connection reset by peer (104)
2019-05-06T14:28:49Z socket recv error Connection reset by peer (104)
2019-05-06T14:32:55Z socket recv error Connection reset by peer (104)
2019-05-06T14:35:40Z socket recv error Connection timed out (110)
2019-05-06T14:37:56Z socket recv error Connection reset by peer (104)
2019-05-06T14:42:42Z socket recv error Connection reset by peer (104)
2019-05-06T14:42:56Z socket recv error Connection reset by peer (104)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-05-06 15:42:00 BST; 1min 7s ago
  Process: 13597 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 13600 (lnd)
   CGroup: /system.slice/lnd.service
[... 63 lines cut ...]
tempWARN=0
tempMAX='4990 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.100.20'
tempCelsius='39.1'
uptime=3262
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=574856
initialSync=0
syncProgress=100.00
startcountLightning=4
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

nsollazzo, 2019-05-06:
1. to open your issue use the three dots on the up right corner and select "reference as new issue"
2. i think the problem is here:

> unable to create server: unable to read channel for chain_hash=6fe28c0ab6f1b372c1a6a246ae63f74f931e8365e15a089c68d6190000000000, node_key=0319fec13dcb6df5b6a0cee49e85b4f6c48e5387d62d67858d6d46615b6e683f81: unexpected EOF

which is a different error from the one that we have

nsollazzo, 2019-05-06:
i see something very strange here to:

> 2019-05-06T14:22:10Z UpdateTip: new best=00000000000000000021037df6661e3f7ee7d104b927d4b0a18b6896c800dbdf height=574856 version=0x20000000 log2_work=90.611491 tx=410373078 date='2019-05-06T14:21:58Z' progress=1.000000 cache=7.3MiB(64647txo) warning='35 of last 100 blocks have unexpected version'
2019-05-06T14:23:05Z socket recv error Connection reset by peer (104)

HansiRumker, 2019-05-06:
Thanks, but what is the to do ?

fluidvoice, 2019-05-06:
> Thanks, but what is the to do ?

open a new issue.

vamp111, 2019-05-07:
> > FAILING the stress test (1200000 microVolt).
> 
> If you are seeing only 1.2 volts in the test this is too low and can be causing HD I/O errors and/or file corruption. I recommend changing your power supply, or powering your HDD with separate PS.
> This error **"critical target error, dev sda, sector 193990704"** is almost certainly a hardware (HDD) error and is not related to LND.
> 
> As it turns out, unfortunately, some of the past items listed in the shopping lists were not very good. As reports from users come in (and now reported via the new HardwareTest script) the shopping lists are being changed to contain better PS's as we learn which ones are better than others.

this is the one from the shopping list, care to show me a better alternative from Amazon.de ?

vamp111, 2019-05-07:
> @vamp111 did you see the errors in [#565 (comment)](https://github.com/rootzoll/raspiblitz/issues/565#issuecomment-489339538) ? And you still think it's an LND issue?

Roasbeef replied like this to my issue opened in LND : https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-489728865
https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-489729021

HansiRumker, 2019-05-07:
I have three RaspiBlitz, one is a for falling back, the first make this Problem , so I play the Backup for Nummer one in the number 3 an become the some Problem. But stresstest is ok. The second is the old version 3 and works fine. So I thing that’s no Hardewareproblem. 
Linux BitcoinServerONE 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue May  7 01:25:29 2019 from 192.168.100.52
Starting the main menu ...
checking bitcoind - please wait ..
can take longer if device was off or first time

****************************************
* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

496 RESTARTS DETECTED - LND might be in a error loop

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

admin@BitcoinServerONE:~ $ /home/admin/00mainMenu.sh
Starting the main menu ...
***********************************
* RaspiBlitz Commandline
* Here be dragons .. have fun :)
***********************************
LND commandline options: lncli -h
Back to main menu use command: raspiblitz

admin@BitcoinServerONE:~ $ /home/admin/00mainMenu.sh
Starting the main/home/admin/00mainMenu.sh: line 109: lnbalance: command not found
Press ENTER to return to main menu.

Starting the main menu ...

Starting the main menu ...
checking bitcoind - please wait ..
can take longer if device was off or first time

****************************************
* The LND service is running with error.
****************************************
If you just started some config/setup, this might be OK.

498 RESTARTS DETECTED - LND might be in a error loop
May 07 07:40:50 BitcoinServerONE lnd[9737]: unable to read channel for chain_hash=6fe28c0ab6f1b372c1a6a246ae63f74f931e8365e15a089c68d6190000000000, node_key=0319fec13dcb6df5b6a0cee49e85b4f6c48e5387d62d67858d6d46615b6e683f81: unexpected EOF

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

(exit after too much restarts/unlocks - restart to try again)
admin@BitcoinServerONE:~ $

fluidvoice, 2019-05-07:
> > @vamp111 did you see the errors in [#565 (comment)](https://github.com/rootzoll/raspiblitz/issues/565#issuecomment-489339538) ? And you still think it's an LND issue?
> 
> Roasbeef replied like this to my issue opened in LND : [lightningnetwork/lnd#3042 (comment)](https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-489728865)
> [lightningnetwork/lnd#3042 (comment)](https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-489729021)

Reinstall Raspiblitz and restore your wallet as per the documentation: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

fluidvoice, 2019-05-07:
@HansiRumker do you understand English?  I've asked you three times to open your own issue.  Is it difficult for you?

HansiRumker, 2019-05-07:
sorry my english ist not so good.

fluidvoice, 2019-05-07:
> this is the one from the shopping list, care to show me a better alternative from Amazon.de ?

> sorry my english ist not so good.

post the links from running:
`sudo cat /mnt/hdd/bitcoin/debug.log`
`sudo cat /var/log/syslog`

vamp111, 2019-05-08:
sudo cat /mnt/hdd/bitcoin/debug.log
2019-05-04T15:11:47Z

2019-05-04T15:11:47Z Bitcoin Core version v0.17.1 (release build)
2019-05-04T15:11:47Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-05-04T15:11:47Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-05-04T15:11:47Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-05-04T15:11:47Z Using the 'standard' SHA256 implementation
2019-05-04T15:11:47Z Default data directory /home/bitcoin/.bitcoin
2019-05-04T15:11:47Z Using data directory /mnt/hdd/bitcoin
2019-05-04T15:11:47Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-05-04T15:11:47Z Using at most 40 automatic connections (1024 file descriptors available)
2019-05-04T15:11:47Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-05-04T15:11:47Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-05-04T15:11:47Z Using 4 threads for script verification
2019-05-04T15:11:47Z scheduler thread start
2019-05-04T15:11:47Z HTTP: creating work queue of depth 16
2019-05-04T15:11:47Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2019-05-04T15:11:47Z HTTP: starting 4 worker threads
2019-05-04T15:11:47Z Cache configuration:
2019-05-04T15:11:47Z * Using 2.0MiB for block index database
2019-05-04T15:11:47Z * Using 12.2MiB for transaction index database
2019-05-04T15:11:47Z * Using 8.0MiB for chain state database
2019-05-04T15:11:47Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-05-04T15:11:47Z init message: Loading block index...
2019-05-04T15:11:47Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2019-05-04T15:11:48Z Opened LevelDB successfully
2019-05-04T15:11:48Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2019-05-04T15:12:10Z LoadBlockIndexDB: last block file = 1624
2019-05-04T15:12:10Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=49, size=57416709, heights=574544...574592, time=2019-05-04...2019-05-04)
2019-05-04T15:12:10Z Checking all blk files are present...
2019-05-04T15:12:10Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2019-05-04T15:12:15Z Opened LevelDB successfully
2019-05-04T15:12:15Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: eadc35e843c5834c
2019-05-04T15:12:16Z Loaded best chain: hashBestChain=0000000000000000000631c5ccb0acd2500d0fb48c54d328663cc2f512ee9b2a height=574592 date=2019-05-04T14:27:48Z progress=0.999984
2019-05-04T15:12:16Z init message: Rewinding blocks...
2019-05-04T15:12:25Z init message: Verifying blocks...
2019-05-04T15:12:25Z Verifying last 6 blocks at level 3
2019-05-04T15:12:25Z [0%]...[16%]...[33%]...[50%]...[66%]...[83%]...[99%]...[DONE].
2019-05-04T15:13:15Z No coin database inconsistencies in last 6 blocks (12340 transactions)
2019-05-04T15:13:15Z  block index           88025ms
2019-05-04T15:13:16Z Opening LevelDB in /mnt/hdd/bitcoin/indexes/txindex
2019-05-04T15:13:17Z Opened LevelDB successfully
2019-05-04T15:13:17Z Using obfuscation key for /mnt/hdd/bitcoin/indexes/txindex: 0000000000000000
2019-05-04T15:13:17Z Wallet disabled!
2019-05-04T15:13:17Z mapBlockIndex.size() = 574617
2019-05-04T15:13:17Z nBestHeight = 574592
2019-05-04T15:13:17Z txindex thread start
2019-05-04T15:13:17Z Syncing txindex with block chain from height 574536
2019-05-04T15:13:17Z torcontrol thread start
2019-05-04T15:13:17Z Bound to [::]:8333
2019-05-04T15:13:17Z Bound to 0.0.0.0:8333
2019-05-04T15:13:17Z init message: Loading P2P addresses...
2019-05-04T15:13:17Z Leaving InitialBlockDownload (latching to false)
2019-05-04T15:13:19Z Loaded 64143 addresses from peers.dat  1581ms
2019-05-04T15:13:19Z init message: Loading banlist...
2019-05-04T15:13:19Z init message: Starting network threads...
2019-05-04T15:13:19Z net thread start
2019-05-04T15:13:19Z dnsseed thread start
2019-05-04T15:13:19Z addcon thread start
2019-05-04T15:13:19Z msghand thread start
2019-05-04T15:13:19Z opencon thread start
2019-05-04T15:13:19Z init message: Done loading
2019-05-04T15:13:20Z New outbound peer connected: version: 70015, blocks=494782, peer=0
2019-05-04T15:13:21Z New outbound peer connected: version: 70015, blocks=574593, peer=1
2019-05-04T15:13:30Z P2P peers available. Skipped DNS seeding.
2019-05-04T15:13:30Z dnsseed thread exit
2019-05-04T15:14:30Z UpdateTip: new best=00000000000000000007bf99cae68085dfadb52891e023b88abfbcbca69d48ce height=574593 version=0x20c00000 log2_work=90.605705 tx=409701156 date='2019-05-04T15:03:22Z' progress=0.999996 cache=1.5MiB(13016txo) warning='35 of last 100 blocks have unexpected version'
2019-05-04T15:14:30Z Syncing txindex with block chain from height 574554
2019-05-04T15:14:34Z New outbound peer connected: version: 70015, blocks=574593, peer=4
[... 664 lines cut ...]
2019-05-08T05:24:50Z UpdateTip: new best=0000000000000000001d21476f45bb212ccb3b4425b2765af068d96dc9306846 height=575079 version=0x20000000 log2_work=90.616379 tx=410949164 date='2019-05-08T05:24:28Z' progress=1.000000 cache=59.0MiB(519560txo) warning='34 of last 100 blocks have unexpected version'
2019-05-08T05:29:04Z Pre-allocating up to position 0x1000000 in rev01628.dat
[... cut ...]
2019-05-08T07:40:19Z UpdateTip: new best=0000000000000000001a24975559be12ad3744188334607ea7ddf40a75d9c501 height=575093 version=0x20000000 log2_work=90.616686 tx=410980242 date='2019-05-08T07:40:15Z' progress=1.000000 cache=65.4MiB(579488txo) warning='38 of last 100 blocks have unexpected version'
2019-05-08T07:42:20Z UpdateTip: new best=00000000000000000025a1711742d6b133d5e84ed9efe35279a9cf1c9ae7cf04 height=575094 version=0x3fffe000 log2_work=90.616708 tx=410982942 date='2019-05-08T07:41:46Z' progress=1.000000 cache=65.9MiB(583533txo) warning='39 of last 100 blocks have unexpected version'

vamp111, 2019-05-08:
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z
-bash: 2019-05-04T15:11:47Z: command not found
admin@FreeTradeBulgaria:~ $
admin@FreeTradeBulgaria:~ $
admin@FreeTradeBulgaria:~ $
admin@FreeTradeBulgaria:~ $
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Bitcoin Core version v0.17.1 (release build)
-bash: syntax error near unexpected token `('
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
-bash: 2019-05-04T15:11:47Z: command not found
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
-bash: 2019-05-04T15:11:47Z: command not found
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2-bash: 2019-05-04T15:11:47Z: command not found
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Using the 'standard' SHA256 implementation
-bash: 2019-05-04T15:11:47Z: command not found
201admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Default data directory /home/bitcoin/.bitcoin
-bash: 2019-05-04T15:11:47Z: command not found
2admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Using data directory /mnt/hdd/bitcoin
15:-bash: 2019-05-04T15:11:47Z: command not found
11admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
01-bash: 2019-05-04T15:11:47Z: command not found
9-0admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Using at most 40 automatic connections (1024 file descriptors availa)le
-bash: syntax error near unexpected token `('
20admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Using 16 MiB out of 32/2 requested for signature cache, able to store24288 elements
2-bash: 2019-05-04T15:11:47Z: command not found
019-admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Using 16 MiB out of 32/2 requested for script execution cache, ablestore 524288 elements
2-bash: 2019-05-04T15:11:47Z: command not found
019-admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Using 4 threads for script verification
20-bash: 2019-05-04T15:11:47Z: command not found
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z scheduler thread start
20-bash: 2019-05-04T15:11:47Z: command not found
19-admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z HTTP: creating work queue of depth 16
-bash: 2019-05-04T15:11:47Z: command not found
2019admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Config options rpcuser and rpcpassword will soon be deprecated. Loc-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
04-bash: 2019-05-04T15:11:47Z: command not found
T1admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z HTTP: starting 4 worker threads
-bash: 2019-05-04T15:11:47Z: command not found
2019admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Cache configuration:
01-bash: 2019-05-04T15:11:47Z: command not found
9-0admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z * Using 2.0MiB for block index database
2019-bash: 2019-05-04T15:11:47Z: command not found
-05-0admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z * Using 12.2MiB for transaction index database
019-05-04T15-bash: 2019-05-04T15:11:47Z: command not found
:11:47Z * Using 8.0MiBadmin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z * Using 8.0MiB for chain state database
2019-05-04T1-bash: 2019-05-04T15:11:47Z: command not found
5:11:47Z * Usadmin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiBmpool space)em
-bash: syntax error near unexpected token `('
9-05-04admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z init message: Loading block index...
19-05-04T15:11:47Z-bash: 2019-05-04T15:11:47Z: command not found
 Openinadmin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:47Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
201-bash: 2019-05-04T15:11:47Z: command not found
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:48Z Opened LevelDB successfully
9-05-04-bash: 2019-05-04T15:11:48Z: command not found
T15:11:48admin@FreeTradeBulgaria:~ $ 2019-05-04T15:11:48Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 00000000000000
19-0-bash: 2019-05-04T15:11:48Z: command not found
admin@FreeTradeBulgaria:~ $ 2019-05-04T15:12:10Z LoadBlockIndexDB: last block file = 1624
9-05-04T-bash: 2019-05-04T15:12:10Z: command not found
15:12:10Z admin@FreeTradeBulgaria:~ $ 2019-05-04T15:12:10Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks57416709, heights=574544...574592, time=2019-05-04...2019-05-04)
-bash: syntax error near unexpected token `('
19-05-04admin@FreeTradeBulgaria:~ $ 2019-05-04T15:12:10Z Checking all blk files are present...
10Z-bash: 2019-05-04T15:12:10Z: command not found
 Openadmin@FreeTradeBulgaria:~ $ 2019-05-04T15:12:10Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
-05-bash: 2019-05-04T15:12:10Z: command not found
-0admin@FreeTradeBulgaria:~ $ 2019-05-04T15:12:15Z Opened LevelDB successfully
2-bash: 2019-05-04T15:12:15Z: command not found
019admin@FreeTradeBulgaria:~ $ 2019-05-04T15:12:15Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: eadc35e843c58
2-bash: 2019-05-04T15:12:15Z: command not found
0admin@FreeTradeBulgaria:~ $ 2019-05-04T15:12:16Z Loaded best chain: hashBestChain=0000000000000000000631c5ccb0acd2500d0b48c54d328663cc2f512ee9b2a height=574592 date=2019-05-04T14:27:48Z progress=0.999984
[... cut ...]
 detected,admin@FreeTradeBulgaria:~ $ 2019-05-07T19:33:33Z Potential stale tip detected, will try using extra outbound ptip update: 1991 seconds ago)
-bash: syntax error near unexpected token `('
admin@FreeTradeBulgaria:~ $ 2019-05-07T19:33:36Z New outbound peer connected: version: 70015, blocks=575021, peer=171
Z Pre-allocating up to position 0x4000000 in-bash: 2019-05-07T19:33:36Z: command not found
admin@FreeTradeBulgaria:~ $ 2019-05-07T19:33:58Z Pre-allocating up to position 0x4000000 in blk01628.dat
07T19:3-bash: 2019-05-07T19:33:58

fluidvoice, 2019-05-08:
> sudo cat /mnt/hdd/bitcoin/debug.log
> sudo cat /var/log/syslog

@vamp111 I'm going to stop responding since you don't answer my questions.

fluidvoice, 2019-05-08:
> sudo cat /mnt/hdd/bitcoin/debug.log
> sudo cat /var/log/syslog

???

vamp111, 2019-05-08:
> > sudo cat /mnt/hdd/bitcoin/debug.log
> > sudo cat /var/log/syslog
> 
> ???

those are the logs you requested

https://github.com/rootzoll/raspiblitz/issues/565#issuecomment-490458267

vamp111, 2019-05-08:
@fluidvoice 

sudo cat /mnt/hdd/bitcoin/debug.log   file:  https://termbin.com/h3t0
sudo cat /var/log/syslog   file: https://termbin.com/ys20

fluidvoice, 2019-05-08:
> Reinstall Raspiblitz and restore your wallet as per the documentation: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

I was requesting those logs from the other guy posting here (who seem to be resisting creating his own issue). That's the problem it gets confusing.  Your issue is a corruption and you need to reinstall.

vamp111, 2019-05-08:
> I was requesting those logs from the other guy posting here (who seem to be resisting creating his own issue). That's the problem it gets confusing. Your issue is a corruption and you need to [...]

I'm sorry i got confused as well, you are right about the reinstall, but i lack the channel.backup file in the main directory, so im afraid it will result in loss of a little under 0.2 btc which are stuck in channels

fluidvoice, 2019-05-08:
> I'm sorry i got confused as well, you are right about the reinstall, but i lack the channel.backup file in the main directory, so im afraid it will result in loss of a little under 0.2 btc which are [...]

Were you not able to create it as I described in the LN issue: 
https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-490477341
I think you're right, but I don't know why. On my node also it's not created. **Try this to create it:**
`lncli unlock && lncli exportchanbackup --all --output_file /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup`

vamp111, 2019-05-08:
> > > > > > @vamp111 did you see the errors in [#565 (comment)](https://github.com/rootzoll/raspiblitz/issues/565#issuecomment-489339538) ? And you still think it's an LND issue?
> > > > > 
> > > > > 
> > > > > Roasbeef replied like this to my issue opened in LND : [lightningnetwork/lnd#3042 (comment)](https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-489728865)
> > > > > [lightningnetwork/lnd#3042 (comment)](https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-489729021)
> > > > 
> > > > 
> > > > Reinstall Raspiblitz and restore your wallet as per the documentation: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz
> > > 
> > > 
> > > I was requesting those logs from the other guy posting here (who seem to be resisting creating his own issue). That's the problem it gets confusing. Your issue is a corruption and you need to reinstall.
> > 
> > 
> > I'm sorry i got confused as well, you are right about the reinstall, but i lack the channel.backup file in the main directory, so im afraid it will result in loss of a little under 0.2 btc which are stuck in channels
> 
> Were you not able to create it as I described in the LN issue:
> [lightningnetwork/lnd#3042 (comment)](https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-490477341)
> 
> > I think you're right, but I don't know why. On my node also it's not created. **Try this to create it:**
> > `lncli unlock && lncli exportchanbackup --all --output_file /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup`

answered you in https://github.com/lightningnetwork/lnd/issues/3042#issuecomment-490584852

turbolift, 2019-05-10:
Same Problem here. Had to recover my funds with channel.backup and seed.

vamp111, 2019-05-10:
@turbolift but my LND 0.6 never generated a channel.backup file, maybe I shouldn't have upgraded it at all

turbolift, 2019-05-10:
> @turbolift but my LND 0.6 never generated a channel.backup file, maybe I shouldn't have upgraded it at all

That was the first thing i did upgrading to 0.6: activate channel.backup and automatic file upload to dropbox.

vamp111, 2019-05-10:
but my LND service never started after the update process.....

turbolift, 2019-05-10:
> but my LND service never started after the update process.....

Upgrading from raspiblitz 1.1 to 1.2 was no problem. I managed to configure backup. 1-2 days later I could not connect via shango. After some research if found this issue: [db corruption](https://github.com/lightningnetwork/lnd/issues/2545). This night i managed to recreate the wallet and applied the backup. After that I did a completly cachout. 

Hopefully you did not loose to many btc.

vamp111, 2019-05-10:
little less than 0.2.......

fluidvoice, 2019-05-10:
> little less than 0.2.......

Damn. That sucks.  Hope it can be recovered some how.
A while ago I created a backup of my HDD because I was tired of having to re-download and re-sync the blockchain.  This backup also has an old copy of LND data.  Do you have this? if so, maybe can recover with that.

vamp111, 2019-05-10:
I will mount my HDD on another computer and will manualy search, not sure since LND 0.6 never ever was able to start, since it generated many of the new files, I asume it didnt delted the old once...

fluidvoice, 2019-05-10:
> I will mount my HDD on another computer and will manualy search, not sure since LND 0.6 never ever was able to start, since it generated many of the new files, I asume it didnt delted the old once...

I don't think it matters.  I don't think channel.backup is created automatically.  When I installed v1.2 it was working at first then started failing later.  Yet my node also has no channel.backup.  I think you have to create it manually?

vamp111, 2019-05-10:
i tryed with this command: lncli exportchanbackup --all --output_file /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup

but since LND is not running, it doesnt work

fluidvoice, 2019-05-10:
> i tryed with this command: lncli exportchanbackup --all --output_file /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
> 
> but since LND is not running, it doesnt work

yes, I'm aware that it won't work if LND won't start.  I don't have any ideas for you, sorry.

turbolift, 2019-05-10:
For all those who wants to get the chance for rescue (from the docs):

For the v1.2 Release this Off-Site Backup options you need to manually editing the raspiblitz config: nano mnt/hdd/raspiblitz.conf (CTRL+o = save & CTRL+x = exit) So this is more for expert users at the moment.

vamp111, 2019-05-10:
@fluidvoice its ok, that's why its called #reckless

nsollazzo, 2019-05-13:
> Same Problem here. Had to recover my funds with channel.backup and seed.
> 
> 2019-05-08 20:19:04.291 [INF] CNCT: ChannelArbitrator(1a1fbe00bbd027c134132b323e00fd6ace7a8e1d9bf7df309df74a61de36f6c4:0): starting state=StateDefault
> 2019-05-08 20:19:04.311 [INF] DISC: Authenticated Gossiper is starting
> 2019-05-08 20:19:04.311 [INF] BRAR: Starting contract observer, watching for breaches.
> 2019-05-08 20:19:04.312 [INF] NTFN: New block epoch subscription
> 2019-05-08 20:19:04.317 [INF] CRTR: FilteredChainView starting
> 2019-05-08 20:19:04.477 [ERR] SRVR: unable to start server: unexpected EOF
> 
> 2019-05-08 20:19:04.477 [INF] RPCS: Stopping RPC Server
> 2019-05-08 20:19:04.477 [INF] RPCS: Stopping SignRPC Sub-RPC Server
> 2019-05-08 20:19:04.478 [INF] RPCS: Stopping ChainRPC Sub-RPC Server
> 2019-05-08 20:19:04.478 [INF] RPCS: Stopping InvoicesRPC Sub-RPC Server
> 2019-05-08 20:19:04.478 [INF] RPCS: Stopping WalletKitRPC Sub-RPC Server
> 2019-05-08 20:19:04.481 [INF] LTND: Shutdown complete

did you just flash the sd and use seed+channel.backup or did you also clean the hdd?

vamp111, 2019-05-14:
@nsollazzo LND 0.6 never had the chance to properly start, so it didn't generated a channel.backup file. 
I didnt touched the HDD, only SD

turbolift, 2019-05-15:
I think the problem is undervoltage of the hdd. Before updating to 1.2 I ironically had no problems. Now I ordered a powerd usb hub. Hopefully it will work.

Currently I got this problem:

[image]

May 15 10:10:04 NodeName kernel: [480025.356531] EXT4-fs (sda1): error count since last fsck: 3157
May 15 10:10:04 Nodename kernel: [480025.356541] EXT4-fs (sda1): initial error at time 1557737446: ext4_iget:4729: inode 46399683

fluidvoice, 2019-05-16:
Please guys, post your own separate issues instead of everyone posting their information in the same issue.  It's hard to track and "LND service not running" can happen for different reasons.

[1 more reply]

closed 2021-04-19 by rootzoll