raspiblitz/raspiblitz #567: Add (optional) Y-cable to shopping lists.

fluidvoice, 2019-05-04:
Add y-cable link to shopping lists and mention recommend using separate power supply for HDD - helps resolve HDD format errors (insufficient power for external USB drive).

@fluidvoice separate power supplies.

Blockchain downloaded, just doing all the other stuff now. Letting it run it's course, thanks for the help.

_Originally posted by @rpjcbs in https://github.com/rootzoll/raspiblitz/issues/552#issuecomment-489318667_

metamarcdw, 2019-05-10:
@fluidvoice, that cable has a USB 3.0 Micro B connector. Is that compatible with the Raspberry Pi?

metamarcdw, 2019-05-11:
Wouldn't using a cable like this with an extra power supply for the HDD make it less necessary to have such a beefy power supply powering the Raspberry Pi?

fluidvoice, 2019-05-12:
There can never be enough beef. :) 
but seriously, these PS's cost often less than $10, and it's money well spent.  
most of the time they are not "beefy" even if they are rated at 3Amps because the spec's are often not accurate (they don't put out 5 volts as the current demand goes up).

metamarcdw, 2019-05-12:
Yeah, that's a fair point. I actually couldn't find my old 2.5A. I went looking for one on Amazon and there was only a $2-3 difference, so I just got the 3A that's on the list.

fluidvoice, 2019-05-12:
> Yeah, that's a fair point. I actually couldn't find my old 2.5A. I went looking for one on Amazon and there was only a $2-3 difference, so I just got the 3A that's on the list.

there are some PS rated at 5.25 volts that probably work better... allowing for voltage sag under load while still staying withing spec (4.9V or above).  Search past github issues. I've passed links in the past.

rootzoll, 2019-06-12:
@fluidvoice after checking some amazon links I think i Y-Cables have too much Variations. I think its easier to add to every shopping list the note that people should consider to add a "Powered USB3 Hub" between HDD and RaspberryPi to stabilize power. Because thats something everybody knows and USB3 Hubs are in generell all powered good enough to run a HDD securely. What do you think?

TODO: Add optional "Powered USB3 Hub" to shoppong lists.
TODO: On weak power detection add info about adding a "Powered USB3 Hub".

openoms, 2019-06-12:
> TODO: Add optional "Powered USB3 Hub" to shoppong lists.

This one proved to be very useful to me: https://www.amazon.co.uk/SuperSpeed-Individual-Switches-Charging-Splitter-USB-3-0-hub-15w-AC-adapter/dp/B07B2HJG8P

Comes with 3A power supply, separate charging port and individual switches.

https://github.com/rootzoll/raspiblitz/issues/368

closed 2019-06-12 by rootzoll