raspiblitz/raspiblitz #568: Update LND 0.6.1

rootzoll, 2019-05-04:
https://github.com/lightningnetwork/lnd/releases
Check if 0.6.1 is out of RC.

- remove LND restart on external IP change?

bitsam, 2019-05-10:
Looks like 0.6.1 is now in beta.

openoms, 2019-05-11:
Can update now:
https://github.com/openoms/bitcoin-tutorials/blob/master/lnd.update.v0.6.1-beta.sh

` wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.update.v0.6.1-beta.sh && sudo bash lnd.update.v0.6.1-beta.sh`

Tested on multiple nodes already: https://twitter.com/openoms/status/1126937869778673664

mamar59, 2019-05-12:
Hi thanks to you @openoms , it works perfectly on the rapiblitz 1.2 :+1:

rootzoll, 2019-05-17:
Keep bitcoin to 0.17.1 until LND >=0.6.2 ... see: https://twitter.com/alexbosworth/status/1129402043360108545

openoms, 2019-05-17:
Downgraded for the incoming PR (and my test branch) https://github.com/rootzoll/raspiblitz/pull/597/commits/6821b14edc082d9951c14468001c9df36dc0bad0

rootzoll, 2019-06-18:
lnd 0.7 is getting prepared and should be ready when 1.3 is ready:
https://github.com/lightningnetwork/lnd/releases/tag/v0.7.0-beta-rc1

then we also can update bitcoind to 0.18
https://github.com/bitcoin/bitcoin/releases/tag/v0.18.0

openoms, 2019-06-18:
Should update Go to v1.12.6 as well.
`The binaries are compiled with go1.12.6`

RTL is also updated to v0.4.1 now https://github.com/ShahanaFarooqui/RTL/releases

rootzoll, 2019-06-18:
RTL Update --> #660
Go Updated

O TODO: Update Protobuffs for LND 0.7

openoms, 2019-06-18:
Made a temporary update script to help testing lnd v0.7.0-beta-rc1:
`wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.update.v0.7.0-beta-rc1.sh && sudo bash lnd.update.v0.7.0-beta-rc1.sh`

openoms, 2019-06-21:
lnd v0.7.0-beta-rc2 is out: https://github.com/lightningnetwork/lnd/releases/tag/v0.7.0-beta-rc2
>We’re targeting lnd v0.7.1. to fix these remaining RPC compatibility issues.

Full compatibility with Bitcoin Core 0.18 is aimed for lnd v0.7.1 so we too should stay on Bitcoin Core v 0.17.1

rootzoll, 2019-07-03:
OK - LND 0.7.0 will now running with v1.3 release.

closed 2019-07-03 by rootzoll

rootzoll, 2019-08-02:
LCD 0.7.1 got released and we got confirmation that its working with bitcoin 0.18.0.
Updated to that versions for v1.3 release.