raspiblitz/raspiblitz #570: Security architecture

Himbeergeld, 2019-05-04:
@rootzoll The RaspiBlitz is a hot wallet, therefore a potential target of physical or remote attacks. With this issue I'd like to a) get a clear overview of the existing security architecture and b) identify improvements.

**Existing security architecture:** 
- where and how are passwords A-C stored (file locations, clear text or some type of encryption)?
- which "random" inputs are used to generate the Lighting wallet seed?
- where and how is the Lightning wallet seed stored (file locations, clear text or some type of encryption)?

**Potential improvements:**
- currently, during the update process of a new RaspiBlitz version, the master password (A) is automatically invalidated and a new master password has to be defined. Could this be a security risk? How can it be improved?
- encryption of passwords on the disk, if not encrypted already
- remove (potential Raspberry Pi-specific) deterministic components from the Lightning wallet seed generation
- what would it take to move the Lightning wallet private keys into a Secure Element (e.g. Ledger Nano S) as an option? Maybe some of the other passwords as well? Could you raise your voice in the LND community and ask for a potential improvement to "externalize" the key storage? Is there someone who could write the Ledger App for LND?

geco91, 2019-05-08:
> where and how are passwords A-C stored (file locations, clear text or some type of encryption)?

password A is stored by the linux os as password for the users - thats standard linux security

password B is stored in the lnd/bitcoin conf in clear text - but thats OK and not possible another way. password B is also used by RTL app to secure the login within local network. 

password C is only stored when user activates the auto-unlock feature. Otherwise LND can just say that password C is correct if its able to decrypt the wallet - so its not stored otherwise.

password D is also not stored on the device (if used to encrypt seed)

In generel during setup passwords could be stored in temp files, but those are deleted with unix shred command afterwards.

From my perspective this looks good. Any concerns here?

> which "random" inputs are used to generate the Lighting wallet seed?

RaspiBlitz calls LND for that - so that question needs to be directed to the LND Dev Slack.

> where and how is the Lightning wallet seed stored (file locations, clear text or some type of encryption)?

Thas also standard LND - as fas as I get it: the seed itself  is not stored. The seed is used to generate different private keys from and those are stored in the wallet.db - which is encrypted by Password C, wich is by default is not stored in the device. But while wallet is unlocked - its a hot wallet and is at risk when someone gains access to the device.

> currently, during the update process of a new RaspiBlitz version, the master password (A) is automatically invalidated and a new master password has to be defined. Could this be a security risk? How can it be improved?

On update the sd card is fresh - it can just contain a standard password. To keep the password A it would be needed to be stored on the HDD in clear text what is not a good idea. So I think as long as the SSH is not exposed to the public and the local network is not compromised - this is OK for now.

But an idea could be that a fresh sd card image on first start would generate a temp password based on time and some other random data (temp, etc) and would set this for first login and just diplay on LCD. That way only the user physically seeing the LCD could login to a fresh RaspiBlitz or during a update. BUT -> that would make the LCD a must have and some users run their RaspiBlitz without an LCD - those user would not be able to make use RaspiBlitz anymore.

> encryption of passwords on the disk, if not encrypted already

In generell a RaspiBlitz is not optimized to provide physical security. If you have physical access you can gain control. As much scary as that sounds - dont forget - even Trezor is not 100% secure against attacks if you have physical access (#walletfail). Thats why RaspiBlitz depends on the security of your home and location where you run it. This could be improved in the future but should not ease of use and ease of recovery of data. Because at the moment the risk of data loss/corruption is much higher as the physical security at your home. And encryption of HDD will make recovery of data much more harder in case of hardware fail.

> remove (potential Raspberry Pi-specific) deterministic components from the Lightning wallet seed generation

We use ARMv7 build of LND - so I cannot see that there are any RaspiPi specific components on wallet seed generation. So that would also be a question to the LND team. BTW if you dont trust the seed gen on RaspiBlitz - you could generate seed on your own/other hardware with LND and use that seed during  RaspiBlitz setup.

> what would it take to move the Lightning wallet private keys into a Secure Element

There is research to build hardware wallets for lightning - but thats early stage and we will have to see if there will be a LND integration - see: https://www.youtube.com/watch?v=Ipv-P1-pIv4

Himbeergeld, 2019-05-12:
@geco91 Thank you so much for your detailed answer and for the reference of the presentation of Crypto Advance GmbH regarding Lightning HW-wallet aspects.

Regarding the passwords: 
Password A: I didn't find it, neither in /etc/passwd nor in /etc/shadow.
Password B: I didn't find the lnd/bitcoin conf in /usr/local/bin/lnd.
Password C: If auto-unlock is enabled, where and how will it be stored?

My concern in general is that e.g. a coffee shop is proud to accept payments with his new Lightning node and posts his node in the web. Then a hacker might quickly get access to funds, maybe first trough a quick physical access reading out or changing the master password and then later quietly accessing the node remotely and executing transactions. 
It seems quite safe if the node is completely stolen because, due to the power loss, the LND wallet would be locked again.
But for the worst case I'd consider that the hacker keeps the (unlocked) LND node running, has quick physical access to the node and is familiar with the system.
How to make sure that funds are still safe in this case?
As I understand, the trick with updating a new RaspiBlitz version could actually work well (during the short period of physical access) to set a new master password, however, at this point the LND wallet will probably be locked again due to restarts.
I think it is worth to specify different attack scenarios in detail, maybe open separate issues for each of them, and discuss how they could be prevented. 
I can imagine a new monitoring / watchdog service which e.g. immediately shuts down the RaspiBlitz if "irregular activities" are identified (e.g. plugging not registered USB devices). After the shutdown it would be great if all passwords / wallets were locked and passwords are stored encrypted so that the attack was useless. On the other hand I'm aware that an immediate shutdown would not be the best approach for a running LND.

Regarding the HW-wallet for LND, the diagrams of Crypto Advance GmbH (screenshots from https://youtu.be/Ipv-P1-pIv4) indeed show the complexity, thank you for this link.
[image]
[image]
For now I'll keep an eye on https://cryptoadvance.io and hope they will keep pursuing their plan.

rootzoll, 2019-05-19:
> My concern in general is that e.g. a coffee shop is proud to accept payments with his new Lightning node and posts his node in the web. Then a hacker might quickly get access to funds, maybe first [...]

You would not run the RaspiBlitz "in the coffee shop". The solution is to run the physical RaspiBlitz in a more private place (at home or in the locked managers office) and to give the staff a device (dedicated POS or a smartphone) that is online and has just the read-only/invoice macaroons from the LND. So it can create invoices and monitor payment - but it cannot send/steal money.

I havent found an easy simply Smartphone-App that allows that - @CandleHater wanted maybe look into this at the next Lightning Hackday in Munich/Germany. But the DIY PoS device "Flux Capacitor" is already aiming in that direction: https://github.com/arcbtc/flux-capacitor

Keeping your RaspiBlitz safe at home and putting a receive-only device connected to it at the store is the best way at the moment. 

And future development of lightning-hardware-wallets or scripts RaspiBlitz-Scripts that automatically loop-out/splice-out channel funds to classic BTC-hardware-wallets will improve the security of your Home-RaspiBlitz further.

nyxnor, 2021-04-18:
Very informative.
pls close this, not to be rude, but issues solved need to be closed to focus on important ones.

closed 2021-04-19 by rootzoll