raspiblitz/raspiblitz #571: [testing 1.2]  Bash autocomplete not working for lncli

gabrieldov, 2019-05-04:
After installing the latest release, lncli bash_completion is not working. I get general bash autocomplete, but not for lncli. I believe this is something that was forgotten - probably this code? https://github.com/breez/lightninglib/blob/master/contrib/lncli.bash-completion

Please advise and let me know if you encounter the same issue. I tested ssh on 2 different machines with same results.

fluidvoice, 2019-05-04:
Raspiblitz has nothing to do with how the `lncli` or `lnd` operate.  It doesn't even compile them.  It just installs and configures the already compiled binaries.  You should ask the LND dev's and/or open an issue with them.

gabrieldov, 2019-05-05:
> https://github.com/breez/lightninglib/blob/master/contrib/lncli.bash-completion

Not sure why it happened, but I've just created a file named lncli within the /etc/bash_completion.d/ folder with the code above and now autocomplete works now. If you experience the same problem with your raspiblitz and what bash_completion to work, this is the quick fix instead of waiting for LND to fix their compiled binaries.

gabrieldov, 2019-05-05:
Also, now I see that this was added to a the 91addBonus.sh script - why it is not installed by default as before?

fluidvoice, 2019-05-05:
the commend in the script says why ... 

> "(auto install deactivated to reduce third party repos)"

gabrieldov, 2019-05-05:
Ok. But it contains 2 important functions that are still part of the main menu interface - the listing of channels and of balance. Right now, it shows error message if user selects lnchannels or lnbalance from the ./00mainmenu.sh after fresh install.

fluidvoice, 2019-05-05:
Yeah, I saw that reported by another user/issue.

geco91, 2019-05-08:
The call of this script could have slipped on the setup refactoring - recommend this to be fixed in v1.3 again.

rootzoll, 2019-05-13:
Listed to be fixed in v1.3 release

rootzoll, 2019-06-13:
OK. Implemented fix for v1.3 - needs final testing.

rootzoll, 2019-07-01:
testing looks good. This fix will be part of 1.3 release.

commit 0c8e0745 (2019-06-13): fix #571
closed 2019-07-01 by rootzoll