raspiblitz/raspiblitz #572: Can't shrink QR code

dsal1985, 2019-05-05:
Hi,
Pretty straight forward question here. Trying to setup my node so I can access it remotely. All works well but when I generate a qr code using any app selection it just won't shrink when I push control and -. Any ideas are appreciated

Thanks

fluidvoice, 2019-05-05:
what do you mean by "using any app selection"?

dsal1985, 2019-05-05:
What I meant is I have tried choosing Shango wallet and the zues wallet and couldn't shrink any of them. I'm running straight from the raspberry Pi itself, did not ssh in, couldn't seem to get that going. I'm not an expert on networking.

fluidvoice, 2019-05-05:
> I'm running straight from the raspberry Pi itself

You mean you connected a keyboard to your Pi and used Alt-F2 to access the command line and Raspiblitz menu?

fluidvoice, 2019-05-05:
Seems it's related to the terminal/shell being used.  When I SSH into my node ctrl plus/minus works for zooming in and out.  

[image]

From this issue: https://github.com/Stadicus/RaspiBolt/issues/354#issuecomment-459031989

> I'm running a light desktop (raspbian stretch lite + pixel) and was able to install gnome-terminal in order to zoom out (**raspbian native terminal does not zoom**). I suppose this would work ssh'ing in via another linux machine too. However, I was never able to get a QR of proper, scanable size using windows, PuTTy. So I guess the work around is to use a linux computer to ssh, zoom out via that linux terminal, and scan the QR, but not sure why it was so huge in the first place or how to zoom out in windows.

dsal1985, 2019-05-05:
Hi, thanks for the reply. Yes I am just plugging a keyboard into the pi itself. I was trying to use putty on windows to login but just couldn't connect. I'm more a beginner in the network side of things, I used advanced IP scanner to see my raspberry ip but it starts with a 10.0.... so I think that's not the correct ip. I will have to get ssh going to make this work. 

Thanks again

fluidvoice, 2019-05-05:
> Hi, thanks for the reply. Yes I am just plugging a keyboard into the pi itself. I was trying to use putty on windows to login but just couldn't connect. I'm more a beginner in the network side of [...]

10.0.... is fine. Routers usually hand out IP addresses like that, or starting with 192.168....
If you can use the connected keyboard and login to get a command line prompt, try running `ifconfig` to see what the full ip-address is.  Then from your Windows machine (with Putty installed) try to ping it...
`ping 10.0.xx.xx` 
if you get ping replies then you should be able to connect via SSH like this...
`ssh 10.0.xx.xx` or whatever the equivalent Putty does to connect (I don't use Windows).
Quick google of Putty gave me this link: https://www.ssh.com/ssh/putty/windows/install#sec-Configuring-and-installing

geco91, 2019-05-08:
v1.2 is displaying the QR code on the LCD (and just by option as fallback as ASCII-code in the terminal). So on a normal RaspiBlitz setup with LCD the need to shrink the QR code should not be a thing anymore and it recommended - it also helps on finding the IP of the RaspiBlitz.

Recommending this issue to close.

intorid, 2019-05-09:
I also can’t zoom to windows putty! How can I download QR cod via ssh?

rootzoll, 2019-05-10:
downloading the QR code via SSH is an interesting idea/work-around ... will add this to a future release - for the "I dont have a LCD" case. Everybody else is advised to scan the QR code from LCD which is already the default for v1.2

dsal1985, 2019-05-10:
I don't have the small 20 dollar lcd but using a monitor connected via HDMI. Even in the new version I don't see the qr code come up. It just hangs there, then I push enter and it then loads the console qr instead. I can't seem to ssh into the machine, there's obviously some sort of network issue on my end that I need to figure out. I did however notice when choosing the zues wallet, it said downloaded qr.png. I thought maybe it was now stored on the SD card but when exploring it I couldn't find such a file.

openoms, 2019-05-12:
Run the same `lndconnect` command in the terminal with the option `--image` added to generate a .png picture file which then can be downloaded with `scp` / WinSCP and opened on any desktop.

rootzoll, 2019-05-13:
@dsal1985 when using HDMI and no LCD-Hat - you need to config the RaspiBlitz manually to switch HDMI back on - see FAQ: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#can-i-run-the-raspiblitz-without-a-displaylcd

dickieadams, 2020-07-17:
It looks like if qrencode is used with "-t ansiutf8"  it will scale better in Putty (since Putty doesn't seem to natively support using ctrl+- to resize text). Example I used for my test:

`qrencode -t ansiutf8 < "filename"`

Where filename was the text from lndconnect -j.

Ben10Brasileiro, 2021-02-05:
@dickieadams Thank you, much better now. But it almost fitted using putty full screen, very close.
Do you know how to make it a little bit smaller?

dickieadams, 2021-02-06:
> @dickieadams Thank you, much better now. But it almost fitted using putty full screen, very close.
> Do you know how to make it a little bit smaller?

Check your putty display scaling. It should fit regardless if the window size.

rootzoll, 2021-02-21:
Thanks @dickieadams ... I will check if I can add the `-t ansiutf8` parameters and the `lndconnect -j text` above the QR code for v1.7 release.

rootzoll, 2021-03-23:
With v1.7RC1 the lndconnect go tool got removed due to build problems and no further maintance. It got replaced with the shell script is now generating the connection infos itself --> #2099

Also with v1.8 it will be targeted to have a webUI that makes pairing even easier.

So I am closing this issue. The optimization on QR code displaying in terminal and on LCD can be further commented on #2099

closed 2021-03-23 by rootzoll