raspiblitz/raspiblitz #575: [v1.2] LND service not running

nsollazzo, 2019-05-05:
After upgrading to v.1.2 i have the experimented an issue "LND Service not running"

My raspberry has an 5V/3A power supply, and I have an HDD plugged into a Hub with an external 5V/2A.
here's my log: https://termbin.com/4060

_Originally posted by @nsollazzo in https://github.com/rootzoll/raspiblitz/issues/565#issuecomment-489419625_

fluidvoice, 2019-05-05:
What is your results of running the HardwareTest?...
`sudo /home/admin/05hardwareTest.sh`

fluidvoice, 2019-05-05:
ah... interesting.  You are getting this "edge not found" error.  My node is also seeing this error on v1.2

> *** LAST LND ERROR LOGS ***
> sudo journalctl -u lnd -b --no-pager -n12
> -- Logs begin at Sun 2019-05-05 09:45:57 BST, end at Sun 2019-05-05 12:50:33 BST. --
> May 05 12:48:46 Fulgur systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
> May 05 12:48:46 Fulgur systemd[1]: Stopped LND Lightning Daemon.
> May 05 12:48:46 Fulgur systemd[1]: Starting LND Lightning Daemon...
> May 05 12:48:46 Fulgur systemd[1]: Started LND Lightning Daemon.
> **May 05 12:49:07 Fulgur lnd[24054]: edge not found**
> May 05 12:49:07 Fulgur systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
> May 05 12:49:07 Fulgur systemd[1]: lnd.service: Unit entered failed state.
> May 05 12:49:07 Fulgur systemd[1]: lnd.service: Failed with result 'exit-code'.
> May 05 12:50:08 Fulgur systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
> May 05 12:50:08 Fulgur systemd[1]: Stopped LND Lightning Daemon.

fluidvoice, 2019-05-05:
Please send links to the full LND logs:
`sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`
`sudo cat /var/log/syslog | nc termbin.com 9999`

nsollazzo, 2019-05-05:
> What is your results of running the HardwareTest?...
> `sudo /home/admin/05hardwareTest.sh`

hardware test output:
```
*** Hardware Test Report ***

RaspiBlitz Hardwaretest v0.1
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
68.8'C 1200/1400 MHz 1.2625V
--> Temp WARN detected
69.3'C 1200/1400 MHz 1.2625V
--> Temp WARN detected
69.3'C 1200/1400 MHz 1.2625V
--> Temp WARN detected
69.8'C 1200/1400 MHz 1.2625V
--> Temp WARN detected
70.4'C 1200/1400 MHz 1.2625V
--> Temp CRITICAL detected
--> Temp WARN detected
70.9'C 1200/1400 MHz 1.2625V
--> Temp CRITICAL detected
--> Temp WARN detected
70.9'C 1200/1400 MHz 1.2625V
--> Temp CRITICAL detected
--> Temp WARN detected
71.4'C 1200/1400 MHz 1.2625V
--> Temp CRITICAL detected
--> Temp WARN detected
71.4'C 1200/1400 MHz 1.2625V
--> Temp CRITICAL detected
--> Temp WARN detected
71.4'C 1200/1400 MHz 1.2625V
--> Temp CRITICAL detected
--> Temp WARN detected
72.5'C 1200/1400 MHz 1.2625V
--> Temp CRITICAL detected
--> Temp WARN detected
68.8'C 1200/1400 MHz 1.2625V
--> Temp WARN detected
67.7'C 1200/1400 MHz 1.2625V
--> Temp WARN detected
67.1'C  600/1400 MHz 1.2625V
--> Temp WARN detected
66.6'C 1200/1400 MHz 1.2625V
--> Temp WARN detected

OK continue ..
OK continue ..
```

> Please send links to the full LND logs:
> `sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`
> `sudo cat /var/log/syslog | nc termbin.com 9999`

lnd log : https://termbin.com/blc0
syslog : https://termbin.com/bzvx

fluidvoice, 2019-05-05:
> hardware test output:

That doesn't look too bad.  There is a little CPU throttling but 1.26V isn't **too** bad.  Could be better but could be worse.

nsollazzo, 2019-05-05:
> > hardware test output:
> 
> That doesn't look too bad. There is a little CPU throttling but 1.26V isn't **too** bad. Could be better but could be worse.

I have 2 copper heat sinks I think it is for that. IMHO the LCD screen is covering the pi to much but it is extremely useful

fluidvoice, 2019-05-05:
> > > hardware test output:
> > 
> > 
> > That doesn't look too bad. There is a little CPU throttling but 1.26V isn't **too** bad. Could be better but could be worse.
> 
> I have 2 copper heat sinks I think it is for that. IMHO the LCD screen is covering the pi to much but it is extremely useful

CryptoCloaks is working on a new 3D printed case, I'm hoping he will include a fan in.  But even just using the "Armor" heatsink with no fan helps - see my tweet: https://twitter.com/fluidvoice/status/1123967360073895936

fluidvoice, 2019-05-05:
> but 1.26V isn't **too** bad.

If I remember, 1.375V is full voltage some people have seen with 1400/1400 Mhz (no throttling).

nsollazzo, 2019-05-05:
> > > > hardware test output:
> > > 
> > > 
> > > That doesn't look too bad. There is a little CPU throttling but 1.26V isn't **too** bad. Could be better but could be worse.
> > 
> > 
> > I have 2 copper heat sinks I think it is for that. IMHO the LCD screen is covering the pi to much but it is extremely useful
> 
> CryptoCloaks is working on a new 3D printed case, I'm hoping he will include _the ability to install a fan_ in. But even just using the "Armor" heatsink with no fan helps. Rootzoll has the Armor case and says it brings temps down a lot.
> https://twitter.com/CryptoCloaks/status/1123951682872848390
> https://twitter.com/fluidvoice/status/1123962942280601601

Awesome, I'm looking forward for the CryptoCloaks' case

fluidvoice, 2019-05-05:
I'm going to try to compile the latest release of LND **0.6-beta-rc1** and test it as they have made a number of code changes related to "edge" stuff - not that I know what that means ;-)  But no promises. I've not built LND before.

nsollazzo, 2019-05-06:
> I'm going to try to compile the latest release of LND **0.6-beta-rc1** and test it as they have made a number of code changes related to "edge" stuff - not that I know what that means ;-) But no promises. I've not built LND before.

thank you, let me know

fluidvoice, 2019-05-06:
> > I'm going to try to compile the latest release of LND **0.6-beta-rc1** and test it as they have made a number of code changes related to "edge" stuff - not that I know what that means ;-) But no promises. I've not built LND before.
> 
> thank you, let me know

No go so far.  I tried to compile it with two different methods and could not. It failed with errors.  Waiting to see if the LND dev can comment and help me out.

nsollazzo, 2019-05-06:
> No go so far. I tried to compile it with two different methods and could not. It failed with errors. Waiting to see if the LND dev can comment and help me out.

I've cloned the lnd repo and I wrote the patch as the comment in the issued you opened said, I was thinking to spin up a container, have you tried it yet?

fluidvoice, 2019-05-06:
> I've cloned the lnd repo and I wrote the [patch](https://github.com/lightningnetwork/lnd/issues/3025#issuecomment-488999784) as the comment in the issued you opened said , I was thinking to spin up [...]

as I said, I was not able to compile so no sense looking at the patch

fluidvoice, 2019-05-06:
> I've cloned the lnd repo and I wrote the [patch](https://github.com/lightningnetwork/lnd/issues/3025#issuecomment-488999784) as the comment in the issued you opened said , I was thinking to spin up [...]

OK got it built now.  Rebooted and testing. My LND is trying to finish syncing.  Will let you know.

btc21cluib, 2019-05-08:
Same Problem LND not running
when I try lncli unlock
 I get
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

 here the debug link https://termbin.com/432a

nsollazzo, 2019-05-08:
> Same Problem LND not running
> when I try lncli unlock
> I get
> [lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = [...]

1. Open your own separate issue here (click "..." and select "reference in new issue")
2. i think it's not the same issue, because i saw this on your logs:
> 2019-05-08T18:01:27Z ThreadRPCServer incorrect password attempt from 127.0.0.1:46232

in your blockchain logs, and this:

>2019-05-08 19:17:37.908 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

on your LND logs.

So i think it's a password error

btc21cluib, 2019-05-08:
@nsollazzo  - Thanks for you assistance, I open a new issue

rektlesspinko, 2019-05-10:
@fluidvoice @nsollazzo

I don;t know whether I should be raising as a separate issue, but  I'm also getting an  [ERR] SRVR: unable to start server: edge not found. 

I Upgraded my raspiblitz from 1.1 to 1.2, it was running quite happily overnight, with 7/9 channels open and about 7 or 8 peer connections. Then I noticed the lnd service had stopped running. I rebooted, lnd service came back up, but no peer connections established and then after 5 mins or so, the lnd service would stop. This loop would then continue ad infinitum. lnd would start, no peers, hit this unable to start server: edge not found issue and the service would shutdown again and again ad infinitum. 

 Does anyone have a fix for this? Looking in the issue you raised in lnd (#3025), (I'm not particularly technical so a lot of it went over my head) but it was suggesting the database is corrupted. Does that mean, I've got to try and restore from the channel.backup file? 

Is this done by running  /home/admin/config.scripts/lnd.rescue.sh backup from the terminal?

I don't have a lot of funds on there (and most in channels), but I'd like to recover them if poss.

If I do get them back, will definitely be buying heat sink and a new powerlead as the hardware test I ran before these issue arose suggested these improvements (although had been running quite happily on v0.99, v1.0, and v1.1 quite happily. Perhaps the move to LND 0.6.0 has made this set up more fragile ?)

lnd log snippet below

2019-05-09 18:26:38.484 [DBG] CNCT: ChannelArbitrator(43d966950a9572bfc8a74a9e6068d64acd85065164799a097ce5093f3a407d48:0): checking chain actions at height=575290
2019-05-09 18:26:38.497 [INF] DISC: Authenticated Gossiper is starting
2019-05-09 18:26:38.498 [INF] BRAR: Starting contract observer, watching for breaches.
2019-05-09 18:26:38.498 [INF] NTFN: New block epoch subscription
2019-05-09 18:26:38.503 [INF] CRTR: FilteredChainView starting
2019-05-09 18:26:38.505 [DBG] DISC: Requesting online notification for peer=02016e4df84a0c729f459d5842529900404aa5469c0f653f1b012c27775021a5d4
2019-05-09 18:26:38.506 [DBG] DISC: Skipping retransmission of channel without AuthProof: 626032234136338432
2019-05-09 18:26:39.575 [ERR] SRVR: unable to start server: edge not found

2019-05-09 18:26:39.575 [INF] RPCS: Stopping RPC Server
2019-05-09 18:26:39.575 [INF] RPCS: Stopping WalletKitRPC Sub-RPC Server
2019-05-09 18:26:39.576 [INF] RPCS: Stopping SignRPC Sub-RPC Server
2019-05-09 18:26:39.576 [INF] RPCS: Stopping ChainRPC Sub-RPC Server
2019-05-09 18:26:39.576 [INF] RPCS: Stopping InvoicesRPC Sub-RPC Server

fluidvoice, 2019-05-12:
> @fluidvoice @nsollazzo
>
> I don;t know whether I should be raising as a separate issue, but I'm also getting an [ERR] SRVR: unable to start server: edge not found.
>
> I Upgraded my raspiblitz from 1.1 [...]

can you run this and post the resulting link to the log files, thanks:
raspiblitz log: `/home/admin/XXdebugLogs.sh | nc termbin.com 9999`
lnd log: `sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`
system log: `sudo cat /var/log/syslog | nc termbin.com 9999`

rektlesspinko, 2019-05-13:
> > @fluidvoice @nsollazzo
> > I don;t know whether I should be raising as a separate issue, but I'm also getting an [ERR] SRVR: unable to start server: edge not found.
> > I Upgraded my raspiblitz from 1.1 to 1.2, it was running quite happily overnight, with 7/9 channels open and about 7 or 8 peer connections. Then I noticed the lnd service had stopped running. I rebooted, lnd service came back up, but no peer connections established and then after 5 mins or so, the lnd service would stop. This loop would then continue ad infinitum. lnd would start, no peers, hit this unable to start server: edge not found issue and the service would shutdown again and again ad infinitum.
> > Does anyone have a fix for this? Looking in the issue you raised in lnd (#3025), (I'm not particularly technical so a lot of it went over my head) but it was suggesting the database is corrupted. Does that mean, I've got to try and restore from the channel.backup file?
> > Is this done by running /home/admin/config.scripts/lnd.rescue.sh backup from the terminal?
> > I don't have a lot of funds on there (and most in channels), but I'd like to recover them if poss.
> > If I do get them back, will definitely be buying heat sink and a new powerlead as the hardware test I ran before these issue arose suggested these improvements (although had been running quite happily on v0.99, v1.0, and v1.1 quite happily. Perhaps the move to LND 0.6.0 has made this set up more fragile ?)
> > lnd log snippet below
> > 2019-05-09 18:26:38.484 [DBG] CNCT: ChannelArbitrator(43d966950a9572bfc8a74a9e6068d64acd85065164799a097ce5093f3a407d48:0): checking chain actions at height=575290
> > 2019-05-09 18:26:38.497 [INF] DISC: Authenticated Gossiper is starting
> > 2019-05-09 18:26:38.498 [INF] BRAR: Starting contract observer, watching for breaches.
> > 2019-05-09 18:26:38.498 [INF] NTFN: New block epoch subscription
> > 2019-05-09 18:26:38.503 [INF] CRTR: FilteredChainView starting
> > 2019-05-09 18:26:38.505 [DBG] DISC: Requesting online notification for peer=02016e4df84a0c729f459d5842529900404aa5469c0f653f1b012c27775021a5d4
> > 2019-05-09 18:26:38.506 [DBG] DISC: Skipping retransmission of channel without AuthProof: 626032234136338432
> > 2019-05-09 18:26:39.575 [ERR] SRVR: unable to start server: edge not found
> > 2019-05-09 18:26:39.575 [INF] RPCS: Stopping RPC Server
> > 2019-05-09 18:26:39.575 [INF] RPCS: Stopping WalletKitRPC Sub-RPC Server
> > 2019-05-09 18:26:39.576 [INF] RPCS: Stopping SignRPC Sub-RPC Server
> > 2019-05-09 18:26:39.576 [INF] RPCS: Stopping ChainRPC Sub-RPC Server
> > 2019-05-09 18:26:39.576 [INF] RPCS: Stopping InvoicesRPC Sub-RPC Server
> 
> can you run this and post the resulting links to the log files, thanks:
> raspiblitz log: `/home/admin/XXdebugLogs.sh | nc termbin.com 9999`
> lnd log: `sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`
> system log: `sudo cat /var/log/syslog | nc termbin.com 9999`

Please see log files  below
https://termbin.com/buq5
https://termbin.com/tsf5
https://termbin.com/m64m

thanks

fluidvoice, 2019-05-16:
> Please see log files below
> https://termbin.com/buq5
> https://termbin.com/tsf5
> https://termbin.com/m64m
>
> thanks

There is no known fix for the "edge not found" problem which may be caused by data corruption.
I recommend you check/test and possibly replace your power supply, then backup, re-install, and recover your Raspiblitz wallet as per the docs/FAQ.

fluidvoice, 2019-06-01:
There is a potential fix in upcoming LND v0.7.0 but it only avoids the problem for working nodes, it does not cure the problem for nodes already with the database corruption:
https://github.com/lightningnetwork/lnd/issues/3025#issuecomment-497458753

> We've found a potential deadlock when polling GetInfo that could cause the daemon to not shut down cleanly, which would explain why users are experiencing this database issue.
> 
> The deadlock would cause lnd not to shut down completely, so if the process gets force killed then it's possible to run into database corruption. The fix is now in master and will be included in the upcoming v0.7.0 release, so it should prevent nodes with this fix from running into this issue. **For any other nodes without this fix that continue to run into this issue, there's not much we can do other than the recommended recovery process.**
> 
> With that said, I'll go ahead and close this for now. Once v0.7.0 is out and most Raspiblitz nodes have upgraded, we can re-examine any further reports if the issue seems to persist.

rootzoll, 2019-06-18:
for progress - check: #638

frennkie, 2020-08-08:
looks good.

closed 2020-08-08 by frennkie