raspiblitz/raspiblitz #576: Always PWR warning at new install

Smiggel, 2019-05-05:
I always get a power supply warning when I run a new install. I use a Raspberry Pi 3, with and LCD display and a 3A power supply.

Next to the above I use a WD My Passport 1TB HDD that is attached to a powered USB hub. So the harddisk power if powered by the hub and not the pi. And still I get a power warning. I tried different power supplies of all 3A. Doesn't make a difference.

Perhaps you can check the script?

fluidvoice, 2019-05-05:
paste results of this...

PowerTest script (all in one line):

`wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh && ./XXpowertest.sh`

Smiggel, 2019-05-05:
> --2019-05-05 18:38:50--  https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.36.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.36.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1835 (1.8K) [text/plain]
Saving to: ‘XXpowertest.sh’

XXpowertest.sh         100%[===========================>]   1.79K  --.-KB/s    in 0s      

2019-05-05 18:38:53 (14.6 MB/s) - ‘XXpowertest.sh’ saved [1835/1835]

***
RaspiBlitz powertest v0.1
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
***

sysbench 0.4.12:  multi-threaded system evaluation benchmark

Running the test with following options:
Number of threads: 4

Doing CPU performance benchmark

Threads started!
66.6'C 1200 MHz  1.2500V
75.2'C 1200 MHz  1.2500V
78.4'C 1200 MHz  1.2500V
81.1'C 1141/1200 MHz 1.2500V
81.7'C 1034/1200 MHz 1.2500V
82.2'C  980/1200 MHz 1.2500V
82.7'C  926/1200 MHz 1.2500V
82.2'C  926/1200 MHz 1.2500V
82.7'C  980/1200 MHz 1.2500V
Done.

Maximum prime number checked in CPU test: 10000

Test execution summary:
    total time:                          49.0839s
    total number of events:              10000
    total time taken by event execution: 196.2512
    per-request statistics:
         min:                                 13.89ms
         avg:                                 19.63ms
         max:                                102.08ms
         approx.  95 percentile:              29.93ms

Threads fairness:
    events (avg/stddev):           2500.0000/85.01
    execution time (avg/stddev):   49.0628/0.01

79.5'C 1200 MHz  1.2500V
74.1'C 1200 MHz  1.2500V
72.5'C 1200 MHz  1.2500V
70.9'C 1200 MHz  1.2500V
69.8'C 1200 MHz  1.2500V
69.8'C 1200 MHz  1.2500V

fluidvoice, 2019-05-05:
That looks OK. 
Where is your debug log when you "get a power supply warning when I run a new install"?
You didn't even say which version you installed.  How can we "check the script" when we don't even know what you ran?

Smiggel, 2019-05-06:
This is the output of my logs:

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 07:30:18 up 13:10,  3 users,  load average: 3.15, 3.25, 3.19

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-05-05 22:38:08 BST; 8h ago
 Main PID: 3074 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─3074 /usr/local/bin/bitcoind -daemo…id
May 05 22:38:07 RaspiBlitz systemd[1]: Starting …..May 05 22:38:08 RaspiBlitz systemd[1]: Started B…n.Hint: Some lines were ellipsized, use -l to show in full.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Mon 2019-05-06 07:30:18 BST. --
May 05 22:38:07 RaspiBlitz systemd[1]: Starting Bitcoin daemon...
May 05 22:38:08 RaspiBlitz systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-05-06T06:16:06Z Pre-allocating up to position 0x600000 in rev01571.dat
2019-05-06T06:16:06Z UpdateTip: new best=00000000000000000013a3804ac2e89bc1b92a6294bb5679de9bfa562dc7722d height=568654 version=0x20c00000 log2_work=90.475421 tx=394588424 date='2019-03-24T23:46:24Z' progress=0.978269 cache=12.4MiB(50912txo)
2019-05-06T06:16:06Z socket sending timeout: 1234s
2019-05-06T06:16:06Z ping timeout: 1234.713522s
2019-05-06T06:16:06Z ping timeout: 1235.580159s
2019-05-06T06:16:06Z ping timeout: 1235.355153s
2019-05-06T06:16:06Z New outbound peer connected: version: 70015, blocks=574805, peer=2376
2019-05-06T06:16:07Z New outbound peer connected: version: 70015, blocks=574805, peer=2421
2019-05-06T06:16:23Z UpdateTip: new best=00000000000000000008c54c1533833a8adea52dd249e10e4de44c29fb0d37c5 height=568655 version=0x20000000 log2_work=90.475444 tx=394591524 date='2019-03-24T23:56:47Z' progress=0.978273 cache=13.3MiB(58552txo)
2019-05-06T06:16:53Z UpdateTip: new best=000000000000000000230242426600677148027be2abe9a8dec3b203604f26bc height=568656 version=0x20000000 log2_work=90.475467 tx=394593458 date='2019-03-24T23:58:29Z' progress=0.978273 cache=14.3MiB(67908txo)
2019-05-06T06:17:00Z UpdateTip: new best=00000000000000000029851fc816670dd78b8bd256278b0c228389e4202f7a1f height=568657 version=0x20800000 log2_work=90.47549 tx=394595728 date='2019-03-25T00:10:47Z' progress=0.978278 cache=14.9MiB(74277txo)
2019-05-06T06:17:02Z UpdateTip: new best=00000000000000000005ab1bbc74a2a5915c07451496987013a4610bb6001211 height=568658 version=0x20000000 log2_work=90.475513 tx=394596203 date='2019-03-25T00:13:33Z' progress=0.978279 cache=15.0MiB(75291txo)
2019-05-06T06:18:45Z UpdateTip: new best=00000000000000000027c290019c51eec723f6bdcb1be5833b7828dd5ef186d2 height=568659 version=0x20400000 log2_work=90.475536 tx=394597199 date='2019-03-25T00:31:38Z' progress=0.978284 cache=15.7MiB(81855txo)
2019-05-06T06:22:30Z UpdateTip: new best=00000000000000000004509074b5d827293da17e910815439cedcd958b0fb0eb height=568660 version=0x20000000 log2_work=90.475559 tx=394597458 date='2019-03-25T00:35:13Z' progress=0.978284 cache=16.5MiB(88634txo)
2019-05-06T06:26:03Z UpdateTip: new best=000000000000000000166c26aa62a969ca2e4df4ad2b29e635c3f906e1551c26 height=568661 version=0x20c00000 log2_work=90.475582 tx=394597845 date='2019-03-25T00:35:35Z' progress=0.978283 cache=17.2MiB(95425txo)
2019-05-06T06:27:07Z Pre-allocating up to position 0x700000 in rev01571.dat
2019-05-06T06:27:07Z UpdateTip: new best=000000000000000000107099fd187f9f7c09b38595025b89cf331b44ec5f8522 height=568662 version=0x20000000 log2_work=90.475605 tx=394600904 date='2019-03-25T00:42:56Z' progress=0.978286 cache=17.8MiB(101157txo)
2019-05-06T06:28:08Z UpdateTip: new best=00000000000000000001945b42307532abd9babe70d277317b26e26f8f5b96c6 height=568663 version=0x20000000 log2_work=90.475628 tx=394603831 date='2019-03-25T01:00:33Z' progress=0.978292 cache=18.5MiB(108078txo)
2019-05-06T06:28:39Z UpdateTip: new best=00000000000000000024602d30086b513f25c13afd3551df1ecd42b56c8ad0e0 height=568664 version=0x20800000 log2_work=90.475651 tx=394605730 date='2019-03-25T01:12:42Z' progress=0.978296 cache=20.4MiB(125170txo)
2019-05-06T06:28:56Z UpdateTip: new best=00000000000000000007a34bac55bacc15f7af40b730d425d041cd98db8dc472 height=568665 version=0x20000000 log2_work=90.475674 tx=394606283 date='2019-03-25T01:14:41Z' progress=0.978296 cache=20.5MiB(126272txo)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-05-05 22:46:29 BST; 8h ago
 Main PID: 6899 (lnd)
   CGroup: /system.slice/lnd.service
           └─6899 /usr/local/bin/lnd --externali…35
May 05 22:46:29 RaspiBlitz systemd[1]: Starting …..May 05 22:46:29 RaspiBlitz systemd[1]: Started L…n.Hint: Some lines were ellipsized, use -l to show in full.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Mon 2019-05-06 07:30:19 BST. --
May 05 22:41:46 RaspiBlitz systemd[1]: Starting LND Lightning Daemon...
May 05 22:41:46 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
May 05 22:46:15 RaspiBlitz systemd[1]: Stopping LND Lightning Daemon...
[... cut ...]
UndervoltageReports in Logs: 1
powerFAIL=0
powerWARN=1
powerMIN='1200000 microVolt'
tempFAIL=1
tempWARN=2
tempMAX='7040 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.0.50'
tempCelsius='64.4'
uptime=47412
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=568667
initialSync=1
syncProgress=97.83 
startcountLightning=2
lndActive=1

fluidvoice, 2019-05-06:
> UndervoltageReports in Logs: 1

If your node is running OK, I wouldn't worry about the power warnings.  Seems like your low power is happening infrequently.

Smiggel, 2019-05-06:
> > UndervoltageReports in Logs: 1
> 
> If your node is running OK, I wouldn't worry about the power warnings. Seems like your low power is happening infrequently.

Ok, Thanks. It looks stable. :-)

geco91, 2019-05-08:
@fluidvoice is there any fine tuning of the power waring detection to recommend for next version from this?

fluidvoice, 2019-05-08:
> @fluidvoice is there any fine tuning of the power waring detection to recommend for next version from this?

don't know. required voltage is something north of 2.5V to be safe.

rootzoll, 2019-05-13:
TODO v1.3: When running with no power warnings on start - delete syslogs to remove/reset Undervoltage Errors.

rootzoll, 2019-06-13:
OK. Its part of the v1.3 release

closed 2019-06-13 by rootzoll