raspiblitz/raspiblitz #577: RESEARCH: enable 2-factor auth for SSH

fluidvoice, 2019-05-05:
supposedly it's possible.  Helpful?
https://www.cnx-software.com/2019/05/03/two-factor-authentication-ssh-armbian/

CommanderPoe, 2019-05-06:
Would something like this mean that i can securely open the port 22 so i can ssh from anywhere in the world from my cellphone without being worried about my security 24/7 ? If so, A big YES !

geco91, 2019-05-08:
That could be an interesting option. The tutorial is aimed at armbian OS ... but there is also a link to setup with custom debian. Can anybody try this and confirm that it is working also on rasbian?

https://www.vultr.com/docs/how-to-setup-two-factor-authentication-2fa-for-ssh-on-debian-9-using-google-authenticator

BUT: If it involves installing some third-party code (on the RaspiBlitz) - that is not part of debian packages, I would not recommend to add this feature.

nyxnor, 2021-04-18:
[Authy ](https://raw.github.com/authy/authy-ssh/master/authy-ssh)option, but it is [unmantained](https://github.com/authy/authy-ssh/blob/master/authy-ssh).
[There is authrelia, maintained but written in go.](https://github.com/authelia/authelia)

nyxnor, 2021-06-04:
https://github.com/rootzoll/raspiblitz/issues/1681#issuecomment-854552788
Openoms gave a better guide here with multiple options.

closed 2023-06-07 by cstenglein