raspiblitz/raspiblitz #579: Feature request: Auto-update bitcoind per default

martinbehrens, 2019-05-06:
I just noticed that most of the listening bitcoind nodes worldwide are still running v0.17.

Wouldn't it make sense to automate an update of bitcoin core on the devices every, for example, minor version bump per default? In order to not patronize the user the automatic update can be disabled per switch in the settings somewhere.

fluidvoice, 2019-05-06:
I don't think so. Important software should not be defaulted to auto-updating. Better would be to detect an update is available and notify the admin/user.

martinbehrens, 2019-05-06:
And with that notification there can be opt-in options for future updates, such as:

- download only (git pull),
- download and build (git/configure/make) and
- download, build and install.

As long as there is consent, it should be fine to simplify the process. Especially since on a raspberry pi building it probably takes ages.

However the specs will look like in detail I just wanted to bring up the feature in general.

fluidvoice, 2019-05-06:
> And with that notification there can be opt-in options for future updates, such as:
>
> * download only (git pull),
> * download and build (git/configure/make) and
> * download, build and install.
>
> As long [...]

I think it's a bad idea.  Do you know which, if any, other full nodes that do auto-updating?  Opt-in isn't always a good idea because some users may not realize the risks/implications of this. The worst would be an auto-update that breaks things and their node won't start, lost funds, and various other things.

CryptoSigmund, 2019-05-07:
Autoupdate should only be available if the auto-backup is enabled and it have to run successfully in the minute before the auto-update starts.

Update options should be:
every version
only major security updates
just inform me

martinbehrens, 2019-05-07:
> Autoupdate should only be available if the auto-backup is enabled and it have to run successfully in the minute before the auto-update starts.

Full ACK.

geco91, 2019-05-08:
At the moment the update mechanism is to reflash the sd-card with a newer image. Why it is done this way see in detail in the FAQ: https://github.com/geco91/raspiblitz/blob/master/FAQ.md#why-do-i-need-to-re-burn-my-sd-card-for-an-update

To add: This way its even possible to change the base linux os system at one point if needed ... like a change from raspbian to a more flexible debian distro that can support more diverse hardware.

In short: The sd-card update gives a lot of flexibility on the development side and robustness against a central source of updates on the user side - because an SD card update can easily be shared within groups and fallback to sneakernet - bypassing central services like GitHub or FTP servers and the update for the user would still be the same mechanism.

geco91, 2019-05-08:
But v1.2 started to at least make it more easy for people to go thru the update process with the "UPDATE" option in the main menu - detecting if a new version is available in general and guiding the user thru the process - even helping on making a backup before changing sd cards. I think this is a working compromise for now.

I recommend this issue/discussion to close for now.

closed 2019-05-13 by rootzoll