raspiblitz/raspiblitz #580: is the lnd cipher seed, also the seed to restore the on chain btc wallet

coinicarus, 2019-05-06:
This isnt an issue as more of a question, is the LND 24 word cipher seed the way you restore the on chain wallet? if not how do you generate the seed for the on chain wallet? 
thank you!

geco91, 2019-05-08:
See FAQ on how to recover funds from the RaspiBlitz and the role the seed words play in that:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

I recommend this issue to close.

zuardi, 2019-05-08:
Thats not what he asked. I believe he is asking how to import his LND wallet into a BIP-39 wallet, transforming his lnd-24 word seed into bip39 24 words seed.

coinicarus, 2019-05-08:
so essentially I F*d up. I thought(mistakenly) that 24 word seed generated when setting up lnd was my 24word seed for the on chain BTC wallet(i am totally wrong) so now i have a certain amount of BTC 'lost' and the only thing i have is that 24 word LND cipher seed. I had checked the FAQ but possibly i missed something.  I tried importing the LND cipher seed into electrum to no avail. i then figured i would try the lightning labs wallet... also to no avail :( I am willing to write this off as a loss but would love to figure out if i can somehow recover the funds. thank you both zuardi & geco91 for your responses this far.

zuardi, 2019-05-08:
Im pretty much at same situation. My node doesn't work when I try to recover my old wallet (some kind of out of the bounds range error) but works fine with a new wallet.

CommanderPoe, 2019-05-09:
The 24 words DO NOT RECOVER either onchain nor offchain funds! In order to recover the funds that once you sent to that wallet you need to have the walletdb for onchain and channeldb for offchain directories that are included in the lnd folder. At least this is my personla experience when trying to recover my funds, i only managed to recover them ALL once i imported those 2 directories i stated before.

#278  is a very very long thread about the topic.

coinicarus, 2019-05-09:
thanks for all your help guys, i will check those topics mentioned by commanderPoe

closed 2019-05-09 by coinicarus

lebowski36, 2019-05-28:
Are you serious? Every bitcoin wallet I ever used only need's the seed for recovery. The raspiblitz is not on this train? I thought my seed is enough to recover any onchain funds?

openoms, 2019-05-28:
@lebowski36 The RaspiBlitz is not a bitcoin wallet. It is running LND on top Bitcoin Core, which works differently beacuse of the added complexity of the Lightning Network. LND is very much in the beta state despite the Lightning Network growing quickly. It works already quite reliably, but there still some things which are not as intuitive as anyone would like.