raspiblitz/raspiblitz #581: Bug: /home/admin/00mainMenu.sh: line 119: lnchannels: command not found

Smiggel, 2019-05-08:
I get the error from the title when I try to display my lightning balance using the admin menu.

I recently restored my node and upgraded to the latest version of RaspiBlitz. I back upped my Lightning channels. Then I updated my Node software, let the blockchain sync and then restored my channels and funds.

After the system was back up and running and I wanted to look at the wallet balance, I got that error. Not sure if it's just my installation of has the do with the latest image. I can however watch the balance using Zap, when I connect that wallet. So, no real harm is done, but just notice the menu item does not work.

bitsam, 2019-05-08:
Did you upgrade to Raspiblitz v1.2? I also have this problem on v1.1.

fluidvoice, 2019-05-08:
It's a known problem.  When you have a problem the first thing to do is search for it, like this...
[image]

The workaround until it gets fixed in the Raspiblitz scripts is to run these commands from an SSH terminal command line:

`wget https://stadicus.github.io/RaspiBolt/resources/lnbalance`
`wget https://stadicus.github.io/RaspiBolt/resources/lnchannels`
`chmod +x lnbalance && chmod +x lnchannels`
`sudo cp lnchannels /usr/local/bin && sudo cp lnbalance /usr/local/bin`

geco91, 2019-05-08:
Recommend to add this issue to be fixed in next v1.3 release.

rootzoll, 2019-05-13:
Listed to be fixed in v1.3 release.

rootzoll, 2019-06-13:
Fixed for v1.3

closed 2019-06-13 by rootzoll