raspiblitz/raspiblitz #584: Error when generating zap mobile wallet QR code

jaramburu, 2019-05-08:
Running v1.2. When I try to generate the QR code to connect the Zap mobile wallet I get an error in the terminal, and the QR code fails to load on the raspiblitz LCD screen. I also cannot display the QR code in terminal. The code for Zeus wallet does seem to display properly.

jaramburu, 2019-05-09:
This is the error message I see when I try to display the QR code in the terminal:

running RPC mode
### Check Framework: GO ###
*** Setup ***

Installing lndconnect. Please wait...

Getting github.com/LN-Zap/lndconnect (please wait - can take several minutes) ...
/usr/local/gocode/src/github.com/LN-Zap/lndconnect/config.go:15:2: 
/usr/local/gocode/src/github.com/jessevdk/go-flags/arg.go:1:1: expected 'package', found s

Building github.com/LN-Zap/lndconnect ...
go get ./...
config.go:15:2: 
../../jessevdk/go-flags/arg.go:1:1: expected 'package', found s
Makefile:4: recipe for target 'dep' failed
make: *** [dep] Error 1
port 10009 forwarding from dynDomain playa1.crabdance.com
./97addMobileWalletLNDconnect.sh: line 102: lndconnect: command not found
./97addMobileWalletLNDconnect.sh: line 119: lndconnect: command not found
(To shrink QR code: OSX->CMD- / LINUX-> CTRL-) Press ENTER when finished.

jaramburu, 2019-05-09:
running RPC mode
### Check Framework: GO ###
*** Setup ***

Installing lndconnect. Please wait...

Getting github.com/LN-Zap/lndconnect (please wait - can take several minutes) ...
/usr/local/gocode/src/github.com/LN-Zap/lndconnect/config.go:15:2: 
/usr/local/gocode/src/github.com/jessevdk/go-flags/arg.go:1:1: expected 'package', found s

Building github.com/LN-Zap/lndconnect ...
go get ./...
config.go:15:2: 
../../jessevdk/go-flags/arg.go:1:1: expected 'package', found s
Makefile:4: recipe for target 'dep' failed
make: *** [dep] Error 1
port 10009 forwarding from dynDomain playa1.crabdance.com
./97addMobileWalletLNDconnect.sh: line 102: lndconnect: command not found

rootzoll, 2019-05-13:
Zap mobile wallet QR code generation works on my test machine. 
Maybe some downloads on install not working?

kaiobe, 2019-06-21:
I have the same issue @jaramburu. Did you find a workaround?

--------------------------------------------------------------------------
fbi: no process found
running REST mode
### Check Framework: GO ###
*** Setup ***

Installing lndconnect. Please wait...

Getting github.com/LN-Zap/lndconnect (please wait - can take several minutes) ...
# cd .; git clone https://github.com/LN-Zap/lndconnect /usr/local/gocode/src/github.com/LN-Zap/lndconnect
Cloning into '/usr/local/gocode/src/github.com/LN-Zap/lndconnect'...
fatal: unable to access 'https://github.com/LN-Zap/lndconnect/': Could not resolve host: github.com
package github.com/LN-Zap/lndconnect: exit status 128
./97addMobileWalletLNDconnect.sh: line 64: cd: /usr/local/gocode/src/github.com/LN-Zap/lndconnect: No such file or directory

Building github.com/LN-Zap/lndconnect ...
make: *** No targets specified and no makefile found.  Stop.
./97addMobileWalletLNDconnect.sh: line 102: lndconnect: command not found
./97addMobileWalletLNDconnect.sh: line 119: lndconnect: command not found
(To shrink QR code: OSX->CMD- / LINUX-> CTRL-) Press ENTER when finished.
./97addMobileWalletLNDconnect.sh: line 119: lndconnect: command not found
(To shrink QR code: OSX->CMD- / LINUX-> CTRL-) Press ENTER when finished.

fbi: no process found
------------------------------
If the connection was not working:
- Check that your WIFI devices can talk to the LAN devices on your router (deactivate IP isolation or guest mode).
- check issues: https://github.com/LN-Zap/lndconnect/issues
- check issues: https://github.com/rootzoll/raspiblitz/issues

Press ENTER to return to main menu.
--------------------------------------------------------------------------

frennkie, 2019-09-28:
Just tested on RaspiBlitz 1,3. The Go download link is outdated. But after fixing that the build of `lndconnect` is working for me.

ChuckNorrison, 2020-08-14:
i get my qr code but Zap fails with unsupported format if i want to connect to my node url... he suggests lndconnect string, btc pay server etc... dont know whats wrong here. lndconnect seems not to be installed, but i get QR codes

ChuckNorrison, 2020-08-15:
all i have found for now is that the QR never gets my domain name. he always use IP. Tls seems to use domain name.

ChuckNorrison, 2020-08-15:
solved with manual lndconnect string build with domain name than ip. import tls certs on android phone seems a precondition. Needs to be renamed to .crt (.cert is not recognized by android 10 built in file open dialogue!)

ChuckNorrison, 2020-08-16:
After one day, duckdns and letsencrypt failed... no ip update no ssl after some hours :( this duckdns ssl solutions are hell

ChuckNorrison, 2020-08-16:
here are some culprits to get aware of:
- some scripts seems to have insufficient permissions (internet.dyndomain.sh)
- raspiblitz.conf on /mnt/hdd can be tweaked manually, but this step seems to needs a repair of macroons and TLS (correct me if iam wrong at this point)
- LetsEncrypt Setup with DynDNS and DuckDns.org seems a bit tricky. Maybe i failed at the subdomain part, my update url was with full url and token rather than with subdomain name and token. This should be a bit more clearly in setup. Here it was called fill in subdomain of duckdns, at this point the full url needs to be set. After that step we give the update url with only the subdomain name manually as input.

In my case i failed at least one time with update url and after ip change duckdns registered 1.1.1.1 and results with forbidden to users. After some try and errors also the cert gets broken and i need a resetup of all that stuff.

Maybe a fix of the permission denied from my log above will help at this point. 

Back to the root issue: 
- Zap Android App needs a valid TLS certificate on that node
- You can use a static ip and LetsEncrypt or some other solutions, but be aware that all the Lightning Wallet Apps wants a trusted Node URL. I would not recommend to go with the self signed cert at default, that will only work after importing manually to the android cert storage

The recommended Setup with DynDNS/DuckDNS and LetsEncrypt is fine but yet not compatible with the Lightning Wallets, cause they want the DynDNS name and not the ip address, cause the certificate is based on the DynDNS.

QR Code can be copied and fixed manually, that will at the end solve the issue with ZAP on Android. Good luck and maybe this can me more clarified in the future with some more help or example setups in documentations

rootzoll, 2020-08-17:
@ChuckNorrison thanks for the report will look into it for upcomming v1.6.1 service release

rootzoll, 2020-09-22:
> all i have found for now is that the QR never gets my domain name. he always use IP. Tls seems to use domain name.

Tested with the code for the v1.6.1 release and if a dynamicDNS is set then it will be picked up on for Zap on the QRcode.
But beware: If you have a IP2Tor tunnel active for that LND API and a dynDNS set under (SETTINGS) then it will use the IP of the IP2Tor.

> After one day, duckdns and letsencrypt failed... no ip update no ssl after some hours :( this duckdns ssl solutions are hell

the IP2Tor service is still new and we are working on improving it. With v1.6.1 there will be an alternative to duckdns if that is not well working for you.

ChuckNorrison, 2020-09-24:
> the IP2Tor service is still new and we are working on improving it. With v1.6.1 there will be an alternative to duckdns if that is not well working for you.

recently added duckdns to some other projects iam running and i can confirm, its the best free dyndns solution, even the commercial no-ip.com is not as good and is not recognizing my public ip behind a tunneled interface. 

Currently at the blitz iam using a static public IP with a VPN tun0 interface. This is not supported and i will try to not report bugs on that system anymore.

thanks to the development, the blitz is doing well, just a bit hard to get into all the shell scripts to understand how things gets done.

rootzoll, 2020-10-12:
Tested again Zap Mobile for RaspiBlitz v1.6.1 release .. works all OK. Closing this issue.

closed 2020-10-12 by rootzoll

marxenegls, 2021-09-15:
>  no ip update no ssl after some hours :(

noip doesn't like us using letsencrypt generated ssl certs - they offer ssl 'deals' themselves : \ still, i had little to no issues using them, i did though have an impossible time when trying the wildcard ssl option.