raspiblitz/raspiblitz #585: Error with dropbox ln backup

jaramburu, 2019-05-08:
Running v1.2. I ssh in via terminal and exit to the console. I enter nano mnt/hdd/raspiblitz.conf. When I open raspiblitz.conf in nano the file is empty. I pasted the dropboxbackup target with my api key and save it but then I get the error [ Error writing mnt/hdd/raspiblitz.conf: No such file or directory ]

fluidvoice, 2019-05-08:
> I pasted the dropboxbackup target with my api key and save it

what does this mean?
what is the output of running this:
`sudo cat /mnt/hdd/raspiblitz.conf`

jaramburu, 2019-05-09:
Pasted below. I've obscured my ip address, dyn domain and token. 

# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.0'
network=bitcoin
chain=main
hostname=Jmoney
publicIP='my_ip_address'
autoPilot=on
autoNatDiscovery=off
runBehindTor=off
rtlWebinterface=on
dynDomain='my_dyn_domain'
autoUnlock=on
dynUpdateUrl='http://freedns.afraid.org/dynamic/update.php?my_dyn_token'
lndPort='9735'
lndAddress=''

jaramburu, 2019-05-09:
I see the issue now. In the readme it says "For the v1.2 Release this Off-Site Backup options you need to manually editing the raspiblitz config: nano mnt/hdd/raspiblitz.conf"

You're missing a leading / after the nano. Proper command should be nano /mnt/hdd/raspiblitz.conf

closed 2019-05-09 by jaramburu

fluidvoice, 2019-05-09:
> You're missing a leading / after the nano.

That's what I thought and why I had you print it out :)