raspiblitz/raspiblitz #586: LND password error

btc21cluib, 2019-05-08:
LND not running
when I try lncli unlock
I get
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

here the debug link https://termbin.com/432a

fluidvoice, 2019-05-09:
Looks like the wrong wallet password was entered - either manually or via auto-unlock.

btc21cluib, 2019-05-09:
no password I correct - now after now blockchain download I got this error:
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
Here are my debug log:
https://termbin.com/sk3x

fluidvoice, 2019-05-10:
Looks like possible data corruption...
> [ERR] SRVR: unable to create server: unable to read bucket for chain

possibly because your power supply is weak (only 1.2 volts)
I recommend to get a better power supply and re-install as per the docs and FAQ

btc21cluib, 2019-05-11:
So Problem solved after many many useless try. 
The best is to follow the 1) Recover LND data
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md
right now its not finish but looks totally good for me
                 ┌──────── Node is Syncing (136) ──────────┐
                 │  Blockchain Progress :  98.45 %         │  
                 │  Lightning Progress  : prepare sync     │  
 
Thank´s rootzoll for your good work.

closed 2019-05-11 by btc21cluib

fluidvoice, 2019-05-12:
> So Problem solved after many many useless try.
> The best is to follow the 1) Recover LND data
> https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md
> right now its not finish but looks totally good [...]

Did you by chance enable TOR then disabled it later?  Could that be what caused the problem?

btc21cluib, 2019-05-13:
@fluidvoice  I never enable TOR. You think I should try?
My syncronization is now 99.92%, I'll wait for the 100, and it shouldn t work, I'll try.