raspiblitz/raspiblitz #587: lnd.setaddress script "set locale"

cryptomulde, 2019-05-09:
Hello,

in version 1.1 I just manually edited the raspiblitz.conf in my hdd folder to change the ip addresses. Sometimes after reboots I had to manually readjust them.
In the config script there is this "lnd.setaddress.sh" and what I understood it will fix raspiblitz.conf manually to an ip-address of my choice.
So far so good. I tried to set the IP and I receive this error message.

[image]

I tried to fix it with the help of google, but none of those worked and after reboot I am getting the same issues. 
Currently LND is working with custom port and custom IP, but bitcoind always turns back to the automatic IP and does change the one in raspiblitz.conf
Any ideas how to solve this? I want the bitcoind to be reachable under the same IP as my LND.

rootzoll, 2019-05-10:
I will take a look at this for the next release. After reboot RaspiBlitz should only change the "publiIcIP" entry (that is always the public IP of your router) but not the "lndAddress" value, that will be used instead if available.

cryptomulde, 2019-05-12:
Is there any way to manipulate the "publicIP" field with a custom-manual set one?

rootzoll, 2019-05-13:
The value "publicIP" is getting ignored once "lndAddress" is set ... so why manipulate it? Setting "lndAddress" is the way to do that.

cryptomulde, 2019-05-16:
Ok, will wait for the next release then.

rootzoll, 2019-06-28:
I tested `lnd.setaddress.sh`and was able to set a fixed domain or IP and it was displayed on the LCD. 
Also I dont get that perl warning on my machine.
So I mark this issue as "final testing" and may close this on v1.3 release if there is not more input on this. But feel free to open it up again if the v1.3 still having the issue.

closed 2019-07-01 by rootzoll