raspiblitz/raspiblitz #589: Load Dropbox Backup

lebowski36, 2019-05-09:
Hey!

This is my first github post so any suggestions are welcome :)

My combination of shango and raspiblitz with a "failing" payment broke my channel states and I need to load a backup.
I couldn't find any way how to load the backup into the raspiblitz.
Is there a manual?

If you are interested in what exactly the order of events were I can later describe it in more detail.

rootzoll, 2019-05-10:
Which version of RaspiBlitz do you run? If you setup a fresh version of the RaspiBlitz it will ask you now if you want to recover an old wallet and offer you to recover from SEED+ChannelBackupFile. See the latest README for details.

lebowski36, 2019-05-12:
Hi rootzoll, 

thanks for your reply.

The failed channel actually showed the right amount again after a few days. 

But now i got the problem that my lnd dont want to start. The raspi display just show's that lnd is not running.

Here is the log:

https://termbin.com/20uh

Any advice?

Many thanks in advance

rootzoll, 2019-05-13:
The reason is "unable to start server: edge not found" thats an LND error others also searching for a solution on this issue: https://github.com/rootzoll/raspiblitz/issues/562 ... please follow/comment there for a solution.

Closing this issue.

closed 2019-05-13 by rootzoll