raspiblitz/raspiblitz #593: How to recover from a failing hard drive

JohnMcW, 2019-05-10:
Hi - my SD card died and my HDD appears to be failing.  I think I've recovered the important files:
/lnd/lnd.conf
/lnd/data/chain/bitcoin/mainnet/wallet.db
/lnd/data/graph/mainnet/channel.db
/lnd/data/graph/mainnet/sphinxreplay.db

I'm going to reformat the SD card and HDD and start from scratch, but before I do, can please can someone tell me a) if I've got all required files off the HDD, and b) how to I restore them - specifically, at what stage do I need to copy them back to the rebuilt HDD?

Thanks!

lucbran, 2019-05-11:
I had to reinstall blockchain twice and inspte of using the original seed, nothing has been ever recovered..

rootzoll, 2019-05-11:
@lucbran try to recover from seed with the lightning desktop app - not sure why, but it has more chance to recover on-chain dunds from seed: https://github.com/lightninglabs/lightning-app

rootzoll, 2019-05-11:
@JohnMcW try to install  https://github.com/lightninglabs/lightning-app and replace files there in the data dir

JohnMcW, 2019-05-11:
> @JohnMcW try to install https://github.com/lightninglabs/lightning-app and replace files there in the data dir

Hi - I'm not sure how this helps - I just want to reinstall raspiblitz from scratch, then reapply my configuration from my broken install.

CommanderPoe, 2019-05-12:
> Hi - my SD card died and my HDD appears to be failing. I think I've recovered the important files:
> /lnd/lnd.conf
> /lnd/data/chain/bitcoin/mainnet/wallet.db
> /lnd/data/graph/mainnet/channel.db
> /lnd/data/graph/mainnet/sphinxreplay.db

Why not copying the whole lnd folder?

> I'm going to reformat the SD card and HDD and start from scratch, but before I do, can please can someone tell me a) if I've got all required files off the HDD, and b) how to I restore them - specifically, at what stage do I need to copy them back to the rebuilt HDD?

I think you can do this anytime you want, but try waiting till the blockchain is synched then simply replace the folder and make a reboot, on-chain and off chain funds should start appearing systematically.

rootzoll, 2019-05-13:
@JohnMcW simplest way would be in your case to macke a backup of your complete lnd-directory, setup a fresh RaspiBlitz (with same passwords) and then replace that lnd directory.

JohnMcW, 2019-05-20:
Hi - so I rebuilt from scratch (now v1.2), then copied over my lnd folder, chown'd to bitcoin, but now lnd won't start:

https://termbin.com/dk18

Any ideas?

JohnMcW, 2019-05-21:
Some more info.  I ran the software test from the main menu:
https://termbin.com/4viq

I noticed this error:

1558389590 ERROR: [lncli] open /home/bitcoin/.lnd/tls.cert: no such file or directory

But I don't understand as it appears to exist:

admin@austinblitz:/home/bitcoin/.lnd $ ls -lt
total 20
-rw------- 1 bitcoin root  227 May 21 09:51 tls.key
-rw-r--r-- 1 bitcoin root  802 May 21 09:51 tls.cert
-rw-r--r-- 1 bitcoin root  728 May 20 21:24 lnd.conf
lrwxrwxrwx 1 bitcoin root   12 May 20 21:03 lnd -> /mnt/hdd/lnd
drwxr-xr-x 4 bitcoin root 4096 Feb 18 21:24 data
drwxr-xr-x 3 bitcoin root 4096 Feb 18 21:16 logs

I tried regenerating from the main menu (EXPORT -> RESET) but that didn't work either
https://termbin.com/k3aa

fluidvoice, 2019-05-22:
> Some more info. I ran the software test from the main menu:
> https://termbin.com/4viq
>
> I noticed this error:
>
> 1558389590 ERROR: [lncli] open /home/bitcoin/.lnd/tls.cert: no such file or directory [...]

LND is getting a "panic" error.  This is a fatal error that usually indicated LN database corruption.
> panic: misplaced bucket header: 7061796d656e7473

JohnMcW, 2019-05-24:
I managed to restore my files, but suspect there was some corruption as I lost my open channel - not much money fortunately and I did manage to recover the funds that were still in my wallet.  Gonna set up from scratch again and the first thing I'm going to do is set up the new backup functionality.  Closing this.  Thanks!

closed 2019-05-24 by JohnMcW