raspiblitz/raspiblitz #595: LND service is not running

RestartingAdmin, 2019-05-11:
Update to version 1.2 successful, system ran for some days now, no problems
Then LND stopped working, rebooting, re-flashing - no change 
So here is debug output after reflashing and restarting the system: 

https://termbin.com/ox25

RestartingAdmin, 2019-05-11:
On the hardware issues: I use the power supply from hardware list, problems arose with heat, so I dismantled the case and use the raspi without. Temp in acceptable range now.

lebowski36, 2019-05-12:
I got the same problem but for now had no time to reflash the sd. 
I think it should be no problem with temp, because I have a fairly big heat sink plus open setup.

Just updated because of the new backup feature but in the future I will wait for the newer version to settle before updating :)

fluidvoice, 2019-05-12:
> I got the same problem but for now had no time to reflash the sd.
> I think it should be no problem with temp, because I have a fairly big heat sink plus open setup.
> 
> Just updated because of the new backup feature but in the future I will wait for the newer version to settle before updating :)

FYI, "I had the same problem" comments will tend to get ignored.  If you want help open a new issue and provide details and logs.

fluidvoice, 2019-05-12:
> Update to version 1.2 successful, system ran for some days now, no problems
> Then LND stopped working, rebooting, re-flashing - no change
> So here is debug output after reflashing and restarting the system:
> 
> https://termbin.com/ox25

In your log I saw an LND "edge not found" error. This problem I also hit and opened an issue with LND dev's but so far have not discovered the cause.  The current theory is it's *possibly* a data corruption issue be we're not sure.  Please provide links to these logs:
`sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`
`sudo cat /var/log/syslog | nc termbin.com 9999`

PatrickScheich, 2019-05-13:
A quite similar phenomenon here.
Updated from v1.1 to v1.2 (today) but LND never managed to start.

It seems to be caught in a loop. 
...
`2019-05-13 06:51:57.058 [INF] CRTR: FilteredChainView starting`
`2019-05-13 06:51:57.309 [ERR] SRVR: unable to start server: edge not found`

as asked here the two logs
syslog: [https://termbin.com/luj6](url)
lnd.log: [https://termbin.com/441c](url)

rootzoll, 2019-05-13:
@fluidvoice From your experience - does lnd 0.6.1 fixes this "edge not found" error?

rootzoll, 2019-05-13:
@fluidvoice To fix the "edge not found" - have you tried deleting the /mnt/hdd/lnd/data/graph and let LND rebuild it? I think that should be safe, because all wallet/channel data is in another directory.

RestartingAdmin, 2019-05-13:
> > Update to version 1.2 successful, system ran for some days now, no problems
> > Then LND stopped working, rebooting, re-flashing - no change
> > So here is debug output after reflashing and restarting the system:
> > https://termbin.com/ox25
> 
> In your log I saw an LND "edge not found" error. This problem I also hit and opened an issue with LND dev's but so far have not discovered the cause. The current theory is it's _possibly_ a data corruption issue but we're not sure. Please provide links to these logs:
> 
> `sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`

https://termbin.com/xnko

> `sudo cat /var/log/syslog | nc termbin.com 9999`

https://termbin.com/krfd

bjadel, 2019-05-13:
> @fluidvoice To fix the "edge not found" - have you tried deleting the /mnt/hdd/lnd/data/graph and let LND rebuild it? I think that should be safe, because all wallet/channel data is in another directory.
> 
> EDIT: NO STOP - channel.db is that directory.

I have the same error "edge not found". Currently I am using the LND version 0.6.1 and the bug remains. Which file should I delete? I just want to make sure that I don't delete the wrong file. :-)

rootzoll, 2019-05-13:
@bjadel DO NOT DELETE FILES WITHOUT MAKING A BACKUP. By deleting that directory I got LND to run again - showing your on-chain funds, but haveing no access to channel data/funds anymore. This this is not the fix you were looking for.

fluidvoice, 2019-05-14:
> @fluidvoice From your experience - does lnd 0.6.1 fixes this "edge not found" error?

No it does not.

fluidvoice, 2019-05-14:
> I have the same error "edge not found". Currently I am using the LND version 0.6.1 and the bug remains.

@bjadel are you, or were you, running TOR?

bjadel, 2019-05-14:
@fluidvoice Yeah, my node uses TOR. My channel.db had 137 errors. Apparently fixed it with "bbolt compact", but it still doesn't work. Should I deactivate the TOR service?

PatrickScheich, 2019-05-14:
> 
> 
> @fluidvoice To fix the "edge not found" - have you tried deleting the /mnt/hdd/lnd/data/graph and let LND rebuild it? I think that should be safe, because all wallet/channel data is in another directory.
> 
> EDIT: NO STOP - channel.db is that directory.

Well, I had have been too quick , and deleted 
- /mnt/hdd/lnd/data/graph/mainnet/channel.db
- /mnt/hdd/lnd/data/graph/mainnet/sphinxreplay.db
after a reboot, lnd started to rebuild both files (which took around 12 hours)
as the rebuild had finished (nearly) everything works fine..
- bitcoind comes up fine
- lnd comes up fine
except that all my channels are gone (I guess I have to recover them from the seed words)

PatrickScheich, 2019-05-14:
> 
> > @fluidvoice From your experience - does lnd 0.6.1 fixes this "edge not found" error?
> 
> @rootzoll No it does not.
> @PatrickScheich and @RestartingAdmin were you ever running TOR?
> My node was working upon upgrade to v1.2 then stopped some time after activating TOR but could be just a coincidence. Turning off TOR did not fix it.

Yes, I once tried to switch my no to TOR (which worked fine) 
then I switched it back to "non TOR" (which also works fine)
I used the "services" menu from raspiblitz for this

the non TOR version works for about 2..3 weeks, then I updated to v1.2 and lnd got stuck from the first moment

PatrickScheich, 2019-05-14:
> @bjadel DO NOT DELETE FILES WITHOUT MAKING A BACKUP. By deleting that directory I got LND to run again - showing your on-chain funds, but haveing no access to channel data/funds anymore. This this is not the fix you were looking for.

@rootzoll As I had done exactly that "deleting those files without making a backup upfront" I'm wondering if there is a procedure that will "bring back" my open channels? If not, I'm ok with that and will write off the approx. 200k sats in that channels.
So if I'm willing to accept at loss, would it be ok to to carry on with funding the lnd-wallet and creation new channels?

rootzoll, 2019-05-15:
@PatrickScheich you really were quick following my bad advice - I think I had that idea just only for 15min without that "STOP" warning. Do you still have the channel.backup file? Just from seed there seems not to be chance to recover those channels.

fluidvoice, 2019-05-15:
> Yes, I once tried to switch my no to TOR (which worked fine)
> then I switched it back to "non TOR" (which also works fine)
> I used the "services" menu from raspiblitz for this
>
> the non TOR version [...]

My experience was the same except I upgraded to v1.2 first. I was working. Then turned on TOR. Still working.  Then turned off TOR and some time later it stopped working with the "edge not found" error in the log.

fluidvoice, 2019-05-15:
> Do you still have the channel.backup file?

I managed to create a channel.backup file.  The problem is I cannot, so far, find where I stashed my seed phrase. So no way to recover my sat's without that right?

rootzoll, 2019-05-15:
@fluidvoice your channel.backup file is encrypted with your private key from your seed phrase. So yes - you will need the seed phrase to make use of the channel.backup file.

PatrickScheich, 2019-05-16:
> @PatrickScheich you really were quick following my bad advice - I think I had that idea just only for 15min without that "STOP" warning. 

Yes, I'm able of a very precise timing when it come to such things

> Do you still have the channel.backup file? Just from seed there seems not to be chance to recover those channels.

sure there are channel.backup files at

- path:/mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup         
  size:45 bytes           
  last accesstime:Wed May 15 14:45:38.5320188400 2019
  
- path:/home/admin/.lnd/data/chain/bitcoin/mainnet/channel.backup     
  size:45 bytes           
  last accesstime:Wed May 15 14:45:48.4420175350 2019
  

but may they have been overwritten by the recreation of channel.db?
nevertheless I will give it a try, can you point me to the HowTo for the channel recovery procedure

PatrickScheich, 2019-05-18:
> 
> 
> @fluidvoice your channel.backup file is encrypted with your private key from your seed phrase. So yes - you will need the seed phrase to make use of the channel.backup file.

@rootzoll is there a way to get the content of channel.backup in "clear text", to see if a given channel.backup contain a particular set of channels?

fluidvoice, 2019-05-18:
> @rootzoll is there a way to get the content of channel.backup in "clear text", to see if a given channel.backup contain a particular set of channels?

see the LND command `lncli --help` output and the `exportchanbackup --all` and `verifychanbackup` flags for that command.
Otherwise I'd ask this question on a LND forum.

fluidvoice, 2019-05-19:
I think there is around 7 nodes with this "edge not found" error where LND keeps stopping.
How many of you guys are running with TOR turned on, or once had it turned on, but now it's off?

Z-and, 2019-05-22:
> @fluidvoice your channel.backup file is encrypted with your private key from your seed phrase. So yes - you will need the seed phrase to make use of the channel.backup file.

Hi rootzoll,

I seem to be one of the many also experiencing this "edge not found" problem and no longer am able to run certain commands on my node. However, I still have my backup seed and would like to somehow retrieve my funds (both onchain and offchain). What would be the best way to do this? Do I need to attempt to create the channel.backup file or was this automatically done when I upgraded to v1.2? Any help would be much appreciate since I have over 1+ BTC at stake >.>

Z-and, 2019-05-22:
Hi @rootzoll ,

I seem to be one of the many also experiencing this "edge not found" problem and no longer am able to run certain commands on my node. However, I still have my backup seed and would like to somehow retrieve my funds (both onchain and offchain). What would be the best way to do this? Do I need to attempt to create the channel.backup file or was this automatically done when I upgraded to v1.2? Any help would be much appreciate since I have over 1+ BTC at stake >.>

fluidvoice, 2019-05-23:
verify if you have a channel.backup file or not:
`sudo ls -al /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/channel.backup`

open one SSH connection to the node to monitor what LND is doing while you run commands, run:
`sudo tail -f /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log`

open another SSH connection run LND commands:
`lncli --help` to get a list of commands and help on running them.

run lncli command to create the channel.backup (I don't remember the exact syntax of the command but it's listed in the --help output) then save the channel.backup to a USB flash drive for safe keeping.

fluidvoice, 2019-05-23:
@Z-and had you been running TOR on the node?

openoms, 2019-05-24:
more docs on channel. backuo https://github.com/lightningnetwork/lnd/pull/2313

syntax:
`lncli --network=mainnet exportchanbackup --all --output_file=channels.backup`

@Z-and
How many channels do you have? 
(all will be force closed when restoring from the SCB) 
Are there significant funds in the channels? 
If not you can attempt to restore without the SCB only from the mnemonic seed with the desktop Lightning App. That gives you access to your on-chain funds. 
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#2-recover-from-wallet-seed

If you know your peers you can contact them to force close the channels, that way the funds are returned on-chain (unless they cheat).

openoms, 2019-05-24:
@fluidvoice you mentioned that patch with which you could ignore the "edge not found" error when building from source.
That could be also attempted to use the existing LND dir, especially if there is no channel.backup or @Z-and wants to avoid force closing his channels.

fluidvoice, 2019-05-24:
@openoms now that I have Raspiblitz running on x86 I'm hoping the LND dev's will be more active in giving me patches to try out to debug this problem: https://github.com/lightningnetwork/lnd/issues/3025#issuecomment-495587663

Z-and, 2019-05-24:
@fluidvoice No I wasn’t using Tor at any point

@opennoms I have 25 channels open and around 1 BTC funded throughout these channels, another .19 BTC on the onchain wallet. 

I was able to retrieve the channel.backup file so I don’t think it’s necessary to try and generate a new one. I also have the backup seed. I’m not sure if I should just wait for a work around on the “edge not found” error or attempt to restore from backup. I’d prefer to take the safest route possible.

openoms, 2019-05-25:
@fluidvoice it would worth a try trying to build from source with the patch

@Z-and
You are not in a bad position having both the seed and the channel backup. I would act rather soon either restoring (resulting to force closes) or trying to build LND from source with the patch (do you need help with that? - clue: the source build script is in the build_sdcard.sh, just commented out) 
Your peers can can cheat you by attempting to close with an old state while you are offline.

Z-and, 2019-05-25:
> @fluidvoice it would worth a try trying to build from source with the patch
>
> @Z-and
> You are not in a bad position having both the seed and the channel backup. I would act rather soon either [...]

Yeah I would definitely need some help if I were to build LND from source. What do you think is the safer route? Building from source or setting up a new node with mnemonic seed/channel.backup? I have an extra Pi setup so I can unplug my existing one and setup a new one.

openoms, 2019-05-26:
I think both options are fairly safe in terms of getting your funds.
Restoring from seed + SBC will result in force closing all the channels.
Trying the LND version built with a patch (ignoring the edge not found error) would not introduce any new risk, but might not work (ignoring the error might not be enough). If it works you might be able to keep your channels open and regain control over them.

I am putting together a helper script which will update your LND with the version built form source+ patch. Just need to test it first.

openoms, 2019-05-26:
Build from the source of my fork where the ignore "edge not found" error is commited.(https://github.com/openoms/lnd/commit/1a59596df3e8ae9e95a858cda33c329d4736d1bd).

Have a look at the script: 
https://github.com/openoms/bitcoin-tutorials/blob/master/lnd.from.source.patched.sh

Only need to run this line on the RaspiBlitz:
`wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.from.source.patched.sh && sudo bash lnd.from.source.patched.sh`

If successful should get this message on the end:
[image]

Fingers crossed it gets your channels back!

fluidvoice, 2019-05-26:
> Build from the source of my fork where the patch to ignore the "edge not found" error is commited.

Nice thanks for doing that. Will try it on my Atomic-Pi.  My VM setup keeps breaking due to UUID error when the external HDD /dev/sdx changes - internal details of how the .vmdk is handled.

openoms, 2019-05-27:
> > Build from the source of my fork where the patch to ignore the "edge not found" error is commited.
> 
> Nice thanks for doing that. Will try it on my Atomic-Pi. Wait, this will work for Arm only or also x86?
> Will try it. I built a patch before for Arm but did not lock the steps down in a script.
> My VM setup keeps breaking due to UUID error when the external HDD /dev/sdx changes - internal details of how the .vmdk is handled.

I have tested it in the VM so will work on X86. Building from source is the same on all platforms. Go ahead and try it. It has similar affect to the update script to lnd v0.6.1 had which is in short: stop lnd, update, start .

fluidvoice, 2019-05-28:
> I have tested it in the VM so will work on X86. Building from source is the same on all platforms. Go ahead and try it. It has similar affect to the update script to lnd v0.6.1 had which is in [...]

I keep breaking my VM setup. Need get it working then stop messing with it.  Atm, I cannot SSH into it. Getting "connection refused" and cannot figure out why.  Plus with the VM I cannot connect a keyboard and go to a terminal prompt with CTRL+ALT+F2 like you can with a physical device.

openoms, 2019-05-28:
@fluidvoice please share the VM specific problems here: https://github.com/openoms/raspiblitz/issues/46

bjadel, 2019-05-28:
@openoms thanks for the script
I got the following error:
`channeldb/graph.go:3029:6: syntax error: unexpected return, expecting expression
`

openoms, 2019-05-28:
>Maybe a go version problem?
/usr/local/go/bin/go version go version go1.11 linux/arm

That is possible. I am running on Go  v1.12.5. Coming with the the next v1.3 RaspiBlitz version or manual SDcard build.

fluidvoice, 2019-05-30:
> sudo bash lnd.from.source.patched.sh

It didn't seem to work, so I tried it by hand:
$ `go get -d github.com/openoms/lnd`
# cd .; git clone https://github.com/openoms/lnd /usr/local/gocode/src/github.com/openoms/lnd
fatal: could not create work tree dir '/usr/local/gocode/src/github.com/openoms/lnd': **Permission denied**
package github.com/openoms/lnd: exit status 128
$ `sudo chown -R admin:admin /usr/local/gocode/src/`
$ `go get -d github.com/openoms/lnd`
# cd .; git clone https://github.com/btcsuite/btclog /usr/local/gocode/src/github.com/btcsuite/btclog
Cloning into '/usr/local/gocode/src/github.com/btcsuite/btclog'...
fatal: unable to access 'https://github.com/btcsuite/btclog/': Failed to connect to github.com port 443: Connection timed out
package github.com/btcsuite/btclog: exit status 128
# cd .; git clone https://github.com/btcsuite/websocket /usr/local/gocode/src/github.com/btcsuite/websocket
Cloning into '/usr/local/gocode/src/github.com/btcsuite/websocket'...
fatal: unable to access 'https://github.com/btcsuite/websocket/': Failed to connect to github.com port 443: Connection timed out
package github.com/btcsuite/websocket: exit status 128
# cd .; git clone https://github.com/btcsuite/btcwallet /usr/local/gocode/src/github.com/btcsuite/btcwallet
Cloning into '/usr/local/gocode/src/github.com/btcsuite/btcwallet'...
fatal: unable to access 'https://github.com/btcsuite/btcwallet/': Failed to connect to github.com port 443: Connection timed out
package github.com/btcsuite/btcwallet/chain: exit status 128
package github.com/btcsuite/btcwallet/waddrmgr: cannot find package "github.com/btcsuite/btcwallet/waddrmgr" in any of:
	/usr/local/go/src/github.com/btcsuite/btcwallet/waddrmgr (from $GOROOT)
	/usr/local/gocode/src/github.com/btcsuite/btcwallet/waddrmgr (from $GOPATH)
package github.com/btcsuite/btcwallet/wallet: cannot find package "github.com/btcsuite/btcwallet/wallet" in any of:
	/usr/local/go/src/github.com/btcsuite/btcwallet/wallet (from $GOROOT)
	/usr/local/gocode/src/github.com/btcsuite/btcwallet/wallet (from $GOPATH)
package github.com/btcsuite/btcwallet/wallet/txauthor: cannot find package "github.com/btcsuite/btcwallet/wallet/txauthor" in any of:
	/usr/local/go/src/github.com/btcsuite/btcwallet/wallet/txauthor (from $GOROOT)
	/usr/local/gocode/src/github.com/btcsuite/btcwallet/wallet/txauthor (from $GOPATH)
package github.com/btcsuite/btcwallet/walletdb: cannot find package "github.com/btcsuite/btcwallet/walletdb" in any of:
	/usr/local/go/src/github.com/btcsuite/btcwallet/walletdb (from $GOROOT)
	/usr/local/gocode/src/github.com/btcsuite/btcwallet/walletdb (from $GOPATH)
$ `sudo make build`
/bin/sh: 1: go: not found
/bin/sh: 1: go: not found
 Building debug lnd and lncli.
GO111MODULE=on go build -v -tags="dev" -o lnd-debug -ldflags "-X github.com/lightningnetwork/lnd/build.Commit=v0.6.1-beta-99-g1a59596df3e8ae9e95a858cda33c329d4736d1bd" github.com/lightningnetwork/lnd/cmd/lnd
/bin/sh: 1: go: not found
Makefile:94: recipe for target 'build' failed
make: *** [build] Error 127

admin@atomicpi:/usr/local/gocode/src/github.com/openoms/lnd$ `/usr/local/go/bin/go build -v -tags="dev" -o lnd-debug -ldflags "-X github.com/lightningnetwork/lnd/build.Commit=v0.6.1-beta-99-g1a59596df3e8ae9e95a858cda33c329d4736d1bd" github.com/lightningnetwork/lnd/cmd/lnd`
go: finding github.com/golang/protobuf v1.3.1
go: finding github.com/btcsuite/btcd v0.0.0-20190426011420-63f50db2f70a
go: finding github.com/coreos/bbolt v1.3.2
go: finding github.com/NebulousLabs/go-upnp v0.0.0-20180202185039-29b680b06c82
go: finding github.com/NebulousLabs/fastrand v0.0.0-20180208210444-3cf7173006a0
go: finding github.com/rogpeppe/fastuuid v0.0.0-20150106093220-6724a57986af
go: finding github.com/davecgh/go-spew v1.1.1
go: finding github.com/btcsuite/btcwallet v0.0.0-20190424224017-9d95f76e99a7
go: finding github.com/btcsuite/btclog v0.0.0-20170628155309-84c8d2346e9f
go: finding git.schwanenlied.me/yawning/bsaes.git v0.0.0-20180720073208-c0276d75487e
go: finding github.com/juju/utils v0.0.0-20180820210520-bf9cc5bdd62d
go: finding github.com/kkdai/bstream v0.0.0-20181106074824-b3251f7901ec
.... build contiues... succeed? TBD

fluidvoice, 2019-05-30:
> UPDATE: It's working now. I'll write more later.

how?

fluidvoice, 2019-05-30:
tried a 2nd time, still errors:

$ `sudo bash lnd.from.source.patched.sh`
*** Build LND from Source ***
repo=github.com/openoms/lnd
up to the commit=1a59596df3e8ae9e95a858cda33c329d4736d1bd
# cd .; git clone https://github.com/davecgh/go-spew /usr/local/gocode/src/github.com/davecgh/go-spew
Cloning into '/usr/local/gocode/src/github.com/davecgh/go-spew'...
fatal: unable to access 'https://github.com/davecgh/go-spew/': Failed to connect to github.com port 443: Connection timed out
package github.com/davecgh/go-spew/spew: exit status 128

openoms, 2019-05-30:
> tried a 2nd time, still errors:
>
> $ `sudo bash lnd.from.source.patched.sh`
> *** Build LND from Source ***
> repo=github.com/openoms/lnd
> up to the commit=1a59596df3e8ae9e95a858cda33c329d4736d1bd
> #. cd .; [...]

you clearly have internet connection issues.
If you are in a VM try setting your network adapter to NAT. 
Port forward for ssh (set up on the host machine):
$ VBoxManage modifyvm <VM-name-here> --natpf1 "ssh,tcp,,3022,,22"
Connect from a host terminal:
$ ssh admin@127.0.0.1 -p 3022

You can build update lnd before connecting a HDD to avoid the problems with .vmdk.

fluidvoice, 2019-05-30:
> If you are in a VM try setting your network adapter to NAT.
> Port forward for ssh (set up on the host machine):
> $ VBoxManage modifyvm --natpf1 "ssh,tcp,,3022,,22"
> Connect from a host terminal:
> $ ssh admin@127.0.0.1 -p 3022
> 
> You can update lnd before connecting a HDD to avoid the problems with .vmdk.

I'm using the Atomic Pi.

fluidvoice, 2019-05-30:
Trying again with explicit/manual command pulled from the Makefile.
Clearly the wifi network connection is working sufficiently - for at least part of the build:

**admin@atomicpi:/usr/local/gocode/src/github.com/openoms/lnd$** `sudo GO111MODULE=on /usr/local/go/bin/go build -v -tags="dev-patch-edgenotfound" -o lnd-debug -ldflags "-X github.com/lightningnetwork/lnd/build.Commit=v0.6.1-beta-99-g1a59596df3e8ae9e95a858cda33c329d4736d1bd" github.com/lightningnetwork/lnd/cmd/lnd`
go: finding github.com/NebulousLabs/fastrand v0.0.0-20180208210444-3cf7173006a0
go: finding github.com/ltcsuite/ltcd v0.0.0-20190101042124-f37f8bf35796
go: finding github.com/btcsuite/btcutil v0.0.0-20190425235716-9e5f4b9a998d
go: finding github.com/btcsuite/btcwallet v0.0.0-20190424224017-9d95f76e99a7
go: finding github.com/Yawning/aez v0.0.0-20180114000226-4dad034d9db2
Fetching https://golang.org/x/crypto?go-get=1
go: finding github.com/davecgh/go-spew v0.0.0-20171005155431-ecdeabc65495
go: finding github.com/btcsuite/btcd v0.0.0-20181130015935-7d2daa5bfef2
Fetching https://google.golang.org/grpc?go-get=1
go: finding github.com/coreos/bbolt v1.3.2
Parsing meta tags from https://golang.org/x/crypto?go-get=1 (status code 200)
get "golang.org/x/crypto": found meta tag get.metaImport{Prefix:"golang.org/x/crypto", VCS:"git", RepoRoot:"https://go.googlesource.com/crypto"} at https://golang.org/x/crypto?go-get=1
Parsing meta tags from https://google.golang.org/grpc?go-get=1 (status code 200)
go: finding golang.org/x/crypto v0.0.0-20190103213133-ff983b9c42bc
get "google.golang.org/grpc": found meta tag get.metaImport{Prefix:"google.golang.org/grpc", VCS:"git", RepoRoot:"https://github.com/grpc/grpc-go"} at https://google.golang.org/grpc?go-get=1
go: finding google.golang.org/grpc v1.18.0
go: finding github.com/kr/pretty v0.1.0
go: finding github.com/juju/clock v0.0.0-20180808021310-bab88fc67299
go: finding github.com/btcsuite/btcutil v0.0.0-20180706230648-ab6388e0c60a
Fetching https://golang.org/x/exp?go-get=1
Parsing meta tags from https://golang.org/x/exp?go-get=1 (status code 200)
get "golang.org/x/exp": found meta tag get.metaImport{Prefix:"golang.org/x/exp", VCS:"git", RepoRoot:"https://go.googlesource.com/exp"} at https://golang.org/x/exp?go-get=1
go: finding golang.org/x/exp v0.0.0-20190121172915-509febef88a4
Fetching https://golang.org/x/lint?go-get=1
Parsing meta tags from https://golang.org/x/lint?go-get=1 (status code 200)
get "golang.org/x/lint": found meta tag get.metaImport{Prefix:"golang.org/x/lint", VCS:"git", RepoRoot:"https://go.googlesource.com/lint"} at https://golang.org/x/lint?go-get=1
go: finding golang.org/x/lint v0.0.0-20180702182130-06c8688daad7
go: finding github.com/lightninglabs/gozmq v0.0.0-20180324010646-462a8a753885
go: finding github.com/btcsuite/websocket v0.0.0-20150119174127-31079b680792

fluidvoice, 2019-05-30:
will see if the errors always happen in the same place?:
$ `grep Failed errors.txt`

> 	fatal: unable to access 'https://github.com/davecgh/go-spew/': Failed to connect to github.com port 443: Connection timed out
> 	fatal: unable to access 'https://github.com/btcsuite/btcutil/': Failed to connect to github.com port 443: Connection timed out
> 	fatal: unable to access 'https://github.com/ltcsuite/ltcd/': Failed to connect to github.com port 443: Connection timed out
> 	fatal: unable to access 'https://github.com/kr/pretty/': Failed to connect to github.com port 443: Connection timed out
> 	fatal: unable to access 'https://github.com/rogpeppe/fastuuid/': Failed to connect to github.com port 443: Connection timed out
>

fluidvoice, 2019-05-30:
To remove potential wifi problems of the Atomic-Pi, tried **building on my notebook** (though still WiFi as it does not have ethernet) with just env changes to the build script.  **It still gets "port 443: Connection timed out" error.**

> #. export GOROOT=/usr/local/go
> export GOROOT=/usr/bin
> #. export PATH=$PATH:$GOROOT/bin
> #. export GOPATH=/usr/local/gocode
> export GOPATH=/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found
> export PATH=$PATH:$GOPATH/bin
> 

*** Build LND from Source ***
repo=github.com/openoms/lnd
up to the commit=1a59596df3e8ae9e95a858cda33c329d4736d1bd
package bytes: unrecognized import path "bytes" (import path does not begin with hostname)
package container/list: unrecognized import path "container/list" (import path does not begin with hostname)
package context: unrecognized import path "context" (import path does not begin with hostname)
package crypto/ecdsa: unrecognized import path "crypto/ecdsa" (import path does not begin with hostname)
package crypto/elliptic: unrecognized import path "crypto/elliptic" (import path does not begin with hostname)
package crypto/rand: unrecognized import path "crypto/rand" (import path does not begin with hostname)
package crypto/sha256: unrecognized import path "crypto/sha256" (import path does not begin with hostname)
package crypto/tls: unrecognized import path "crypto/tls" (import path does not begin with hostname)
package crypto/x509: unrecognized import path "crypto/x509" (import path does not begin with hostname)
package crypto/x509/pkix: unrecognized import path "crypto/x509/pkix" (import path does not begin with hostname)
package encoding/binary: unrecognized import path "encoding/binary" (import path does not begin with hostname)
package encoding/hex: unrecognized import path "encoding/hex" (import path does not begin with hostname)
package encoding/pem: unrecognized import path "encoding/pem" (import path does not begin with hostname)
package errors: unrecognized import path "errors" (import path does not begin with hostname)
package fmt: unrecognized import path "fmt" (import path does not begin with hostname)
# cd .; git clone https://github.com/btcsuite/btcd /home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd
Cloning into '/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd'...
fatal: unable to access 'https://github.com/btcsuite/btcd/': Failed to connect to github.com port 443: Connection timed out
package github.com/btcsuite/btcd/blockchain: exit status 128
package github.com/btcsuite/btcd/btcec: cannot find package "github.com/btcsuite/btcd/btcec" in any of:
	/usr/bin/src/github.com/btcsuite/btcd/btcec (from $GOROOT)
	/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd/btcec (from $GOPATH)
package github.com/btcsuite/btcd/chaincfg: cannot find package "github.com/btcsuite/btcd/chaincfg" in any of:
	/usr/bin/src/github.com/btcsuite/btcd/chaincfg (from $GOROOT)
	/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd/chaincfg (from $GOPATH)
package github.com/btcsuite/btcd/chaincfg/chainhash: cannot find package "github.com/btcsuite/btcd/chaincfg/chainhash" in any of:
	/usr/bin/src/github.com/btcsuite/btcd/chaincfg/chainhash (from $GOROOT)
	/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd/chaincfg/chainhash (from $GOPATH)
package github.com/btcsuite/btcd/connmgr: cannot find package "github.com/btcsuite/btcd/connmgr" in any of:
	/usr/bin/src/github.com/btcsuite/btcd/connmgr (from $GOROOT)
	/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd/connmgr (from $GOPATH)
package github.com/btcsuite/btcd/rpcclient: cannot find package "github.com/btcsuite/btcd/rpcclient" in any of:
	/usr/bin/src/github.com/btcsuite/btcd/rpcclient (from $GOROOT)
	/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd/rpcclient (from $GOPATH)
package github.com/btcsuite/btcd/txscript: cannot find package "github.com/btcsuite/btcd/txscript" in any of:
	/usr/bin/src/github.com/btcsuite/btcd/txscript (from $GOROOT)
	/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd/txscript (from $GOPATH)
package github.com/btcsuite/btcd/wire: cannot find package "github.com/btcsuite/btcd/wire" in any of:
	/usr/bin/src/github.com/btcsuite/btcd/wire (from $GOROOT)
	/home/brad/working/raspiblitz/dev/lnd/patches/edge-not-found/src/github.com/btcsuite/btcd/wire (from $GOPATH)

openoms, 2019-05-30:
Still works on my X86 VM Blitz: https://pastebin.com/r5CSU8yQ

fluidvoice, 2019-05-30:
> I have tested it in the VM so will work on X86. Building from source is the same on all platforms. Go ahead and try it. It has similar affect to the update script to lnd v0.6.1 had which is in [...]

> I am putting together a helper script which will update your LND with the version built form source+ patch. Just need to test it first.

@openoms you tested running the script, but did you actually run the resulting binaries?  I don't even see where they are. It seems it built only `*.go` files?  Where are the `lnd` and `lncli` binaries?

Z-and, 2019-05-30:
> @openoms thanks for the script
> I got the following error:
> `channeldb/graph.go:3029:6: syntax error: unexpected return, expecting expression `
> Maybe a go version problem?
> `/usr/local/go/bin/go version go version go1.11 linux/arm `
> 
> UPDATE: It's working now. I'll write more later.

Please share when you get a chance, I'd like to give this a shot.

fluidvoice, 2019-06-01:
> Please share when you get a chance, I'd like to give this a shot.

update on this problem: https://github.com/rootzoll/raspiblitz/issues/562#issuecomment-497939315
tl;dr fix is to avoid the problem but does not fix nodes already with this error.  Running the above patch might work or not. It would be nice to try if anyone can get it to build.  Otherwise reinstall/restore is in order.

openoms, 2019-06-02:
> Build from the source of my fork where the patch to ignore the "edge not found" error is commited.([openoms/lnd@1a59596](https://github.com/openoms/lnd/commit/1a59596df3e8ae9e95a858cda33c329d4736d1bd)).
> 
> Have a look at the script:
> https://github.com/openoms/bitcoin-tutorials/blob/master/lnd.from.source.patched.sh
> 
> To be safe as possible make a backup of the lnd directory (https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-backup-my-lightning-node)
> run:
> `/home/admin/config.scripts/lnd.rescue.sh backup`
> 
> To update lnd with the patch run this line on the RaspiBlitz:
> `wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.from.source.patched.sh && sudo bash lnd.from.source.patched.sh`
> 
> If successful should get this message on the end:
> [image]
> 
> Fingers crossed it gets your channels back!

I have modified the patch, please test again. @fluidvoice @Z-and @bjadel 
Renamed the file to avoid conflict, new command to run on your RaspiBlitz:

`$ wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.from.source.patched.corr.sh && sudo bash lnd.from.source.patched.corr.sh`

Output, which is looking better now:
https://pastebin.com/raw/xzqU7jYE

openoms, 2019-06-03:
TL;DR: the patch does not help.

Got the patched lnd running on a node with the error:
[image]

Still fails due to `[ERR] SRVR: unable to start server: edge not found`
```
2019-06-02 19:15:18.284 [INF] BRAR: Starting contract observer, watching for breaches.
2019-06-02 19:15:18.284 [INF] DISC: Authenticated Gossiper is starting
2019-06-02 19:15:18.284 [INF] NTFN: New block epoch subscription
2019-06-02 19:15:18.616 [INF] CRTR: FilteredChainView starting

2019-06-02 19:15:20.489 [ERR] SRVR: unable to start server: edge not found

2019-06-02 19:15:20.489 [INF] RPCS: Stopping RPC Server
2019-06-02 19:15:20.490 [INF] RPCS: Stopping SignRPC Sub-RPC Server
2019-06-02 19:15:20.490 [INF] RPCS: Stopping ChainRPC Sub-RPC Server
2019-06-02 19:15:20.490 [INF] RPCS: Stopping InvoicesRPC Sub-RPC Server
2019-06-02 19:15:20.490 [INF] RPCS: Stopping WalletKitRPC Sub-RPC Server
2019-06-02 19:15:20.500 [INF] LTND: Shutdown complete
```

Ignoring the error did not help, this might suggest that the database is corrupted irreversibly.

Will proceed with restoring from the seed and channel.backup file on this node (https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#2-recover-from-wallet-seed)

There is fix to prevent this happening already in the lnd source, coming in the next release: https://github.com/rootzoll/raspiblitz/issues/638

fluidvoice, 2019-06-03:
pretty much what I thought. Dev thought it was a long shot too.

openoms, 2019-06-03:
@fluidvoice No, it is from one of the already known nodes. Not adding to this number luckily.

jodobear, 2019-06-13:
here's another node struck by `edge not found`
logs:  https://termbin.6bn36/com
lng.log: https://termbin.com/qlx7
syslog: https://termbin.com/7tu8

> TL;DR: the patch does not help.
> 
> Got the patched lnd running on a node with the error:
> [image]
> 
> Still fails due to `[ERR] SRVR: unable to start server: edge not found`
> 
> ```
> 2019-06-02 19:15:18.284 [INF] BRAR: Starting contract observer, watching for breaches.
> 2019-06-02 19:15:18.284 [INF] DISC: Authenticated Gossiper is starting
> 2019-06-02 19:15:18.284 [INF] NTFN: New block epoch subscription
> 2019-06-02 19:15:18.616 [INF] CRTR: FilteredChainView starting
> 
> 2019-06-02 19:15:20.489 [ERR] SRVR: unable to start server: edge not found
> 
> 2019-06-02 19:15:20.489 [INF] RPCS: Stopping RPC Server
> 2019-06-02 19:15:20.490 [INF] RPCS: Stopping SignRPC Sub-RPC Server
> 2019-06-02 19:15:20.490 [INF] RPCS: Stopping ChainRPC Sub-RPC Server
> 2019-06-02 19:15:20.490 [INF] RPCS: Stopping InvoicesRPC Sub-RPC Server
> 2019-06-02 19:15:20.490 [INF] RPCS: Stopping WalletKitRPC Sub-RPC Server
> 2019-06-02 19:15:20.500 [INF] LTND: Shutdown complete
> ```
> 
> Ignoring the error did not help, this might suggest that the database is corrupted irreversibly.
> 
> Will proceed with restoring from the seed and channel.backup file on that node (https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#2-recover-from-wallet-seed)
> 
> There is a fix to prevent this happening already in the lnd source, coming in the next release: #638

So, we don't have a solution for this yet. And even if rebuilt from source it will persist. So, we can only wait until it's resolved? Am i missing something here?

And thanks @openoms @fluidvoice

openoms, 2019-06-14:
>So, we don't have a solution for this yet. And even if rebuilt from source it will persist. So, we can only wait until it's resolved? Am i missing something here?

Unfortunately from my experience this error means that there is database corruption, which we were unable to fix by ignoring the error.
It seems to happen when LND is not shut down correctly. Could be caused by a power outage or hard reset without stopping services (I am speculating here). 

The new version of LND (built up from source to a recent commit: https://github.com/rootzoll/raspiblitz/issues/638#issuecomment-498135968) is fixing the problem by preventing it to occur, but cannot fix the faulty nodes.

The way to go to my best knowledge is to restore the lightning wallet from seed+SCB: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#2-recover-from-wallet-seed
The onchain funds tend not to reappear on the RaspiBlitz, in this case you can use the Desktop Lightning App.
Also there is a problem if there was a passphrase used on top of the seed as the Lightning App is not compatible with that yet. See the issue: https://github.com/rootzoll/raspiblitz/issues/632

[16 more replies]

closed 2021-05-17 by rootzoll