raspiblitz/raspiblitz #596: Error reading from database: Fatal LevelDB error: Corruption: block checksum mismatch:

bitsam, 2019-05-12:
Hi All, I have been searching through the existing issues, as I know this has happened many times. Some time ago I scrapped my chain data and re-downloaded it again. I recall using a script, or an option from the mainmenu that stepped me through the process. Like, do you want to delete the chain data.....

I am running v1.1 and have been offline for a few days trying to re-sync the blockchain. basically, I want to delete the chaindata, and re-download it.

I have closed all my channels and was paid recieved about 50% of my funds. However 50% of my funds were in Pending closed channel waiting for the blocks to mature. That maturity time has now passed, but my node has been offline when the blocks matured. 

So my two questions are:
1) is there a script to delete the chain-data and re-download the blockchain (either by local copy or torrent) ?
2) If I update from v1.1 to v1.2 while I have channels in "Pending force close" state, will I be able to recover my funds on v1.2 ? or should I try and reciver my node as v1.1 and makesure I get the payment from the closed channels first before updating to v1.2?

fluidvoice, 2019-05-12:
It's almost always better to be running the latest code.  You should follow the doc's and FAQ to see what is the recommended backup and upgrade procedure.

rootzoll, 2019-05-13:
> 1. is there a script to delete the chain-data and re-download the blockchain

You can copy the blockchain again from another computer with calling `/home/admin/50copyHDD.sh` - it will ask you if you want to delete the old blockchain data.

> If I update from v1.1 to v1.2 while I have channels in "Pending force close" state, will I be able to recover my funds on v1.2?

Once a pending force close is activated - it should be now problem to update the RaspiBlitz in between.

Closing this issue.

closed 2019-05-13 by rootzoll

YuckFouBTC, 2023-10-26:
This sounds similar to my issue https://github.com/raspiblitz/raspiblitz/issues/4230, and the 50copyHDD.sh script could have helped.  However, that script is gone...

admin@192.168.1.247:~ ₿ /home/admin/50copyHDD.sh
-bash: /home/admin/50copyHDD.sh: No such file or directory
admin@192.168.1.247:~ ₿ locate 50copyHDD.sh
admin@192.168.1.247:~ ₿ locate copyHDD.sh
admin@192.168.1.247:~ ₿