raspiblitz/raspiblitz #599: LND service failure

StyriaTrading, 2019-05-12:
I have following problem which occured days after I started with the node:

https://termbin.com/iid9

LND worked fine for several days, I also activated TOR in the advanced settings and everything was fine, PSU plug out plug in no problem, Pi was starting normally.

Then I deactivated TOR in the advanced options and this resulted in this error, tried to activate it again, but getting this error again and again.

StyriaTrading, 2019-05-12:
It writes also this at startup:

* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

2 RESTARTS DETECTED - LND might be in a error loop
err='bitcoin.conf: missing rpcuser (needs to be same as set in lnd.conf)'
err='bitcoin.conf: missing rpcpassword (needs to be same as set in lnd.conf)'
More Error Detail:
err='bitcoin.conf: missing rpcpassword (needs to be same as set in lnd.conf)'

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

fluidvoice, 2019-05-12:
The main problem is your bitcoin.conf file is missing entries.  
secondarily, your power supply is weak - I would replace it with something better.  See the shopping lists and/or search the past github issues, I've already given various PS recommendations.

run this and paste the output:
`sudo cat /mnt/hdd/bitcoin/bicoin.conf`

StyriaTrading, 2019-05-13:
admin@RaspiBlitz:~ $ sudo cat /mnt/hdd/bitcoin/bicoin.conf
cat: /mnt/hdd/bitcoin/bicoin.conf: No such file or directory

StyriaTrading, 2019-05-13:
I have no clue why the config-file is missing. I changed nothing, the SD-card was always mounted in the Pi. 

So this would eventually mean, I have to reflash the card with Raspiblitz - will the HDD be reformatted completely or will the blockchainfiles remain, so that the download has not to be done again?

rootzoll, 2019-05-13:
Similar errors with the config files on TOR on/off were reported in this issue:
https://github.com/rootzoll/raspiblitz/issues/564

Its marked for fixing on next release.

closed 2019-05-20 by StyriaTrading