raspiblitz/raspiblitz #601: Enhancement: Using the RaspiBlitz remotely with ZeroTier

openoms, 2019-05-13:
Ketominer`s guide works great on the RaspiBlitz as well:
https://medium.com/@ketominer/using-nodl-remotely-with-zerotier-a9a17cbb48cf

The only change is that need to use sudo:

rootzoll, 2019-05-13:
I think this could be interesting - but I am a bit skeptical. You need to install a package from a unknown provider ... thats something I want to reduce as much as possible - at least for the default installs and thats why the NODL also is not adding it by default. I think the auto-ssh with SSH forwarding at least a safer way todo that - from the client side needed code.

But its a nice-2-have that could be added as a config script that will install the needed code on request. Not sure I I find the time for that in the near future - but anybody sending a PR with a config script for that I am open to add it.

openoms, 2019-05-13:
Fully agree with not adding it as default, however it is a good option for someone wanting a remote control GUI and not willing or able to do port forwarding or uses an Iphone. 
Two other remote methods I have used and documented:
*  RTL access through HTTPS/SSL (needs port forwarding and a dynamicDNS)
https://github.com/openoms/bitcoin-tutorials/blob/master/nginx/README.md
* Zeus access with a Tor auth cookie (needs and Android phone and Tor activated)
https://github.com/openoms/bitcoin-tutorials/blob/master/Zeus_to_RaspiBlitz_through_Tor.md

Out of these using ZeroTier is the easiest to set up, but also the most trusted. Would worth a menu entry. I will happily look into it.

openoms, 2019-06-19:
I wrote up the method to connect with ZeroTier: https://github.com/openoms/bitcoin-tutorials/blob/master/zerotier/README.md

I have tested it successfully with RTL and Zeus, but did not see a demand for it so far.
Will see about implementing it to the service menu if anyone wants it, the manual setup takes only a few steps on the RaspiBlitz,

justinmoon, 2019-10-13:
@rootzoll @openoms Are there any other ways you guys might support remote RPC?

openoms, 2019-10-13:
> @rootzoll @openoms Are there any other ways you guys might support remote RPC?

There are plenty of ways to do this, less secure to most secure:
* open ports directly (should be avoided)
* serveo.net reverse ssh tunnel (3rd party risk - similar to ZeroTier)
* SSH tunnel through own VPS: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-setup-port-forwarding-with-a-ssh-tunnel
* Tor Hidden Service (easy if the client is Tor enabled (not on iOS yet), can be done like this: https://github.com/openoms/bitcoin-tutorials/blob/master/electrs/Tor_Hidden_Service_for_Electrs.md)

justinmoon, 2019-10-14:
Why is an SSH tunnel through a VPS more secure than an SSH tunnel directly to raspiblitz? Because you can whitelist just the VPS IP??

openoms, 2019-10-14:
> Why is an SSH tunnel through a VPS more secure than an SSH tunnel directly to raspiblitz? Because you can whitelist just the VPS IP??

You are right that reverse tunnel has a similar risk as forwarding a port on a router. The access-controll is either done on the VPS or on the router.

The reverse SSH tunnel is beneficial when the user does not have access or knowledge to forward a port on their router, and can be automated.

The Tor Hidden service can achieve this without either a public server(VPS) or the need to modify the router settings.

Same described here fro the BitBoxBase: https://base.shiftcrypto.ch/networking.html

rootzoll, 2020-06-23:
OK feature is merged (#1253) ans will be part of the v1.6RC2

rootzoll, 2020-06-24:
Maybe todo for another release:
- [ ] LND TLS update (hard todo automatically if added to private network)
- [ ] device stays in private network after update/recovery

rootzoll, 2020-07-24:
@21isenough I tested the on update/recover the raspiblitz jopins the zerotiwer network with a new address .. but that should be good for the v1.6 release.

But looks good so far. Will be part of final v1.6. Closing issue.

closed 2020-07-24 by rootzoll