raspiblitz/raspiblitz #602: The LND service is running with error v.1.2

btc21cluib, 2019-05-13:
After new SDK/HDD/Power.plug/Raspberry/Blockchain download and sync. - wallet recover 
But still LND not working

               RaspiBlitz v1.2  21btc.club
               bitcoin Fullnode + Lightning Network 
        ,/     -------------------------------------------
      ,'/      CPU load 0.75, 0.82, 0.73, temp 50°C 82°F
    ,' /       Free Mem 605M / 927M  Free HDD 613G (66%)
  ,'  /_____,  ssh admin@192.168.0.197 ▼27.8MiB ▲11.6MiB
 .'____    ,'  
      /  ,'    bitcoin v0.17.1 mainnet Sync OK 100.00%
     / ,'      DynDNS 21btc.club.master-node.cc 8 connections
    /,'        
   /'          LND   Not Started | Not Ready Yet
               
When I try unlock wallet manually

admin@21btc:~ $ lncli unlock
Input wallet password: 
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

After reboot is the same problem
First comes
9 times
[image]

then 
* The LND service is running with error.
****************************************
If you just started some config/setup, this might be OK.

4 RESTARTS DETECTED - LND might be in a error loop

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

(exit after too much restarts/unlocks - restart to try again)

debug here
https://termbin.com/ulve1

rootzoll, 2019-05-13:
The cat/tail errros are not a critcal problem - just a debug output that you can ignore.

Reading your debug - your LND has the following problem and shutsdown:

`2019-05-13 13:51:20.976 [ERR] SRVR: unable to create server: unable to read bucket for chain=f0b74181eeff906b6c254d475c7d5ff85793e29bd757f4eee96eb41148fa338c00000001`

Havent seen that before, might be a data corruption if its not able "to read" stuff - best would to check hardware if power supply needs improvement for the future and research on LND GitHub if there is a fix for this error.

btc21cluib, 2019-05-14:
Hardware test is ok. i restorer on a running empty wallet like discreet from here. 
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#i-dont-have-a-lan-port-on-my-laptop---how-to-connect-to-my-raspiblitz
but after restorer my old back up I got the same error
LND service is running with error.
my debug log file
https://termbin.com/5ybc

rootzoll, 2019-05-15:
@btc21cluib did you find any detail on that error message on the LND GitHub? Or any repsonse on it from the LND Slack? https://dev.lightning.community ... if they say its due to "data corruption", please ask how to repair the LND data then.

btc21cluib, 2019-05-16:
@rootzoll I open a new issues here at LND-still waiting for a solution
https://github.com/lightningnetwork/lnd/issues/3079#issuecomment-492793232

fluidvoice, 2019-05-16:
> @rootzoll I open a new issues here at LND-still waiting for a solution
> [lightningnetwork/lnd#3079 (comment)](https://github.com/lightningnetwork/lnd/issues/3079#issuecomment-492793232)

I think the solution is to re-download and re-sync the blockchain.  I think the corruption is with the blockchain DB not LND.

btc21cluib, 2019-05-16:
@fluidvoice  Thanks for your response.
I was thinking the same before also, but after two times download form other HDD and one time via torrent download, I got the same error. Then I start with the same blockchain data and create a new empty wallet (no funds, no open channels) and all is working nice. After this I recover my old Wallet .. and the same error.

fluidvoice, 2019-05-16:
> After this I recover my old Wallet .

what steps/commands did you perform when you did this?

amcitaly, 2019-05-16:
Me too, same error
https://termbin.com/w0a5

btc21cluib, 2019-05-17:
@fluidvoice  like Just like this explains  
 1) Recover LND data
https://github.com/rootzoll/raspiblitz/blob/dc5d8c6ee1968cf5aed98efb9d3ecb0ec8f1b4e1/FAQ.md#1-recover-lnd-data

fluidvoice, 2019-05-18:
> Me too, same error

C'mon man, you can be more explicit than that.  Which error?  That log file did not have "unable to read bucket" in it.

fluidvoice, 2019-06-01:
"edge not found" problem update: https://github.com/rootzoll/raspiblitz/issues/562#issuecomment-497939315

nyxnor, 2021-05-17:
Please close this issue.

closed 2021-05-17 by rootzoll