raspiblitz/raspiblitz #604: LND Service doesn't syncing anymore

ediying, 2019-05-14:
Hi,
i have an issue with my node and i really dont know whats the problem. After a reset (power off) 2 day ago LND isn't syncing with the network.
I would really appreciate if somebody could help me out.

https://termbin.com/gc3k
https://termbin.com/u8bd

rootzoll, 2019-05-15:
It seems like LND has the following problem:
```
Failed to process consensus server notification (name: `blockconnected`, detail: `failed to store sync information 00000000000000000009f5b281217cabdba84f1db7d965448efa43d4216c6922: failed to fetch block hash for height 575949: block not found`)
```
Not sure what that is. Could be a corruption of the blockchain data. Did you do a torrent download or dis you copyied the blockchain from another computer in the beginning?

ediying, 2019-05-15:
I started with raspiblitz v1.0 and downloaded the blockchain data via torrent. After the update to v1.2 raspiblitz still worked fine. Then i opened a channel (with 5sat/byte fee for testing). One or two days later i did a hard reset and this issue occured.

rootzoll, 2019-05-15:
So you can try to get more info on the "block not found" error and how to fix it from LND GitHub/Slack .. or try to reset/redownload the blockchain data and see if that fixes it.

fluidvoice, 2019-05-15:
I think the problem may not be with LND but with bitcoind...
The log shows 
> Unable to synchronize wallet to chain: t: short read

Lets see these:
`/home/admin/XXdebugLogs.sh`

ediying, 2019-05-16:
https://termbin.com/r0xu
https://termbin.com/5dp1

fluidvoice, 2019-05-16:
> Hi,
> i have an issue with my node and i really dont know whats the problem. After a reset (power off) 2 day ago LND isn't syncing with the network.
> I would really appreciate if somebody could help me out.
> 
> https://termbin.com/gc3k
> https://termbin.com/u8bd

as per this LND issue w/same error, your node may just still be finishing it's rescan which can take a long time: https://github.com/lightningnetwork/lnd/issues/3043#issue-440361585

What does your LND log look like now?...
`sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`

ediying, 2019-05-17:
I decided to setup my raspiblitz totally new from scratch.
Thx for helping me out.

closed 2019-05-17 by ediying