raspiblitz/raspiblitz #605: Unable to Restore Old Wallet

rektlesspinko, 2019-05-15:
Having upgraded to 1.2, my lnd node became corrupted ([ERR] SRVR: unable to start server: edge not found of which there are a few open incidents with users having similar issues going to lnd v0.6).
I decided to rebuild my node and then use my seed and channel.backup to restore (Choose SEED+SCB - Seed & channel.backup file option)

The issue I have is that my seed was protected with a password (Password D). However, that password contains special characters ( specifically a single $ character). However, there is now logic built in to 1.2 that prevents the use of special characters so it won't accept the password that my seed is protected with. So is there a way to bypass this logic to enable me to enter the actual password that my seed is protected with? 

 I built my node using  raspiblitz v0.99 and given I created my Password D with that special character in using that version, I guess the same logic preventing the use of special characters was not in place at that point. I don't know whether the fact it contains a special character is going to be an issue even if I can somehow bypass the logic that is preventing me from trying to  restore my wallet with the actual password protecting it or not, but I'd at least like to get to that point and try. Thanks

rootzoll, 2019-05-15:
When doing recovery from seed - at the moment I would still recommend using the LND Desktop Wallet, because it somehow is more relaible to recover on-chain funds from seed (this still on investigation why RaspiBlitz is not able to get the same results) and it should accept special chars in password. I never used the channel.backup with the desktop app, but it should be possible (maybe ask on LND slack).

https://github.com/lightninglabs/lightning-app

Once you recovered all your funds with the desktop app then you can setup I fresh RaspiBlitz and fund it from that lightning desktop app.

fluidvoice, 2019-05-22:
> [ERR] SRVR: unable to start server: edge not found of which there are a few open incidents with users having similar issues going to lnd v0.6).
> I decided to rebuild my node and then use my seed and channel.backup to restore (Choose SEED+SCB - Seed & channel.backup file option)

were you successful at restoring your node and wallet?  Please provide details for others if possible.

fluidvoice, 2019-06-01:
> were you successful at restoring your node and wallet? Please provide details for others if possible.

I think not... 
"edge not found" problem update: https://github.com/rootzoll/raspiblitz/issues/562#issuecomment-497939315

rektlesspinko, 2019-06-03:
Hi all, sorry for the delay in replying. Unfortunately I was not successful in restoring my node using the LND Desktop wallet, but more down to my stupidity than anything technical. I was unable to restore from the seed and passphrase, it didn't like my passphrase so I must have either written it down wrong or got it confused with a previous passphrase. So going to start again from scratch.  Interesting update to #562 where this situation should be avoided in v 0.7.0. That (along with doing a better job recording my recovery seed and passphrase!) should leave me in a better position going forward. Suggest this issue is closed

closed 2019-06-05 by rootzoll